pbatard/rufus #1317: Flash drives are not shown in the list, when external HDD is connected

Jon-Smit, 2019-05-22:
Starting versions 3.+ I've met such problem.

When there is no external HDD connected, flash drives appears in the list.
When external HDD connected, there are no flash drives in the list.

Windows 7 Ultimate 64bit.

[image]
[image]

pbatard, 2019-05-22:
Thanks for the report, I'll look into it. But can you please also provide the logs (no screenshots, copy/paste please) after you press <kbd>Alt</kbd>-<kbd>.</kbd> (to enable USB debug).

This may contain additional information that I could use to troubleshoot this issue.

Jon-Smit, 2019-05-23:
I added comments in log file with # mark what I'm doing at current moment.
[rufus.log](https://github.com/pbatard/rufus/files/3213940/rufus.log)

Jon-Smit, 2019-05-23:
> 
> 
> Thanks for the report, I'll look into it. But can you please also provide the logs (no screenshots, copy/paste please) after you press Alt-. (to enable USB debug).
> 
> This may contain additional information that I could use to troubleshoot this issue.

pbatard, 2019-05-24:
Thanks for the additional logs. I've been trying to replicate your issue, but I'm afraid I can't. None of the hardware I tested (Win10, Win7, Win8, using USB 3.0 and USB 2.0 devices) seemed to exhibit this problem.

Moreover, from looking at the code, I don't believe this is a Rufus bug but an issue higher up in the system.

The thing is we're not even getting the second `Processing 'USBSTOR' device:` message that we should get for your flash drive (there should be two messages like this, one for the HDD and one for the flash drive), and there really is nothing in our code that can break the loop that iterates over `USBSTOR` devices as long as the system APIs report them.

Thus, it looks to me like, on your system, `SetupDiEnumDeviceInfo()` is only turning up one device when it should report two, and I'm afraid this is not something Rufus has any control on. Possibly the problem is with `SetupDiGetClassDevsA()` failing to properly fill in the device list that `SetupDiEnumDeviceInfo()` can process, but again, that is not something I have any control on.

And yes, I understand that you may say something like "But Device Manager has no trouble seeing both devices!", to which I will reply that different applications use different APIs to enumerate devices, and a problem with one API mode does rarely translate to other APIs, and I can't change the APIs currently being used by Rufus to list devices, as I wouldn't be able to access the data required for formatting otherwise.

All in all, I suspect this is a very limited issue (else I'm pretty sure I would have had plenty of reports about it), that most likely only manifests itself with a specific set of hardware, and probably only on Windows 7 (even if your flash drive is USB 2.0, the fact that Windows 7 doesn't have native USB 3.0 drivers, and that I have seen way too many buggy instances of those in the past doesn't provide me with great confidence that the custom USB 3.0 driver your system uses is not screwing up enumeration for USB 2.0 devices...). As such, considering that Windows 7 is reach End of Life in about 6 months, unless people using Window 8 or later start to reporting a similar issue, I don't think I will investigate this any further.

Of course, I do understand that this is not the answer you were hoping for, but I'm afraid that to process further, I'd have to have access to the same kind of hardware you have, and go into a time consuming debugging of API calls, which is not something I am prepared to do for an issue that can easily be worked around by unplugging external HDDs.

As such, I will close this issue. Of course, is anybody else is facing a similar problem they will still be able to add their comments or reference this.

closed 2019-05-24 by pbatard

Jon-Smit, 2019-05-24:
Just one thing. As I wrote when opened this case: "Starting versions 3.+".
Rufus version 2.9, for example, doesn't have such problem. Everything works just fine. So there **is** something, what differs in 3.+ versions comparing 2.+ versions.

pbatard, 2019-05-25:
In that case I'm going to have to ask you to tell me exactly which version of Rufus starts to display the issue. There are quite a few versions between 2.9 and 3.0, so please download and test the versions you find [here](http://rufus.ie/downloads/) and let me know precisely which version started to exhibit the problem.

But again, because I cannot replicate this issue, nobody else seems to be reporting it and there's a very easy workaround, I'm going to treat it as very low priority.

Jon-Smit, 2019-05-25:
Ah, OK! Sorry, I misunderstood version numbering. I thought 2.9 is the latest version before 3.0. In fact it is not a version 2.90, but 2.09. So, the problem showed up starting version 2.13 and higher. Latest version which works fine is 2.12.

JonnyTech, 2019-05-25:
[How comes the portable and regular versions are binary identical?](https://github.com/pbatard/rufus/wiki/FAQ#How_comes_the_portable_and_regular_versions_are_binary_identical)

> The way it works is like this: if the file name contains the letter p, then the code will run in portable mode. And if there is no p, then regular mode is used.

pbatard, 2019-05-25:
Thanks for the update. I'll look at the changes between 2.12 and 2.13 when I get a chance, to see if I can figure out something.

> It means there is no difference in funcionality for these "versions". Why?

It's not because 2 files are binary identical that they will provide the same functionality. Even if the bits are the same, an executable can have code that checks its name during startup, to decide what functionality it should provide (this is something that's commonly used by the `busybox` minimal set of commandline utilities for Linux). So that's why you can have a single executable that, if named a certain way or another, will provide portable or non-portable functionality. This is indeed explained in the FAQ link @JonnyTech shared.