pbatard/rufus #1321: After crating USB with some distro it's kinda impossible to format the usb (need special program)

spirulin, 2019-06-02:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
So I create bootable USB with some distro and the USB is unformattable. At least on Windows 10. I try to format it the normal way - right click - format - "Acess denied" is what I get.

When creating the USB I just pick the iso and follow the recommended settings (I change nothing). 

Then I try with CMD to format it. No luck - Acess denied.

I need to download EaseUS program so I can finally format it.

Log
---
```
<FULL LOG>
```

pbatard, 2019-06-02:
> So I create bootable USB with some distro

Can't help you unless you tell precisely which distro and the exact image you downloaded.

For the record, there does exist some images (Chromium ones) that trick Windows into declaring the drive as unformatable (#759), and this has nothing to do with Rufus. It's just a Windows __bug__ (a bit like the one that will crash Windows if you plug an USB drive with a FreeBSD image on Windows XP or Windows 7). Rufus does what it can to work around such bugs, but when the issue is __internal__ to Windows, there's only so much we can do. Even if some other software does appear to alleviate these issues, adding a lot of workarounds just for a specific image is not always a great use of development time...

You may also try the steps detailed in #1319 to see if they help.

But at any rate, unless you provide the exact details of the image you use, so that we can try to replicate what you see, there's no way you should expect a fix, as you don't provide any of the details we need. A log of how you created the drive in Rufus would also be nice, as I don't know what specific version of Windows 10 you are using and other stuff. Please understand that the checklist is not optional, and if you treat it as such, then don't be surprised if the issue you are reporting is ignored.

spirulin, 2019-06-03:
The latest one was Ubuntu 19.04. Created it with no changes in rufus settings and it did the job. I was able to install Ubuntu but I saw it wasn't for me, so went back to Window 10 Home. Plug to format the CD - access denied. Before that had the same problem with Lubuntu 19.04. I get Access denied when trying to format - both from the normal way and cmd.

pbatard, 2019-06-03:
Did you use DD or ISO mode?
And what happens when you try to reformat them in Rufus. Just select _Non bootable_ under _Boot selection_.

You do realize that, if you are not able to reformat a drive outside of Rufus, then this is a Windows problem (most likely an environmental one), not a Rufus one, and I don't really have the ability to patch Windows...

Also, access denied is a pretty clear cut error. It means another application is accessing the drive. So you need to find what that application is. If your drive is slow, it may very well be Windows creating a bunch of files under `System Volume Information` as it tends to do that on its own when a drive is inserted. In that case you just need to wait (or force an eject and then re-plug the drive). Or it could be a security solution. Again, I __really__ don't see what you expect me to do when it comes to what is clearly an external problem to Rufus. I don't have a magic wand to fix __external__ applications if they are the ones with a problem, and there's nothing fancy with how Rufus formats a drive that can cause "Access Denied". Rufus relinquishes all access and locks to the drive when it completes its operations, so there's absolutely nothing more I can do here to prevent access issues. There are no hidden "locks" that are being held, and there aren't a zillion ways to create a bootable drive either, which means that, if you expect me to say _"Oh, I know, I can just create the flash drive in __this__ different way and it will solve the access denied issue"_, you're in for a long wait...

spirulin, 2019-06-03:
ISO mode.

Just tried Manjaro in DD mode - same thing. I need the special program so I can format it.

pbatard, 2019-06-03:
Again:

> And what happens when you try to reformat them in Rufus? Just select Non bootable under Boot selection.

pbatard, 2019-06-09:
No answer &rarr; Closed.

closed 2019-06-09 by pbatard