pbatard/rufus #1322: Cant write iso image (Qubes os) on flashdrive

qubesfan, 2019-06-03:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [X] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi all,
For 2 days now I'm trying to write an iso file (Qubes-os) on my flashdrive. It appears as if everything is fine untill I want to reboot and the Qubes install won't start after selecting the right usb drive in bios.  The log files keep giving the errors: "timeout while waiting for logical drive" and "drive has an unknown Master Boot Record"

How can i fix this?

The laptop i'm using has an i7 7700 x64 processor. I have both vt-d and vt-x enabled in BIOS.

Log
---
```
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive3':
● Partition 1 (offset: 530432, size: 27.7 MB)
Opened \\.\PhysicalDrive3 for exclusive write access
Writing Image...
Timeout while waiting for logical drive

Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&1b153018&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[211]: USB\VID_13FE&PID_4300\190085720110701921080F98105
  Found ID[242]: USB\VID_04CA&PID_3016\5&225C1337&0&7
  Found ID[010]: USB\VID_04F2&PID_B5F7\5&225C1337&0&9
  Found ID[241]: USB\VID_090C&PID_1000\18061313010533
Processing IDs belonging to 'USBSTOR':
  USB\VID_090C&PID_1000\18061313010533
  USB\VID_13FE&PID_4300\190085720110701921080F98105
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskHFS256G39TND-N210A______3000'
Found non-USB non-removable device 'HFS256G39TND-N210A' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_MQ01ABD100______AX1P'
Found non-USB non-removable device 'TOSHIBA MQ01ABD100' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskHEMA_16GHEMA_16GB_______PMAP'
  Matched with ID[211]: USB\VID_13FE&PID_4300\190085720110701921080F98105
  Matched with Hub[1]: '\\?\usb#root_hub30#4&1b153018&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'HEMA_16G HEMA_16GB USB Device' (13FE:4300)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1889, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x64B1934E
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 27.7 MB (29061120 bytes)
  Start Sector: 1036, Boot: No

pbatard, 2019-06-03:
1. You truncated the log. Please don't do that as I don't know what version of Windows you are using as well as other information that may be critical to helping you out.
2. Can you try the steps described [here](https://github.com/pbatard/rufus/issues/1319#issuecomment-496481875)?

qubesfan, 2019-06-03:
Thanks for the help, i will let you know if it helped.

Don't know what you mean bij truncated the log? I am using Windows 10.

pbatard, 2019-06-03:
> Don't know what you mean bij truncated the log?

As __explicitly__ stated in the checklist, in an item you checked:

> [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Your log does __not__ start by `Rufus version: x.y.z` which means you __did__ truncate it. A complete log always starts with the line `Rufus version: x.y.z` and no, there is no bug where Rufus might fail to produce that line. If the line is not there, along with all subsequent lines that should exist before `Format operation started`, it means that you truncated the log. Don't do that, as it impairs the ability of people to help you!

pbatard, 2019-06-06:
Any update on this? If not, then I will close this issue.

pbatard, 2019-06-09:
No update &rarr; Closed.

closed 2019-06-09 by pbatard