pbatard/rufus #1328: [FAIL] Could not locate 'efi\boot\bootx64.efi'

loic2665, 2019-06-13:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------

Hi,

(first of all, sorry for my medium English, i hope i can be understood ^^)

I've been using rufus for years now, such a mervelous software…
But yesterday, i was not able to boot from my disk.

I'm repairing a computer and the computer doesn't find the `bootx64.efi` file.

(picture of EFI Loader)

[image]

The two files are here in their respective folder. (2 pics here) 

The worst part is i've been repairing this computer 2 years ago, and I was still using rufus. (and it worked)

[image]

[image]

(Windows doesn't show the two partitions) these two screens capture are made on my linux machine. Rufus is running in a Windows 10 virtual machine, i've Always done that before)

I hope i given all information needed, i uploaded the log file in the Log part, i could copy paste, but it's a pain to scroll then. This is the log after i burned the iso in the disk.

Last information, the iso file is the rufus downloaded one. I just discovered this option after a while. I already tried with my (from Microsoft website) iso, but still Don't want to boot...

I'm here for any questions ;) 

Have a nice day ! :) 

Log
---
```
[rufus.log](https://github.com/pbatard/rufus/files/3284907/rufus.log)
```

pbatard, 2019-06-13:
Hmm, that's interesting. Both the log and your report show that the `efi\boot\bootx64.efi` definitely exists on the primary NTFS partition (this is the file that matters, the other `bootx64.efi`, on the small FAT partition labelled "UEFI_NTFS" is actually the bootloader that ultimately displays the `Could not locate 'efi\boot\bootx64.efi'` error and, of course, it's always present).

So, technically, there is no reason why UEFI:NTFS should fail.

__However__, one very interesting thing we are seeing as UEFI:NTFS runs is this:
```
[WARN] An NTFS service is already loaded
```
What this means is that, as far as UEFI:NTFS is concerned, there already appears to exist an NTFS driver, and therefore we don't install our own NTFS driver.

So we have tow possibilities here:
1. Your UEFI firmware really includes an NTFS driver, but that driver is faulty or incomplete, and cannot actually access the `efi\boot\bootx64.efi` residing on the NTFS partition.
2. Your UEFI firmware is __pretending__ it has a driver that can access the NTFS partition, when it doesn't, in which case you will also see a similar error.

Right now, I'm pretty confident these are the 2 only possibilities, and that this is not a problem or bug with the UEFI:NTFS bootloader.

At any rate, __this is an issue with your UEFI firmware__ and not Rufus (or UEFI:NTFS which is the component used by Rufus to boot from NTFS in UEFI mode), and I can't really think of a good way to work around it.

For one thing, even if that is possible, I don't think I want to modify UEFI:NTFS to unload a potential "faulty" UEFI NTFS driver, provided by the firmware, so that we can install our working one. Unloading firmware provided drivers during boot goes way beyond the mandate of what UEFI:NTFS should do, and goes against what the application is supposed to do.
Secondly, if this computer manufacturer is providing a problematic firmware, it is in the best interest of users of that computer that they fix it, rather than have other software work around it. In other words, if an UEFI firmware __advertises__ that it can handle an NTFS partition, which is what your firmware does here, then it better be able to do so, by either providing you with the option to boot from it in UEFI mode (Do you see 2 options to boot your USB drives in UEFI mode? Coz if your UEFI firmware really has a working NTFS driver you should), or by actually embedding a proper NTFS driver that can, rather than a "thing" that pretends it can provide file system access to an NTFS partition but doesn't.

For the sake of other people who might be confronted to this issue, can you please provide the exact make and model of the computer (and/or motherboard) you are using, and that exhibits this behaviour?

loic2665, 2019-06-13:
Hi !

Thanks for your answer, 
for the warning, with the NTFS service already loaded, I think I understand what you said, but I have no clue of how to unload the firmware NTFS driver. I understand the fact that this is not the job to the main program.

For the computer model, I was thinking of repairing the computer in hours this morning, but for now I don't have access to it until Saturday / Sunday... But any suggestions are welcome. I know this is an Asus computer with a model starting with J..... But it's too ambiguous.

For the behaviour, at boot before posting, I press the escape key multiple time to get the menu, then I select the drive I connected, before i turn on the computer. Like always on my computers, this is really the first time ever I got that then now I'm stuck.

For the boot menu I know it was showing me 3 options :

```
    - The windows Boot Manager
    - The UEFI hard drive which Rufus just burned the ISO (in the UEFI mode) (starting with UEFI on screen)
    - The BIOS setup 
```

I didn't have the mbr mode since I wanted only the EFI mode.

(I'm writing on mobile phone sorry of there is any typos / grammar errors)

pbatard, 2019-06-13:
> I have no clue of how to unload the firmware NTFS driver.

I'm not sure you can. As I said, the issue appears to be with the hardware manufacturer of your computer (which, __again__, I would really like you to disclose: What is the brand and model type of your computer or motherboard?), so it's up to the manufacturer to fix that with a firmware update.

Also, if you didn't get 2 choices for UEFI boot of the USB devices, then it means that the firmware for your computer is making it look like it can access the NTFS partition, but it actually doesn't have an NTFS driver for it, which, again, is something your manufacturer should fix, as you shouldn't pretend a partition is accessible when you actually don't have a file system driver for it.

loic2665, 2019-06-16:
Hi,

Some news.

Here is the computer model.

[image]

Here is the boot menu.

[image]

BIOS menu / specs

[image]

And secure boot obviously is disabled.

loic2665, 2019-06-16:
Ok I have some (bad) news, i don't know if it's normal, but I looked with GParted to see the partition table and after a burn With rufus, i got... this. 

[image]

I press ignore, if not, i can't continue.

[image]

Then "Valider". I can re-create the GPT partition table, but then I lost all but I don't have the error anymore. After flashing (again) with Rufus, the error come back... 

I updated the BIOS from the version 207 to 208, but still the same exact error. File not found, but it exists.

I even changer the SATA controller but same issue. Now I'm downloading a Non Official (but not cracked) windows ISO file with a .wim file less than 4Gb, we will see with one NTFS partition and one FAT21 Large partition... Maybe It can't read the bootx64.efi file from the second NTFS partition where the Windows Installer is located, who knows...

loic2665, 2019-06-16:
Ok, I can boot, BUT, only with the older (1704) unofficial ISO where is is a .Wim file less than 4Gb... burned it in FAT32 partition. I will try on my other computer if I can boot with the newer version...

loic2665, 2019-06-16:
Last answer, this is solved, this is a bug FROM the manufacturer.
My computer could boot on the EFI file with the official ISO provided by rufus. 

So, for this model, this bug is not resolvable. Newer version of windows can't boot properly on it. (never seen that).

Thanks again for your help.

closed 2019-06-16 by loic2665