pbatard/rufus #1333: Converting multiple 'LABEL' in single line in grub.conf

soohoon, 2019-06-21:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Sometimes grub.conf has multiple LABELs in single line something like

linuxefi /images/pxeboot/vmlinuz inst.stage2=hd:LABEL=CentOS\x207\x20x86_64 inst.ks=hd:LABEL=CentOS\x207\x20x86_64 acpi_osi="Linux" quiet

But the conversion code in iso.c converts LABEL once so the second LABEL doesn't get converted.

iso.c:
fix_config() -> replace_in_token_data()

Would it be possible to call replace_in_token_data() multiple times until nothing's converted?
Or any better solutions will be appreciated.

Thanks,
Soohoon.

pbatard, 2019-06-22:
I'll see what I can do, but I would very much like to have an example of an __official__ Linux ISO with multiple `LABEL`'s, as adding features to the code just to help people who create super custom ISOs and situations that very few people are expected to encounter automatically gets assigned a __very low__ priority on my TODO list.

soohoon, 2019-06-26:
It was my own ISO but I think the example in the first comment is everything.
Anyway my fix in rush looks like.

--- a/src/iso.c
+++ b/src/iso.c
@@ -293,8 +293,12 @@ static void fix_config(const char* psz_fullpath, const char* psz_path, const cha
 		usb_label = replace_char(img_report.usb_label, ' ', "\\x20");
 		if ((iso_label != NULL) && (usb_label != NULL)) {
 			if (replace_in_token_data(src, (props->is_grub_cfg) ? "linuxefi" : ((props->is_conf) ? "options" : "append"),
-				iso_label, usb_label, TRUE) != NULL)
+				iso_label, usb_label, TRUE) != NULL) {
 				uprintf("  Patched %s: '%s' ➔ '%s'\n", src, iso_label, usb_label);
+				if (replace_in_token_data(src, (props->is_grub_cfg) ? "linuxefi" : ((props->is_conf) ? "options" : "append"),
+					iso_label, usb_label, TRUE) != NULL)
+					uprintf("  Patched again %s: '%s' ➔ '%s'\n", src, iso_label, usb_label);
+			}
 		}
 		safe_free(iso_label);
 		safe_free(usb_label);

pbatard, 2019-06-27:
> It was my own ISO

Okay. I'm going to defer this issue then, since it won't affect many people

> Anyway my fix in rush looks like.

Ugh!

Sorry, but there's no way I can accept something like that, because then someone who uses 3 labels will want to add another branch, and this "fix" is simply not up to the standard I have for code that I want to add in Rufus. If there is an issue, it should to be fixed at the source wherever possible rather than by adding a quick band aid at the first location where you can do it.

&rarr; Deferred.

soohoon, 2019-06-27:
> Okay. I'm going to defer this issue then, since it won't affect many people
All yours.

> Ugh!
I told you it's "my fix in rush".
You can simply convert it to a loop or anything that meets your standard.
It's the simplest fix without touching replace_in_token_data() which will drag in more changes which is I wanted to avoid.
I tried to provide more info but looks like it was a wrong day.

pbatard, 2019-06-28:
Not at all. The example you provided is clearly a short term fix. The long term fix is to have multiple substitution handling in `parser.c`, which is both the logical place to have it reside and the best way to be in a position to alleviate similar issues in the long run (instead of adding band aids here and there).

But hey, feel free to pass a developer's decision to want to have what they consider a more appropriate fix as them having a "wrong day". Can't help but wonder how you'd qualify the other possibility I considered, the minute you said it was a custom ISO, which was to close this issue altogether (instead of deferring it) and wait for a mainline distro to start to use multiple labels in their config files before I would consider adding a multiple labels fix (since I can always use development time that I save on issues that only a handful of people are expected to encounter on more worthwhile endeavours)...

git4wht, 2019-12-23:
I encountered the same problem. When I specify the `inst.ks` path. My current approach is to replace `hd: LABEL` in the file when creating the ISO.

Schmill, 2020-02-05:
Just been caught out by this same issue.
We use a (customised) CentOS ISO image and after much head scratching determined that the only difference between those USb drives that worked, and those that didn't was that some had a non-default name specified in Rufus.

Ours takes the following format:

```
search --no-floppy --set=root -l 'defaultName'

### BEGIN /etc/grub.d/10_linux ###
menuentry 'Install MySystem' --class fedora --class gnu-linux --class gnu --class os {
	linuxefi /images/pxeboot/vmlinuz inst.ks=hd:LABEL=newName:/ks.cfg inst.stage2=hd:LABEL=defaultName quiet consoleblank=0
	initrdefi /images/pxeboot/initrd.img
}
```

I've changed the actual names used to "defaultName" (the one in the ISO) and "newName" (the one specified in Rufus) to illustrate the issue.

pbatard, 2020-02-05:
Thanks for the report. I'm still planning to fix the issue, but I'm afraid that, as long as it only occurs with custom ISOs and not official ones, it's quite low on my list of priorities. But I will get there, eventually.

commit 590b89a5 (2020-03-23): [parser] add multiple occurrence support to replace_in_token_data()

* Closes #1333
closed 2020-03-23 by pbatard