pbatard/rufus #1335: Computer froze and reset when i tried to copy particular iso to particular SD card.

slrslr, 2019-06-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [-] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [-] I also tried one or more of the following:
  - [-] Using a different USB drive.
  - [-] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [-] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I tried to copy an "AcronisBootableMedia.iso" to my flash drive using this Rufus (version and details in the following log). This iso was created by the Acronis True Image WD edition. And two times i launched the process of copying the iso (it was running for a few seconds) always computer frozen resulting in frozen sound, black screen and after like 5 seconds i see computer automatically restart/reset. I already used this Rufus version and this computer and this USB stick reader and this USB port to successfuly create bootable USB disk, but i am not sure i tried on this particular SDHC card and i never tried on this .iso image. I want to help fix this issue, but tell me what to do so i do not need to reset my computer multiple times (i do not have time to waste, my computer boots extremely slow). Thank you in advance.

Log
---
```
<FULL LOG>

Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0405
Will use default UI locale 0x0405
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'cs-CZ'
Found USB 2.0 device 'Maxtor Basics Desktop USB Device' (0D49:7410)
Could not get device number for device \\.\T:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\V:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\Z:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\T:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\V:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\Z:: [0x00000001] Nesprávná funkce.
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'WD Elements 25A3 USB Device' (1058:25A3)
Could not get device number for device \\.\T:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\V:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\Z:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\T:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\V:: [0x00000001] Nesprávná funkce.
Could not get device number for device \\.\Z:: [0x00000001] Nesprávná funkce.
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
```

pbatard, 2019-06-23:
1. Please confirm that the image you are using can be written without issue on other media than this specific SDHC. If it's only that specific SDHC that's affected, I'm afraid I'm going to have to decline investigating, as this would either required a lot of time to help a __very limited__ number of users (those with that specific SDHC card, or possibly the same SD card reader as you have), which isn't exactly a good use of my __limited__ development time, or it might end up being a complete wild goose chase that will result in declaring that writing this specific image is simply incompatible with this SDHC card/SD card reader.

2. If you find that the issue occurs with different media, I would need you to upload your ISO somewhere, as the __only__ way I'm even going to begin consider looking into this problem is if I can access that ISO. Once again, it's just way too time consuming shooting in the dark and hoping that one of these shots may address the issue. This is just not how software development work: To fix an alleged software problem, one must first be able to consistently replicate that problem on multiple environments (including vanilla versions of the OS), which means the developer must have the required elements to replicate the problem. Unlike what movies would like you to believe, unless it's a very critical situation, a developer and especially an Open Source one will not invest time working with an end-user, whose machine is the only environment that seems to be able to produce the issue, to try to get to the root of the issue. Once again, we just don't have that kind of time on our hands, and, unless an issue affects more than a handful of users, we're not going to lose much sleep if there exists a very specific and hard to replicate set of conditions here and there that produces an issue. There are way too many factors that can play negatively against some usage scenarios to have the luxury to try to address the all...

3. Are you writing the image in DD or ISO mode? Provided that your image is an ISOHybrid, you should be prompted to write it in either of these two modes. And if it is an ISOHybrid, what happens if you use other software such as win32diskimager to write it? Do you see the same issue?

pbatard, 2019-06-27:
Any update for the questions I asked above?

If not, I will have to close this issue.

slrslr, 2019-06-27:
i should update within a few days

slrslr, 2019-07-01:
@pbatard 

1. i just tried to connect different USB storage device (flash pendrive) into different USB port.
and tried to write same .iso and it also crashed my computer as before. It wiped partition scheme and such actions but the crash happened when it said something like writing partition scheme or filesystem. So while i tried multiple card reader and cards and also this flash drive, i think problem may be in the .iso

2. so i uploaded it and sent download link to the p**e@akeo.ie

3. it prompted me and i selected default mode iso

pbatard, 2019-07-01:
Seeing no issue there. 3 tests (2 in ISO mode, one in DD mode) and no crash:
```
Rufus x86 v3.5.1497
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x12D3F3A7
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16013878272 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 31277169, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ACRONIS_MEDIA'
  Size: 345.1 MB (Projected)
  Uses: EFI
Using image: AcronisBootableMedia (Issue #1335).iso (529.6 MB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive6':
● Partition 1 (offset: 32256, size: 14.9 GB)
● Partition 2 (offset: 16013910528, size: 31.5 KB)
Opened \\.\PhysicalDrive6 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Using zeroed MBR
Found volume GUID \\?\Volume{0c2f472c-9acd-11e9-9a10-88d7f6e01c0c}\
Successfully remounted \\?\Volume{0c2f472c-9acd-11e9-9a10-88d7f6e01c0c}\ on F:
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: F:\C1.png (1.5 KB)
Extracting: F:\C2.png (1.8 KB)
Extracting: F:\C3.png (1.5 KB)
[... 222 lines cut ...]
Created: F:autorun.ico

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'ACRONIS_MEDIA'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes
Local time: 13206486121
Next update check in 60298 seconds.
```

NB:

1. I tested mostly with 3.5 because this is the latest version and the one you should obviously use that one, as, if there is any issue with the earlier versions, I will obviously need to confirm that it applies to latest before even considering a fix. I did run the same test with 3.4 though, and found no issues
2. In case you want to point out that you are using the "portable" version of Rufus whereas I am no, please note that there is actually no binary difference between the two versions, and that it's simply impossible that portable vs. non-portable could be the cause of your issue.
3. Only thing I see is that you seem to be using a fairly old version of Windows 10 (Build 10586  whereas latest in Build 18362). Considering that Windows has been known to have __bugs__ with its handling of partitions, that led to crashes, and that Microsoft does try to fix Windows issues, I would strongly advise you to upgrade to the latest feature update and test again.

At any rate, it is pretty clear to me that your issue is environmental and not something that has to do with Rufus, else I would be observing the crash with your ISO. I would therefore encourage you to test on other Windows platforms (preferably up to date ones) so that you might be able to identify the problem, as I'm afraid none of the data I have allows me to do anything that could even remotely help you with your issue.

slrslr, 2019-07-04:
thank you, i think i will not be updating MS Win so i think we can not move on so i am closing this.

closed 2019-07-04 by slrslr