pbatard/rufus #1336: already mounted, but volume GUID doesn't match

prabhakaranp05, 2019-06-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am posting this peculiar case of error after some basic investigation. 
I need to prepare a FREEDOS bootable USB drive. There's a case where I am required to flash old devices ( Target machines ) , so I have to check the option "Fixes for old BIOSes". When I do this operation on a Windows 10 PC, the error "already mounted, but volume GUID doesn't match:" occurred (repeatedly with certain USBs on certain PCs). It successfully prepared the same USBs when the "Fixes for old BIOSes" is unchecked ( but obviously it doesn't work on my old machines which need to be flashed ). From the logs you can understand that extra partition created ( for BIOS fixes ), causes the confusion ( between the volume mounted by OS & that volume the tool expects to mount ). Kindly correct me, if I missed out on something.
Else, we will require an urgent fix.

Log
---
```
Rufus x86 v3.5.1497
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:312B)
Found non-USB removable device 'NVMe KXG50ZNV512G NVM' => Eliminated
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1888, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x6D63DDBD
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15534653440 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 14.5 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Adding extra partition
Reserved 1 tracks (31.5 KB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{66567564-990e-11e9-aca4-54bf648e88a5}\
E:\ already mounted, but volume GUID doesn't match:
  expected \\?\Volume{66567564-990e-11e9-aca4-54bf648e88a5}\, got \\?\Volume{7c7e2169-7818-11e9-ac9b-54bf648e88a5}\
Could not remount \\?\Volume{66567564-990e-11e9-aca4-54bf648e88a5}\ on E:\: [0x000000EA] More data is available.
E:\ already mounted, but volume GUID doesn't match:
  expected \\?\Volume{66567564-990e-11e9-aca4-54bf648e88a5}\, got \\?\Volume{7c7e2169-7818-11e9-ac9b-54bf648e88a5}\

Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:312B)
Found non-USB removable device 'NVMe KXG50ZNV512G NVM' => Eliminated
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1888, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0F884F0F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15536134656 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 30344076, Boot: No

```

ProfessorStrawberry, 2019-07-01:
rufus 3.5: same here with a windows 1903 image

with rufus 3.4 i had not such problem

pbatard, 2019-07-01:
@ProfessorStrawberry, please run more tests with Rufus 3.4. You will find that it behaves the same.

You are assuming that we introduced a regression between 3.4 and 3.5 with regards to the way partition are mounted (and therefore that there might be an easy fix), but that is simply not correct. None of the relevant code changed, therefore Rufus 3.4 and 3.5 behave the same way in that respect. The only issue here is that it's a semi-random behaviour of Windows 10 not to select the proper partition when we ask it to, which is a known issue that I am well aware of (since I have seen it in forever, but not in a manner that warranted a fix, since it only seems to occur occasionally and retrying usually seems to fix the problem), and for which I am trying to take action in the upcoming Rufus 3.6.

@prabhakaranp05, do you really see this issue consistently?

On my Windows 10 platform I only ever see it occasionally, and retrying a few times is usually enough to make it go away. You may also try to zero the drive first (<kbd>Alt</kbd>-<kbd>Z</kbd>) and cancel after a while, then retry to see if that help.

Also, with regards to your:

> Else, we will require an urgent fix.

This is Open Source software that you got absolutely for free. There's no such thing as the expectation for an urgent fix. On the contrary, if the fix is that urgent to you, then you are more than welcome to look at the current code, test it, then submit a pull request for a fix that we could integrate.

For one thing, if it's really that critical to you I would encourage you to do the following:
- Install Visual Studio 2017 Community Edition (which Microsoft provides for free and that you are fully entitled to use to compile an Open Source project like Rufus)
- Clone the latest Rufus source from this repository
- Build Rufus in Visual Studio and see if the current version improves thing. You may also try <kbd>Alt</kbd>-<kbd>V</kbd> with this version (which forces the use of VDS) which might help.

ProfessorStrawberry, 2019-07-05:
i will test more, but for now i have a windows usb stick. thanks though, keep the work up!

pbatard, 2019-07-11:
I'm going to close this issue for now. You will still be able to comment on it if you want.

closed 2019-07-11 by pbatard