pbatard/rufus #1337: [Feature Request] Support internal HDD

RUSshy, 2019-07-03:


pbatard, 2019-07-03:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_support_internal_HDDsSSDs

Also please bear in mind that, if you have an internal drive that was set as removable/hot plug in your BIOS (which may or may not be available for all drives or for all platforms), you can use <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> to list them from Rufus (but if you do use this __unpublicized__ cheat mode, you are 100% on your own, whether it works for you or not).

pbatard, 2019-07-03:
I'll also add that, if you use a __proper__ USB 3.0 flash drive or USB enclosure, you can easily reach 300 MB/s read or write speed over USB, which isn't exactly what I would qualify as slow.

If you find that your USB stick is slow, try to purchase a better one, coz you definitely have means to perform fast transfers over flash based USB media if you invest in decent hardware.

closed 2019-07-03 by pbatard