pbatard/rufus #1338: Rufus fails to create boot disk with 1T USB drive

Pherbo123, 2019-07-05:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I formatted drive exFat successfully in Win, it's accessible thru File Explorer but Rufus 3.5 detects the USB 1T drive then says it's a fixed disk and it won't proceed with creating disk with Hiren's Boot disk .iso img. Hiren's boot disk creation failed too; it copied some boot dirs to it but stopped at 18%

<FULL LOG>
```
Rufus x86 v3.5.1497
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'WD My Passport 0820 USB Device' (1058:0820)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'WD My Passport 0820 USB Device' (1058:0820)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121597, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x46A7FC66
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 931.5 GB (1000167440384 bytes)
  Start Sector: 2048, Boot: No

pbatard, 2019-07-05:
See that part of the log that says:
```
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
```
Also, you allegedly read the FAQ, because you checked the box. If so, did you read [this fairly prominent entry](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected) that is there precisely for the issue you report.

Please make sure you pay attention to all the elements that are only there to help you!

Since I consider that I have answered your query, I will close this issue.

closed 2019-07-05 by pbatard

Pherbo123, 2019-07-05:
The problem isn't whether or not the USB hard drive was detected,..as I stated in the first sentence of my submission that you allegedly read carefully because you love helping people so much. I checked the fairly prominent checkbox every time; it just won't complete the "Copying ISO files..." part.

Pherbo123, 2019-07-05:
The problem isn't whether or not the USB hard drive was detected,..as I stated in the first sentence of my submission that you allegedly read carefully because you love helping people so much. I checked the fairly prominent checkbox every time; it just won't complete the "Copying ISO files..." part.

 

I tried again, got farther but it failed again.  Probably faulty hardware. Thanks for your attempt to help.

pbatard, 2019-07-05:
I am basing my analysis on the log you provide. The log you pasted shows no attempt whatsoever at copying files. It only shows that your drive wasn't listed.

If you want me to help you, then please provide the __proper__ log, i.e. the one that matches the problem you are facing. I am not aware of a scenario where you shouldn't be able to open the log and copy and paste from it, even if the operation doesn't complete (and even if somehow Rufus wasn't able to display the log, you can still access it with an application such as [DebugView](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview). The more accurate data you provide from the log you paste, the better help you'll receive.