pbatard/rufus #1339: [FAIL] Could not locate '\efi\boot\bootx64.efi': [14] Not Found

stan-tikhonov, 2019-07-06:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hey there!

I keep getting the following error when trying to boot from a usb stick with an Ubuntu image:

[image]

In order to avoid booting via the legacy boot mode (because, apparently, this is what causes problems with Linux bootloader installation in the later stages of Ubuntu installation), I tried to flash the Ubuntu image onto the usb stick under the UEFI:NTFS setting in Rufus. The bootx64.efi is present on the pen drive:

[image]

I have seen similar issues here, and I gather that the problem might be with the bootx64.efi file somewhere in my Windows. If this is the case, do you have any idea what should I do? If not, then how to modify the settings in Rufus so that the problem goes away? My current settings are:

[image]

My computer is Acer Aspire E 15 E5-575G-57X6, if this is of any help.

Thank you in advance!

Stan

Log
---
```
Rufus x86 v3.5.1497
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Using autorun.inf label for drive G: 'Ubuntu 18.04.2 LTS amd64'
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {7F42A205-5E79-4A83-B7EB-A2DAD2B90E3B}
Max parts: 128, Start Offset: 17408, Usable = 15676176896 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {52BCF857-062D-4C1E-A3A5-B2CC05F8C7E3}
  Size: 14.6 GB (15674061312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {EF019DCF-DE79-43E1-A720-1ADAA9F8467C}
  Size: 512 KB (524288 bytes)
  Start Sector: 30615449, Attributes: 0x0000000000000000
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Using autorun.inf label for drive G: 'Ubuntu 18.04.2 LTS amd64'
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {7F42A205-5E79-4A83-B7EB-A2DAD2B90E3B}
Max parts: 128, Start Offset: 17408, Usable = 15676176896 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {52BCF857-062D-4C1E-A3A5-B2CC05F8C7E3}
  Size: 14.6 GB (15674061312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {EF019DCF-DE79-43E1-A720-1ADAA9F8467C}
  Size: 512 KB (524288 bytes)
  Start Sector: 30615449, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04.2 LTS amd64'
  Size: 1.9 Гб (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-18.04.2-desktop-amd64 (1).iso (1.9 Гб)
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Using autorun.inf label for drive G: 'Ubuntu 18.04.2 LTS amd64'
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {7F42A205-5E79-4A83-B7EB-A2DAD2B90E3B}
Max parts: 128, Start Offset: 17408, Usable = 15676176896 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {52BCF857-062D-4C1E-A3A5-B2CC05F8C7E3}
  Size: 14.6 GB (15674061312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {EF019DCF-DE79-43E1-A720-1ADAA9F8467C}
  Size: 512 KB (524288 bytes)
  Start Sector: 30615449, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 14.6 Гб)
● Partition 2 (offset: 15675109888, size: 512 Кб)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 119599 (1 block = 128 Кб)
Bad Blocks: Using offset 12111 for fake device check
Bad Blocks: Writing test pattern 0x00
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Using Rufus protective MBR
Found volume GUID \\?\Volume{497cf1f6-9ffc-11e9-a56e-5800e39d6040}\
\\?\Volume{497cf1f6-9ffc-11e9-a56e-5800e39d6040}\ was already mounted as G:\
Disabling file indexing...
Successfully remounted \\?\Volume{497cf1f6-9ffc-11e9-a56e-5800e39d6040}\ on G:
Finalizing, please wait...
Created: G:\autorun.inf
Created: G:\autorun.ico
Successfully remounted \\?\Volume{497cf1f6-9ffc-11e9-a56e-5800e39d6040}\ on G:

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Using autorun.inf label for drive G: 'Ubuntu 18.04.2 LTS amd64'
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {A07709D7-060C-4DDF-831A-2B83170850F9}
Max parts: 128, Start Offset: 17408, Usable = 15676176896 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {C32247B1-FD1F-43B4-BCE7-3D78323AF8DD}
  Size: 14.6 GB (15674061312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {7E3E07B9-CCCE-4EF6-A3A2-A05140883476}
  Size: 512 KB (524288 bytes)
  Start Sector: 30615449, Attributes: 0x0000000000000000
```

pbatard, 2019-07-06:
> I keep getting the following error when trying to boot from a usb stick with an Ubuntu image

The log shows that you __didn't__ write an Ubuntu image. Instead you created a blank UEFI:NTFS media, which is not what you want. This is also confirmed by your screenshot.

You __MUST__ select an Ubuntu ISO file if you want to create a bootable Ubuntu media. Rufus does not provide that ISO file, you need to download it first and then select it in Rufus.

All your other issue derive from the fact that you created a blank UEFI:NTFS media. For the record, the `\efi\boot\bootx64.efi` UEFI:NTFS complains about is __NOT__ the one you find on the second non-NTFS partition (which is the actual UEFI:NTFS bootloader and therefore is obviously always present) but the one that should reside on the NTFS partition itself, which, if you created a blan UEFI:NTFS drive, you need to add yourself.

At any rate, I would encourage you to watch a youtube tutorial on how to install Ubuntu with rufus, or find a more detailed guide, as I think all your problem derive from a lack of understanding of what you need to provide before you can create a bootable drive in Rufus. Google is your friend!

stan-tikhonov, 2019-07-06:
Thanks a lot!

The issue is solved.

closed 2019-07-06 by stan-tikhonov