pbatard/rufus #1340: How to create dual-compatible USB flash drive for UEFI AND(!) Legacy BIOS?

bsto, 2019-07-11:
﻿I have two bootable ISO files for the well known software tool Aomei Backupper.
One ISO file is created for Legacy-BIOS-based computers the other ISO file for UEFI-based computers.

I can use Rufus v3.5p to put one of them onto an USB flash drive and boot successfully on a corresponding computer.
ok so far.

The disadvantage is that I need TWO USB flash drives: One for UEFI-based computer and another for Legacy-BIOS computers.

What I want to achieve is to put both variants onto ONE USB flash drive.

Therefore I dragged the ISO file with the UEFI bootloader onto Rufus v3.5p window, then hit ALT + E.
And yes, now the drop down "Target system" shows "BIOS or UEFI".

So I guess this must run on both computers: UEFI and/OR Legacy-BIOS.

However when I boot on the/an Legacy BIOS based system I get at startop an error:

"Bootmgr is missing.
Press key to restart".

Whats wrong?

How can I create an USB flahs drive for UEFI AND Legacy BIOSes otherwise?

Other values were:

Partition scheme=MBR
File System=Large FAT32
Cluster Size=32 kilobytes

Thank you
Ben

pbatard, 2019-07-11:
Please produce a log. The log is not optional, and, unless you want to see this issue closed without further notice (and even if you think it is pointless, which I can assure you it's not), I __need__ to see the complete log of how you created the flash drive(s).

Also, I think the FAQ, which you are supposed to look at if you follow the check list, is clear about <kbd>Alt</kbd>-<kbd>E</kbd> being for official Microsoft Windows installation. Aomei Backupper is not a Windows installer.

bsto, 2019-07-11:
Thank you for your answer.
However I don't think that this is a bug.
Nothing crashed. No error message was produced during USB flash drive creations.
Its the result during boot of the flash drive.

Your last paragraph about ATL+E  is interesting. It implies that the ALT+E tricks works ONLY for Windows.
Is that correct?

That would mean users can NEVER create a dual-boot (UEFI + Legacy BIOS) USB flash drive for other, non-Windows software.
So I need no further investigation.
Sorry about this newbie question.

pbatard, 2019-07-11:
If you can't provide what I ask for and follow the rules, and instead are trying to second-guess what I need to see, sorry but I can't help you. You may think providing a log is pointless, but I'm afraid that it is not.

Closed on account of user repeatedly refusing to provide the data that is asked of them.

closed 2019-07-11 by pbatard