pbatard/rufus #1343: Rufus notes about Hebrew translation requiring an update

ItielMaN, 2019-07-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I ran the new Rufus 3.6.1551 portable version (locale being Hebrew), opened the Log window and I saw that one of the lines is stating that:
```
NOTE: The Hebrew (עברית) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
```
AFAICT I haven't missed any update to Rufus's translation, and to my knowledge Hebrew should be 100%.
What am I missing? :-(

Log
---
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x040D (he-IL)
Will use default UI locale 0x040D
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'he-IL'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
NOTE: The Hebrew (עברית) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
0 devices found
0 devices found
0 devices found
```

ItielMaN, 2019-07-20:
Oh, just now I've noticed it also says to contact you via email. Sorry!

closed 2019-07-20 by ItielMaN

pbatard, 2019-07-20:
I'm not able to replicate this issue.

Are you absolutely sure you don't have a `rufus.loc` in the same directory as the one where you have the Rufus executable.

One way I'd expect to generate that message is if you manually edited a local `rufus.loc` and set `l "en-US" "English` to `v 3.6`. Be mindful that the version used by the loc file is completely independent from the version of the application, and is set to the version of the application where the last translation round occurred. And because there was no translation round for 3.6, and the last one was for 3.5, then the English version is set to `v 3.5` even in Rufus 3.6.

ItielMaN, 2019-07-20:
> Are you absolutely sure you don't have a `rufus.loc` in the same directory as the one where you have the Rufus executable.

Well, that's what I did wrong :)
In the Downloads folder I had old Rufus files, one of which is rufus.loc.

Sorry for the noise!

pbatard, 2019-07-20:
No worries. Glad it's sorted out now. 😉