pbatard/rufus #1344: checksums for rufus releases

snow8gaia, 2019-07-21:
<details>

<summary>irrelevant </summary>

<p>
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

</p>
</details>

---

**no checksums(or hashes) on ->**
https://rufus.ie/ & https://github.com/pbatard/rufus/releases

---

 example file  https://cdimage.debian.org/debian-cd/current-live/i386/iso-hybrid/SHA256SUMS ->

```
6b8a90cc6efcd84c6eb4d357e98d198e53fe633201a18a8ab35389e8c6afc476  debian-live-10.0.0-i386-standard.packages
28d4ae6219c0d0e390796d21b21215925fb1cdc0f27c51a5cadcb8f63f68bfc5  debian-live-10.0.0-i386-xfce.contents
49c8d18314d6f91be5bef61dd5f18ee9230d9bdef0a7c8be6887e29050f7a578  debian-live-10.0.0-i386-xfce.iso
7f9d6cf7ae5460736d10e1dfa15017d34fb04b7554033c71ab0c2360de88b6c1  debian-live-10.0.0-i386-xfce.log
3314743bfd003bfb612333c8aad5992c81dd9ef6605196934112a7be362ce7ee  debian-live-10.0.0-i386-xfce.packages
```

---

**example projects->**

good:
1 for all; correct format:
https://cdimage.debian.org/debian-cd/current-live/i386/iso-hybrid/

1 for 1; wrong format:
https://github.com/VSCodium/vscodium/releases

bad:
https://github.com/PowerShell/PowerShell/releases

---

**suggestion 1:  SHA256 or better.**

pbatard, 2019-07-21:
The executable is digitally signed, __therefore__ the provision of checksum is unnecessary as Windows __automatically__ validates the checksum as part of the digital signature check.

Checksums make sense for binaries that aren't digitally signed, such as ISOs.
For signed binary executables, they are redundant, therefore there is no point in providing them.

closed 2019-07-21 by pbatard

snow8gaia, 2019-07-21:
@pbatard, thanks for the checksums.😊