pbatard/rufus #1346: v3.6.1551 No devices being displayed.

Deneteus, 2019-07-26:
Issue: No devices being displayed
Version: Rufus v3.6.1551
OS: Windows x64

I was about to install Raspian on a 32GB SDCARD and realized Rufus did not recognize the device, so I switched devices and the drive comes up but Rufus still doesn't recognize the device. I even checked the setting for usb drives and ran the application as administrator. 

[image]

[image]

Log:

_Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'ST2000DX001-1NS164' => Eliminated
Found USB device 'Generic STORAGE DEVICE USB Device' (8564:4000) [ID]
Device eliminated because it appears to contain no media
Found USB device 'Generic STORAGE DEVICE USB Device' (8564:4000) [ID]
Device eliminated because it appears to contain no media
Found non-USB removable device 'ST500LM012 HN-M500MBB' => Eliminated
Found USB device 'Generic STORAGE DEVICE USB Device' (8564:4000) [ID]
Device eliminated because it appears to contain no media
0 devices found
Found non-USB removable device 'ST2000DX001-1NS164' => Eliminated
Found USB device 'Generic STORAGE DEVICE USB Device' (8564:4000) [ID]
Device eliminated because it appears to contain no media
Found USB device 'Generic STORAGE DEVICE USB Device' (8564:4000) [ID]
Device eliminated because it appears to contain no media
Found non-USB removable device 'ST500LM012 HN-M500MBB' => Eliminated
Found USB device 'Generic STORAGE DEVICE USB Device' (8564:4000) [ID]
Device eliminated because it appears to contain no media
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.12/Fido.ps1.lzma
File length: 7.3 KB
Successfully downloaded 'Fido.ps1.lzma'
Signature is valid ✓
Please Wait...
Exited download script with code: 100
Scanning image...
ISO analysis:
  'C:\Users\denet\Desktop\Raspian\2019-07-10-raspbian-buster-full.img' doesn't look like an ISO image
Disk image analysis:
  Image has a Zeroed Master Boot Record
  Image is a bootable disk image
Using image: 2019-07-10-raspbian-buster-full.img (6.0 GB)_

pbatard, 2019-07-26:
Please press <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus to enable enumeration debug and send the log you get (which should be a lot more verbose).

Also, please properly read the guidelines next time and don't remove the checklist items you don't like. For instance your log is truncated, with the first few lines missing, so I have no idea what version of Windows you are running, and you clearly didn't read the part that told you to copy/paste the log into the triple backquotes. One of the checklist item is actually to make sure that you copy/paste the full log, so it's pretty annoying to have users just remove the precise part I want them to pay attention to. It makes it more difficult for developers to help you!

pbatard, 2019-07-28:
Any update here? If not I will have to close this issue.

Deneteus, 2019-07-28:
I found that the issue was happening because of Windows Update not installing an update on my desktop. The update itself was not completing because any USB disk connected was causing it to fail. I only found out by looking through all of the update issues for Windows x64 Update 1903. Now that the update completed I can see all the drives.

pbatard, 2019-07-28:
Okay then I will close this issue.

closed 2019-07-28 by pbatard