pbatard/rufus #1347: Unusable device

painteru, 2019-07-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
As I used Rufus for the 1st time, I wanted to make a bootable win10 installation to finally switch from Linux to Windows. So, I selected the find? bad blocks option, too. 
I saw it takes too long, imho, for a 16G pen so I decided to cancel the process EVEN that I have read the warning it MAY cause getting an unusable device. That word MAY made me think this is something like Safely remove device. 
Well, it does result in an unusable flash pen. Kingston dtse9 16gb. I tried to get it usable again by diskpart and then by other several 3rd party apps. 

My device is not shown in my computer and when I try to format in Disk Management or diskpart it tells me no media is present.
I also tried to get it on in Linux using gparted and hdparm wirh no results.

Is there any way to repair that flash pen? Should I try to flash its firmware like how I saw suggested on some webpages? Thanks!

Log
---
```
<FULL LOG>
```

pbatard, 2019-07-28:
Your device reports `No media` which means that the flash memory is probably dead as per: https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media

> Is there any way to repair that flash pen?

Not if the flash memory has become defective, which it eventually does for __all__ flash memory based device. You may try to fool the device into thinking that the flash memory is okay, which is what people who advocate flashing the device firmware as a "solution" seem to swear by, because it resets the dead flash cell count that the firmware maintains to zero, but I'm afraid that, if you have dead memory cells, it's not by pretending that these dead cells don't exist that you are going to "fix" anything. On the contrary you are going to make the problem worse, because now the flash drive will be trying to use these dead cells as if they were working.

At any rate, the software for re-flashing the firmware of a flash drive is super specific and very rarely provided by the software manufacturer (because, once again, re-flashing the firmware will never not "cure" the underlying problem of dead flash memory). So I can't help you with that.

painteru, 2019-07-28:
Thank you very much for your fast reply!
The device failed because of my dumb action PLUS of some 7 (or maybe more) bad blocks I saw there were on the pen until I decided to stop the procedure.
Fortunately, a 16GB flash is affordable these days and also, I didn't have any data on that pen.
I've already bought a new flashpen and made it bootable using Rufus. This reply is written from within the new Win10 installation.
Thanks for the great app!

pbatard, 2019-07-28:
Okay. The fact that you got bad blocks is clear confirmation that your drive was about to fail or in the process of failing. There's no reversing that.

Since you managed to get yourself sorted out with the different flash drive, I will close this issue.

closed 2019-07-28 by pbatard