pbatard/rufus #1350: Rufus option (Old BIOSes)

malachyallen, 2019-07-29:
I've been using Rufus for a while now to burn Linux ISOs to USBs. One of my laptops is an old Samsung (released in 2005 iirc). I used Rufus to burn an Arch Linux ISO to a USB but didn't check the "Add fixes for old BIOSes (extra partition, align, etc.)" option because I'm not sure what year you count new BIOSes being made (I'm not sure how else to write that). What I mean is "Does 19XX-200X count as old BIOSes and 200X-present count as new ones?".
Hopefully you understand what I mean and can clear this up for me!

pbatard, 2019-07-29:
If your machine was manufactured after 1995, you shouldn't need those fixes. At any rate, are you actually __observing__ an issue if you don't use the option?

You really shouldn't check that box unless you tried unchecked and ran into an issue, and are actually using a pre-2000 machine.

pbatard, 2019-08-01:
Unless you think there's something more you want to know, I will close this issue.

closed 2019-08-01 by pbatard