pbatard/rufus #1351: Cannot mount GUID volume

MrCoconut, 2019-07-30:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered. _And there is no specific question regarding "GUID" things._
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log. _Yeah, but issue #476 ended with "I think I have an idea what's going on. Will add a possible fix for this with the 2.3 release." And I'm using Rufus 3.6_.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below. _Yeah, please see it._
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive. _Rufus run successfully on my USB drive with the same setting and Windows ISO, but it's far to slow to run a Windows To Go. The performance was unacceptable._
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi, all. I continuously experienced error message "cannot mount GUID volume" everytime when trying to construct a Windows To Go drive on my SSD. It appers right after status "writing MBR". I successfully made a WTG on my USB drive prior to this so I don't think the ISO is to be blame and I run a bad block check but the SSD was alright. I tried several times with the SSD repluged into another port, Windows virtual machine restarted and the error message never disappers.

Log
---
```
Rufus x86 v3.6.1551
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Found USB 3.0 device 'ASMT 2115 USB Device' (174C:1153)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found non-USB removable device 'VMware Virtual SATA Hard Drive' => Eliminated
0 devices found
Found USB 3.0 device 'ASMT 2115 USB Device' (174C:1153)
Found non-USB removable device 'VMware Virtual SATA Hard Drive' => Eliminated
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {3E2828D5-F849-456A-BC8A-ADE2D0B4390F}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI System Partition'
  ID: {3396DF70-B553-4A11-833E-F5324060A9D5}
  Size: 300 MB (314592768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft Reserved Partition'
  ID: {346F1B95-4C19-4CBD-A48A-934855E87109}
  Size: 128 MB (134217728 bytes)
  Start Sector: 616487, Attributes: 0x0000000000000000
Partition 3:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {02CCB867-4F44-4934-99B5-4C3EB82AF735}
  Size: 465.3 GB (499657986048 bytes)
  Start Sector: 878631, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_ZH-CN_DV9'
  Size: 4.6 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1903_V1_Chinese(Simplified)_x64.iso (4.6 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\yezid\AppData\Local\Temp\Ruf1B2B.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 专业版' (Build: 18362, Index 4) for Windows To Go

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 300 MB)
● Partition 2 (offset: 315641344, size: 128 MB)
● Partition 3 (offset: 449859072, size: 465.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
● Creating EFI System Partition Partition (offset: 1048576, size: 300 MB)
● Creating Microsoft Reserved Partition Partition (offset: 315641344, size: 128 MB)
● Creating Main Data Partition (offset: 449859072, size: 465.3 GB)
Waiting for logical drive to reappear...
Closed Windows format prompt
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Found volume \\?\GLOBALROOT\Device\HarddiskVolume9\
Could not remount \\?\GLOBALROOT\Device\HarddiskVolume9\ as E: [0x00000012] 没有更多文件。
\\?\Volume{7df4b5a6-0659-4163-b585-bc5c71d47127}\ is already mounted as E:
Re-mounted volume as E: after error

Found USB 3.0 device 'ASMT 2115 USB Device' (174C:1153)
Found non-USB removable device 'VMware Virtual SATA Hard Drive' => Eliminated
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {81E6E4BB-BDDE-4DBE-89C9-8117F5FB92B2}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI System Partition'
  ID: {7354A864-CC76-427C-A694-3F87146E6AD4}
  Size: 300 MB (314592768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft Reserved Partition'
  ID: {52D00507-CFBE-41B5-B8B9-B1164FEB3754}
  Size: 128 MB (134217728 bytes)
  Start Sector: 616487, Attributes: 0x0000000000000000
Partition 3:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {7DF4B5A6-0659-4163-B585-BC5C71D47127}
  Size: 465.3 GB (499657986048 bytes)
  Start Sector: 878631, Attributes: 0x0000000000000000
```

MrCoconut, 2019-07-30:
Translation：没有更多文件。-> No more files.

DaKoalP, 2019-07-30:
Indeed, same issue. 
Previous version (3.5) works fine. 
I also noticed that it says "Improve(?) Windows To Go support by following Microsoft's recommended partition order" in the changelog.

pbatard, 2019-07-30:
Please run more tests with 3.5. Be mindful that a single test with 3.5 is not enough to determine if there was a regression.

Also, did any of you enable VDS (<kbd>Alt</kbd>-<kbd>V</kbd>) by any chance? If so please disable it an try again, as this feature is __very__ experimental.

pbatard, 2019-07-30:
Also, I see that you are getting:
```
Timeout while waiting for logical drive
Logical drive was not found!
```
This is not usual and indicates that Windows is interfering with the normal flow of operation. When that happens, you may want to cycle the USB port with <kbd>Alt</kbd>-<kbd>C</kbd> as this issue tends to indicate that Windows is trying to use an old cached partition table, instead of the actual one, and the only way I have found to force Windows to forget the old table is through port cycling (I have tried many, many API calls that Microsoft say are supposed to refresh the layout, which is actually part of the reason why there is a new VDS mode in Rufus, but __none__ actually work)...

Heinzable, 2019-07-30:
I've tested v3.3 - 3.5, all of which worked immediately. Tried it with v3.6 multiple times (and the usual stuff such as rebooting, trying different USB ports, formating the drive in Disk Management, even deleting the partition table, low level format) but it never got past that point. 

Always said 
`Timeout while waiting for logical drive
Logical drive was not found!`

And no, I didn't hit ALT-V.
Can you elaborate on the ALT-C fix? When should it be pressed?

Heinzable, 2019-07-30:
Just tried it with hitting ALT-C before pressing "Start". Worked twice in a row.
Thank you!

frh, 2019-07-30:
I'm experiencing what appears to be the same issue, although ALT-C doesn't fix it. My log also has the same "Timeout while waiting for logical drive" and "Logical drive was not found!" messages. Using Rufus 3.5 instead of 3.6 works. This is a SATA SSD in a Ugreen USB enclosure. Rufus is running in a Windows 10 VM on KVM using USB passthrough for the target disk. Let me know if I can help by testing in any way.

pbatard, 2019-07-30:
I see a lot of people using VM's there... Please be mindful that I can only support real hardware, as there is no telling how virtualization can interfere with either of Rufus or Windows behaviour. For instance I strongly suspect that the reason <kbd>Alt</kbd>-<kbd>C</kbd> doesn't work on a VM is because no actual USB port reset takes place (it's probably filtered by the VM application). In that case, please try to physically unplug and replug the drive.

frh, 2019-07-30:
I can understand if you reckon that the VM aspect will make this problem out of scope. I very much appreciate Rufus nonetheless.
I tried physically replugging the drive, but it made no difference. It consistently works every time with 3.5 and fails with 3.6 with everything I've tried so far on my setup.

MrCoconut, 2019-07-31:
Sincerely thank everyone in this thread. It turned out that rolling back to `v3.5` was the solution for me. I did not accidentally turn on VHS, and `Alt + C` did not fix the problem. `_(:з」∠)_`
I know that running such a system-mode intensive task in a VMWare virtual machine is not a good idea, but that's the only viable option for me anyway... Thank you @pbatard for your hardwork and quick response.

ckrzen, 2019-08-01:
Just wanted to add that I hit this just now and the only solution, after trying all mentioned methods above, was to revert to 3.5.

Let me know if I can add any debug logs for things not yet tried.

pbatard, 2019-08-01:
(Full) logs are __ALWAYS__ welcome. I'm afraid a "me too" without a log doesn't tell me much. Also please indicate if you are running Rufus on real hardware or on a VM.

frh, 2019-08-01:
Here's the log from my Windows To Go attempt in 3.6, running in a KVM VM with USB passthrough.
[rufus.log](https://github.com/pbatard/rufus/files/3458946/rufus.log)

frh, 2019-08-01:
Out of curiosity I tried the same thing with a virtual USB disk in the VM, and the result was the same. I don't know if this will be useful in any way, but here's the log from that:
[rufus.log](https://github.com/pbatard/rufus/files/3459037/rufus.log)

I also tried with MBR instead of GPT, and when I do that it seems to work.

ckrzen, 2019-08-02:
> (Full) logs are **ALWAYS** welcome. I'm afraid a "me too" without a log doesn't tell me much. Also please indicate if you are running Rufus on real hardware or on a VM.

Running 3.6 on bare-metal(HDD) amd attaching full-log with screenshot.

[rufus-3.6_fail.log](https://github.com/pbatard/rufus/files/3461807/rufus-3.6_fail.log)

[image]

ckrzen, 2019-08-02:
I can also confirm that this seem to only happen when using "GPT" partitioning for the WTG target. Selecting "MBR" allows the process to progress past the GPT fail point.

pbatard, 2019-08-02:
Thanks for the report. Did cycling the port (<kbd>Alt</kbd>-<kbd>C</kbd>) help in any way at all?

ckrzen, 2019-08-02:
> Thanks for the report. Did cycling the port (Alt-C) help in any way at
> all?

No. And I did try that _multiple_ times.

pbatard, 2019-08-02:
Okay. I'm going to reopen this issue since it appears to affect a lot of people. Will see what I can do to try to get this fixed for 3.7.

ForestJohnson, 2019-08-10:
Thank you for this wonderful software!!  This happened to me on windows 10 update 1903 (no virtualization) with rufus 3.6.1551, Alt-C did not make any difference, 

```
WIM extraction method(s) supported: wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.esd:[0] (API)
Extracting: C:\Users\forest\AppData\Local\Temp\Ruf332.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.esd
Will use 'Windows 10 Home' (Build: 18362, Index 1) for Windows To Go

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 300 MB)
● Partition 2 (offset: 315641344, size: 128 MB)
● Partition 3 (offset: 449859072, size: 931.1 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
● Creating EFI System Partition Partition (offset: 1048576, size: 300 MB)
● Creating Microsoft Reserved Partition Partition (offset: 315641344, size: 128 MB)
● Creating Main Data Partition (offset: 449859072, size: 931.1 GB)
Waiting for logical drive to reappear...
Closed Windows format prompt
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Found volume \\?\GLOBALROOT\Device\HarddiskVolume40\
Could not remount \\?\GLOBALROOT\Device\HarddiskVolume40\ as E: [0x00000012] There are no more files.
\\?\Volume{5e18a454-f877-4c31-a96c-89a93ac8914a}\ is already mounted as E:
Re-mounted volume as E: after error

Found USB 3.0 device 'WD My Passport 0748 USB Device' (1058:0748)
1 device found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121597, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {224D27E3-98AB-4CA4-AD5E-B19C0A3E3429}
Max parts: 128, Start Offset: 17408, Usable = 1000170551808 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI System Partition'
  ID: {2A690BAA-16CE-4CC2-9CFF-DDD29E261E5D}
  Size: 300 MB (314592768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft Reserved Partition'
  ID: {E4D1069D-FBB0-4234-8749-E0070F2971E2}
  Size: 128 MB (134217728 bytes)
  Start Sector: 616487, Attributes: 0x0000000000000000
Partition 3:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {5E18A454-F877-4C31-A96C-89A93AC8914A}
  Size: 931.1 GB (999720710144 bytes)
  Start Sector: 878631, Attributes: 0x0000000000000000

```

pbatard, 2019-08-10:
Yeah, the problem (which can be verified with `diskpart` as well as the Disk Manager UI) is that Windows sometimes seems to "forget" to allocate a proper volume ID to all the partitions after you repartition a drive.

In our case, we'll get the ESP (partition 1) and MSR (partition 2) with an effective accessible volume, but not partition 3, which is the main NTFS partition where we apply the WIM image.

In other words, instead of getting the partitions enumerated like this:
```
GOT \\?\Volume{7ebca266-b883-11e9-9a1e-88d7f6e01c0c}\ @1048576
GOT \\?\Volume{7ebca267-b883-11e9-9a1e-88d7f6e01c0c}\ @315641344
GOT \\?\Volume{7ebca268-b883-11e9-9a1e-88d7f6e01c0c}\ @449859072
```
where the value behind the `@` represents the offset of each partition, you only ever get this:
```
GOT \\?\Volume{7ebca266-b883-11e9-9a1e-88d7f6e01c0c}\ @1048576
GOT \\?\Volume{7ebca267-b883-11e9-9a1e-88d7f6e01c0c}\ @315641344
```
where Windows "forgets" to assign a volume GUID to the 3rd partition.

This is the reason why I advise a USB port reset, as a __proper__ reset should force Windows to rescan the partition layout and assign an ID to the 3rd partition. However, it appears that, in their great wisdom, Microsoft have neutered USB port reset in Windows 1903, because it doesn't seem to produce the usual disconnection one sees with previous versions of Windows. And none of the APIs I tried to force a partition layout refresh seem to help either (and, believe me, I've tried all of the ones that Microsoft makes available). So my recommendation now is to yank out the USB drive and then plug it back to see if that helps (which may or may not work, since Windows appears to do its hardest to cache a partition layout, instead of properly refreshing it when it should).

Now, those with a keen eye might notice that the first longword in the GUID of each partition is incremented (`7ebca266`, `7ebca267`, `7ebca268`) whereas the rest of the GUID of the same, so you might be tempted to think that, even if Windows does not report the GUID, Rufus could still "guess" it, but I'm afraid it's the other way round: The fact that Windows doesn't report a volume GUID for the 3rd partition is because it can't be accessed, __at all__. So even if we can logically guess the Volume ID to use, we still won't be able to do anything with the partition, since none of the Windows APIs will allow us to access it (and I'm afraid that, no, using a different ID to access the volume, such as `\\?\GLOBALROOT\Device\HarddiskVolume##`, which is supposed to be equivalent to a GUID volume, doesn't help either).

So, provided the issue is that Windows only manages to sort its volume GUIDs for the first 2 partitions, I think I might revert to what I was doing in 3.5, which is have the partitions in MSR, Data and ESP order, rather than ESP, MSR, Data as was changed in 3.6. This might make things less compatible with some __poorly designed__ UEFI systems (that aren't specs compliant if they expect the ESP as the first partition), but hopefully it will sort things out until Microsoft really gets its shit together with regards to their __ABSOLUTELY ABYSMAL__ handling of multiple partitions...

xemex, 2019-08-10:
Thanks for the insight in the root cause.
I have just tried unplugging the device and observed no difference.

Note: I am using build Windows 18362, not 1903 and yet having same problems (on a bare metal).
MBR partitioning works just fine.

```
Rufus x86 v3.6.1551
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'TOSHIBA MQ01ABF050 USB Device' (18A5:0408)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'TOSHIBA MQ01ABF050 USB Device' (18A5:0408)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {52E14D01-14E5-4D6A-AEBF-A6EED68C7ADD}
Max parts: 128, Start Offset: 17408, Usable = 500107825664 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI System Partition'
  ID: {0E2B4BB4-B158-4F44-81AD-9F04B548D9D4}
  Size: 300 MB (314592768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft Reserved Partition'
  ID: {40750DBE-9C9B-4264-BEFB-3BDE5B1E51D5}
  Size: 128 MB (134217728 bytes)
  Start Sector: 616487, Attributes: 0x0000000000000000
Partition 3:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {E8479485-C77B-47CE-8321-B253B4999D89}
  Size: 465.3 GB (499657984000 bytes)
  Start Sector: 878631, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CENA_X64FREE_EN-US_DV9'
  Size: 4.1 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 18362.30.190401-1528.19h1_release_svc_refresh_CLIENTENTERPRISEEVAL_OEMRET_x64FRE_en-us.iso (4.1 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\Przemek\AppData\Local\Temp\Ruf3DF2.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 Enterprise Evaluation' (Build: 18362, Index 1) for Windows To Go

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive4':
● Partition 1 (offset: 1048576, size: 300 MB)
● Partition 2 (offset: 315641344, size: 128 MB)
● Partition 3 (offset: 449859072, size: 465.3 GB)
Opened \\.\PhysicalDrive4 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
● Creating EFI System Partition Partition (offset: 1048576, size: 300 MB)
● Creating Microsoft Reserved Partition Partition (offset: 315641344, size: 128 MB)
● Creating Main Data Partition (offset: 449859072, size: 465.3 GB)
Waiting for logical drive to reappear...
Closed Windows format prompt
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Found volume \\?\GLOBALROOT\Device\HarddiskVolume45\
Could not remount \\?\GLOBALROOT\Device\HarddiskVolume45\ as G: [0x00000012] There are no more files.
\\?\Volume{159487f0-4445-4b0f-9187-69112318783d}\ is already mounted as G:
Re-mounted volume as G: after error

Found USB 2.0 device 'TOSHIBA MQ01ABF050 USB Device' (18A5:0408)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {EB536DEA-6E59-408B-9718-B709773E2DF9}
Max parts: 128, Start Offset: 17408, Usable = 500107825664 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI System Partition'
  ID: {4A33CC25-FA18-46E9-8D50-8E57F0FB956E}
  Size: 300 MB (314592768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft Reserved Partition'
  ID: {F34C1D5D-5AC5-4833-9187-C82F7EDEA28D}
  Size: 128 MB (134217728 bytes)
  Start Sector: 616487, Attributes: 0x0000000000000000
Partition 3:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {159487F0-4445-4B0F-9187-69112318783D}
  Size: 465.3 GB (499657984000 bytes)
  Start Sector: 878631, Attributes: 0x0000000000000000
```

pbatard, 2019-08-13:
Confirmed that moving the ESP back to being the last partition should make Rufus perform as it did in 3.5. I will push the fix shortly and it will be part of Rufus 3.7 (which I may release earlier than planned as there are now quite a few fixes that would probably be better pushed sooner rather than later).

closed 2019-08-13 by pbatard

pbatard, 2019-08-26:
Note, for those who have been facing this issue, I have just released a BETA for the upcoming Rufus 3.7, which you can download from [HERE](http://rufus.ie/downloads/).

If you feel like testing and confirm that this fixes the issues you were experiencing with 3.6, please do so!

frh, 2019-09-04:
I tested with 3.7 beta in the same environment as before (Windows 10 VM running in KVM with USB passthrough for the target disk). I can confirm that the problem appears to be fixed.

commit 936bd9be (2019-08-13): [core] revert to having the ESP as the last partition

* This is to avoid Microsoft's appalling refresh of the partition layout,
  which can result in partitions not being assigned a volume GUID.
* Mostly reverts a change that was applied in 1c39a80d72ce3f6914925ad18518857063dfa413.
* Also add some more enum output and bail if we can't get a logical drive.
* Closes #1351