pbatard/rufus #1352: Format error: The volume label is invalid.

yoyokenny, 2019-07-31:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Setting a blank label produces an error and the format fails.

Log
---
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
LPG handling disabled, per settings
Localization set to 'en-US'
Found USB 2.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 0, size: 29.5 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (SFD)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Bad label
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 2.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No
```

InstantAli3n, 2019-08-01:
Having same issue here, as well as
[0x00000012] There are no more files.
Also tried two different images, Windows To Go install on official image didn't work, then format failed first time (I deleted the volumes in disk management) then trying a second time in a row without doing anything else succeeded.

Here's the log.
[rufus.log](https://github.com/pbatard/rufus/files/3455603/rufus.log)

pbatard, 2019-08-01:
@InstantAli3n, please see #1351 for the `Timeout while waiting for logical drive` / `Logical drive was not found!` (which are the origin of `Could not remount ... [0x00000012] There are no more files`). I'm afraid this is a different issue from what OP reported (, and __I can only handle a single issue per thread__ (otherwise each issue would devolve into _"I have this problem... and I have also this separate problem... and I have also this separate problem"_ which soon becomes unmanageable).

Therefore, unless you can prove that these two issues are directly related (MB: It's not because they don't seem to be present in 3.5 that they are), this thread will __ONLY__ focus on the `Bad label` error and `[0x0000007E] The specified module could not be found. - Retrying...` when creating the file system. Any other error will have to be handled in a different issue.

This being said, and that applies to both you and OP, what happens if you try <kbd>Alt</kbd>-<kbd>C</kbd> or unplug/replug the drive?

bjpafa, 2019-08-01:
>    - I clicked the 'Log' button or pressed Ctrl-L in Rufus, and

If the stick is to boot from uefi W10 it can be formatted ntfs. At least
with KDE 19.04 neon plasma dev or Mint 19.04 cinnamon... it is booting
correctly wit Rufus 3.5

pbatard, 2019-08-01:
@bjpafa, I'm not sure what you are commenting on.

Are you trying to format a drive __As a Super Floppy Disk__ (like OP does) and encountering a `Bad label` /  `[0x0000007E] The specified module could not be found. - Retrying...`.

Also, I don't understand at all what you mean with the above. You need to indicate what options you used in Rufus, rather than what you created the USB for, which practically tells me nothing, since it doesn't tell me if you partitioned it MBR or GPT.

I appreciate the willingness to help here, but, __unless you provide a log__, that details exactly what you tried to accomplish in Rufus and that shows an explicit `Bad label` error, I'm going to have to ignore your report, because it's only through a log that I can make some progress.

Also, once again, I would really like to hear what happens if you plug/unplug the drive or try <kbd>Alt</kbd>-<kbd>C</kbd> in Rufus to cycle the USB port.

yoyokenny, 2019-08-01:
Sorry for the delay, I tried again, used <kbd>Alt</kbd>-<kbd>C</kbd> (which doesn't work since I'm still on Windows 7) and then proceeded to try all 3 Partition Schemes with the same results.

Happens exactly the same regardless of USB drive and unplugging/replugging.

Following is the full log:
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
LPG handling disabled, per settings
Localization set to 'en-US'
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004F32A5
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.8 GB (4047503360 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004F32A5
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.8 GB (4047503360 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 3.8 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (SFD)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Bad label
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
[... 110 lines cut ...]
Partitioning (SFD)...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Bad label
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 3.8 GB (4048551936 bytes)
  Start Sector: 0, Boot: No
```

pbatard, 2019-08-01:
Try <kbd>Alt</kbd>-<kbd>Z</kbd> then, let it zero the drive for a while then cancel, and retry after unplugging and replugging the drive.

yoyokenny, 2019-08-01:
Did as asked, same results.

Following is the log:
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
LPG handling disabled, per settings
Localization set to 'en-US'
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0054F393
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.8 GB (4047503360 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 3.8 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Zeroing drive...
Cancelling
Re-mounted volume as E: after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 3.8 GB (4048551936 bytes)
  Start Sector: 0, Boot: No
0 devices found
0 devices found
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 3.8 GB (4048551936 bytes)
  Start Sector: 0, Boot: No
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 3.8 GB (4048551936 bytes)
  Start Sector: 0, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 0, size: 3.8 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (SFD)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Bad label
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 3.8 GB (4048551936 bytes)
  Start Sector: 0, Boot: No
```

yoyokenny, 2019-08-01:
Here's another log with both a different drive and port:
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
LPG handling disabled, per settings
Localization set to 'en-US'
0 devices found
0 devices found
Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No
Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No
Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 0, size: 29.5 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (SFD)...
Closing existing volume...
Waiting for logical drive to reappear...
FAT label is mostly underscores. Using '29.5 GB' label instead.
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Bad label
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No
```

pbatard, 2019-08-01:
I know you mentioned that you tried other partition schemes, but why do you seem to insist on partitioning the drive as SFD (Super Floppy Disk)? First of all, it's only available as an advanced option, and I currently know of only one case where one absolutely needs to use SFD rather than MBR or GPT. One.

So can you please explain why you want to use SFD? If not, then please make sure to only test with GPT or MBR partitioning. Also, since according to Windows, the problem is with the label choice, what happens if you try to change the label in Rufus?

yoyokenny, 2019-08-01:
The partitioning doesn't actually matter to me, was just using that option because I could.

Using the default assigned label (for this drive "29.5 GB" without the quotes), the format completes fine. When trying to not have any label set (blank label field), it seems to be treating it as if there are too many characters getting converted to underscores (__________ fails as expected). Any label that doesn't produce too many underscores works fine.

Following is a log where I did exactly that (i.e: "29.5 GB", "", "__________", and "NO_LABEL" respectively and without the quotes):
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
LPG handling disabled, per settings
Localization set to 'en-US'
Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 29.5 GB (31633440768 bytes)
  Start Sector: 0, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 0, size: 29.5 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 29.5 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{8d46c074-b35f-11e9-ac42-90b11c96b61f}\
Successfully remounted \\?\Volume{8d46c074-b35f-11e9-ac42-90b11c96b61f}\ as E:

Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.5 GB (31632392192 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 29.5 GB)
[... 91 lines cut ...]
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{8d46c074-b35f-11e9-ac42-90b11c96b61f}\
Successfully remounted \\?\Volume{8d46c074-b35f-11e9-ac42-90b11c96b61f}\ as E:

Found USB 3.0 device 'USB DISK 3.0 USB Device' (6557:5400)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3845, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.5 GB (31632392192 bytes)
  Start Sector: 2048, Boot: No
```

yoyokenny, 2019-08-01:
Also, whenever it states ```FAT label is mostly underscores. Using '29.5 GB' label instead.``` it isn't actually using that label instead.

pbatard, 2019-08-01:
> Using the default assigned label (for this drive "29.5 GB" without the quotes), the format completes fine. 

Aha, at least we are getting somewhere.

> When trying to not have any label set (blank label field), it seems to be treating it as if there are too many characters getting converted to underscores (__________ fails as expected).

If you are talking about the `FAT label is mostly underscores.` __notice__ (it's just a notice), that's mostly because when we convert labels with international characters to something FAT32 can accept (English only, with a limited set of ASCII), because we convert any character that won't be accepted to underscore, we may end up with a label that's completely pointless (`_____2_____` or something), so it makes a lot more sense to just use a more straightforwards label. And usually, this would be completely transparent to the user, since Windows would use the `autorun.inf` label to display in the UI, except we have a minor __unrelated__ bug where the `autorun.inf` is not created, which will be fixed in 3.7 (here I have to really stress out that it's unrelated, even if both problems seem to apply to labels. We just don't create `autorun.inf` in 3.6 because the checkbox gets unchecked when START is pressed. We didn't change anything when it comes to how the underlying FAT label is applied, which is completely independent of `autorun.inf`).

So that note about the label being underscore is not indicative of an issue. Especially, we very much sanitize the label before we apply it (make sure it's 11 characters or less, with all characters being ones that FAT support, which is the precise reason you are seeing a notice about underscores in the first place, as it is issued during label sanitization), so it should be a matter of Rufus not properly sanitizing the label for FAT32.

As a matter of fact, no matter what I try, I am unable to replicate the issue, whether I use a blank label, one with plenty of underscores and so on... Even when I'm getting the message `FAT label is mostly underscores. Using '##.# GB' label instead.` I never get `Bad label` being reported, so this issue is definitely not as clear cut as your testing makes it look.

One possibility is that it might be a Windows 7 vs Windows 10 behaviour, so I'll try to test on Windows 7 when I have a change.

However, by all accounts, it seems that, in your case, the issue is the result of trying to apply a "weird" label, which therefore I don't expect should affect that many people in the first place. Of course, I'll see what I can do to fix it, but it's not as high a priority as I thought it might be.

Do you have the possibility to test on Windows 10 (or on a different machine) to see if you can replicate the same issue there? Considering that my testing shows that there is an environmental component (in other words it's not programming issue that will consistently produce the same behaviour regardless of your Windows environments), I'm still going to need more details about your testing conditions before I can even explore the possibility of a fix.

pbatard, 2019-08-01:
For the record, for all who wonder, the `Bad label` message occurs [here](https://github.com/pbatard/rufus/blob/master/src/format.c#L162-L165), and is pretty much a report from the black box that is the Windows formatting API (i.e. not something we have much control on).

In other words, when Rufus formats a FAT32 drive, it basically tells Windows (through the `FormatEx()` API call): _"Here's a drive. Can you please format it to FAT32 using label 'XYZ'?"_ and then we wait for Windows to either complete the request or report an error.

In this case `FCC_BAD_LABEL` is what Windows' `FormatEx()` reports when it is unhappy with the label provided. Therefore, if one label works on Windows 10, but not on Windows 7, it would look to me more like a Windows 7 issue than anything, though of course I can add additional sanitization on Windows 7... if I can figure out what Windows 7 is unhappy about.

This is why I will need to get confirmation of:
1. This error consistently occuring when you apply a specific label
2. What is the exact label you use (tip: use single backquotes to enclose the label so that formatting will not be lost even if there are spaces such as `THIS LABEL HAS TWO SPACES AT THE END  `)

yoyokenny, 2019-08-01:
Unfortunately I cannot test it on Windows 10 or another machine. It could be a Windows 7 issue, although the previous version of the program worked fine.

The specific label I'm trying to apply is no label at all. (empty `Volume label` textbox).

It also doesn't seem to be falling back to setting the label as the size of the USB.

yoyokenny, 2019-08-01:
Upon looking through the code, is it possible that the `utf8_to_wchar` function isn't returning the equivalent of an empty string for the label?

pbatard, 2019-08-01:
> is it possible that the utf8_to_wchar function isn't returning the equivalent of an empty string for the label when it's blank?

Nope:

```C
	// Convert the empty string too
	if (str[0] == 0)
		return calloc(1, sizeof(wchar_t));
```

Besides, again, if this was a straight code issue, then I would be able to replicate the problem on Windows 10, which I can't (empty label works just fine fore me. No error whatsoever.

But are you saying that you are seeing the `FAT label is mostly underscores.` notice when you use an empty label too? Because it shouldn't appear then and I'm not seeing it during my testing either.

yoyokenny, 2019-08-01:
That is exactly what I'm saying.
It may be a Windows 7 specific issue.

pbatard, 2019-08-01:
> I'll see if I can replicate the problem in a Windows 10 VM through VirtualBox.

No need. I have now been able to replicate the issue in a Windows 7 VM. And the nice thing is I could test the exact same drive for both Windows 10 and Windows 7 (since the Windows 7 VM was being host by the Windows 10 machine). And of course, both cases were tested using the exact same version of Rufus (3.6 release)

Windows 7 does indeed produce `Bad label` for an empty label. Windows 10 does not. So it is indeed a Windows 7 specific issue (not sure where Windows 8 falls yet).

Now of course, I can also confirm that Rufus 3.5 seems to be fine on Windows 7, which gives me an idea of where to look. But the underlying problem is that Windows 7 is stricter with the kind of volume labels it accepts compared to Windows 10.

Now, IIRC, I think I may have modified the label sanitization process from Rufus between 3.5 and 3.6, so that it would also accommodate Linux `ext2`/`ext3` labels (since you can now format a volume to `ext2`/`ext3` in 3.6), so, if I were to take a guess, I'd say I probably removed some sanitization that might have still been needed for Windows 7. I'll see if I can figure out what that is, and try to fix this for Rufus 3.7.

yoyokenny, 2019-08-01:
Okay I've cancelled my VM.

While I was waiting for it, I started trying every filesystem that can be formatted on my machine.
So far every single one has had some problem with a blank label.

`ext2 / ext3` both fail regardless (Probably safe to say this is a separate [I'm assuming Windows 7] issue with `ext2 / ext3`) No idea if this one is related.

Labels used: blank, `3.8 GB`

Full log:
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 (0x06)
  Size: 3.8 GB (4047503360 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 3.8 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.8 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT)...
Using cluster size: 65536 bytes
Quick format was selected
Creating file system...
Bad label
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01052531
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.8 GB (4047503360 bytes)
[... 290 lines cut ...]
● Creating Main Data Partition (offset: 1048576, size: 3.8 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (ext3)...
247040 possible inodes out of 988160 blocks (block size = 4096)
49408 blocks (5.0%) reserved for the super user
Creating 31 inode sets: [1 marker = 1.0 set(s)]
+++++++++++++++++++++++++++++++
Could not zero inode set at position 988160 (1024 blocks): [0xC0030057] The parameter is incorrect.
Format error: [0xC0030057] The parameter is incorrect.
Re-mounted volume as E: after error

Found USB 2.0 device 'Sony Storage Media USB Device' (054C:0439)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01117A33
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Size: 3.8 GB (4047503360 bytes)
  Start Sector: 2048, Boot: No
```

pbatard, 2019-08-01:
Okay, I should have a fix now.

The root of the issue is that we change the readout of the label from the User Interface from `GetWindowTextW()` to `GetWindowTextU()` between 3.5 and 3.6, the latter being our own wrapper over Microsoft's call to make working with UTF-8 easier (since Microsoft still don't want to provide their APIs in UTF-8, __as they should do__). But we had a small issue in `GetWindowTextU()` because `GetWindowTextW()` returns 0 when it reads an empty string __while still setting the string variable requested to the empty string__, whereas, we only proceeded with the UTF-16 to UTF-8 conversion if `GetWindowTextW()` returned non zero.

Which means that our implementation of `GetWindowTextU()` "returned" whatever already existed in the passed string when the control being queried was empty. 

Now, while this is a bug that obviously needs to be fixed, it still didn't matter that much if you used MSVC to compile the app, regardless of the platform, or even MinGW on Windows 10, as, in these scenarios, the executable would have set the label string to be all zeroes anyway (automatic local variable initialization by the compiler), so, even if the code didn't properly set the variable to the empty string, the compiler already took care of that anyway, so you still got what you expected.

The problem however is, if you are compiling using MinGW (which is how the release version of Rufus is built) and running the app on Windows 7, then local function variables are not initialized to zero, but contain whatever garbage exists in memory. And most of the time, this will contain low level ASCII characters, which we don't eliminate as part of our label sanitization (because, sanitization is meant to apply to elements that people might rightfully try for not knowing better, such as accented characters, lowercase or `*?,;:/\|+=<>[]"` characters, since it's not obvious that are forbidden for FAT32 &mdash; On the other hand, if someone's going crazy with obvious invalid stuff, such as lower ASCII, then you want them to end up with an explicit `Bad label` error), so, even after sanitization, you will end up with something like `'___U_'W_`, with (in this case) 2 low level ASCII chars that of course will make any formatting facility reject the label as invalid. For instance, if you copy the label above, Rufus will produce a bad label on Windows 10, even with the aforementioned fix applied, and I am not planning to change that.

So, I'm just going to point out that the `Bad label` problem for blank fields will be fixed in 3.7 (I have tested that it works), and I will therefore close this issue with the fix.

Oh, and with regards to `ext2`/`ext3` formatting errors, please be mindful that I have slapped a big `[EXPERIMENTAL]` in the release notes about that feature. What you have found is part of the reason why.

commit 307e2f70 (2019-08-01): [core] fix 'Bad label' errors with empty labels

* Only happens on Win7 due to MinGW not automatically initializing variables to zero on that platform.
* Root of the problem was that GetWindowTextU() did not properly handle empty text controls, due to
  GetWindowTextW() returning 0 on empty strings. As a result, the UTF-8 label was not properly set to
  the empty string, but kept whatever data it contained, which is garbage on Windows 7 and resulted in
  an invalid label (even after sanitizing, since we don't sanitize low-level ASCII characters).
* Closes #1352
closed 2019-08-01 by pbatard