pbatard/rufus #1353: Creation of a Windows-To-Go using 10(1903) ISO took over 7-hours on Patriot Rage2 128-GB(USB-3.0)

ckrzen, 2019-08-01:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [o] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [o] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [o] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [o] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [o] Running Rufus on a different computer.
- [o] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Firstly, THANK YOU for Rufus!

I apologize that I did _not_ copy/paste the log as I did not keep it(unless it can be obtained post-reboot? If so, I will gladly update this post).

I have created WTG drives using Rufus for many years, mostly on a Sandisk Cruzer Ultra(32-GB) drive, without a problem. However, I'm using Rufus(latest version as of today), running on the Cruzer Ultra WTG(1809) in a USB-3.0 port, to create a new WTG(1903) on my sparkling new "Patriot Rage2 128-GB USB-3.0" drive in a USB-3.0(backplane port of tower motherboard) and it ran all night, completing successfully, in over 7-hours. Benchmarking this new target drive shows 380-mbps read / 100-mbps write for 1-GB files & 50-mbps write for smaller 10-MB files.

1. Is this normal?

2. Is the size(128-GB) of the target drive a factor?

Log
---
```
<FULL LOG>
```

pbatard, 2019-08-01:
> Is this normal?

No. But it's possible that Windows, for a wide variety of reasons, may have dropped the USB 3.0 speeds to 1.1 speeds (Rufus would have told you that in the log) in which case 7 hours is not far from what I'd expect.

At any rate, Rufus always tries to create the Windows To Go drives as fast as it can, and the main process is so straightforward (we're simply calling on the Windows internal WIM API to apply an image to the drive - Rufus does not even handle the copy/extraction process itself) that if there any slowdowns, then these come from either Windows or the hardware, not Rufus.

> Is the size(128-GB) of the target drive a factor?

Nope. There's no part of the code that cares about the size of your drive, as long as it's large enough for the image.

Note however that creating Windows To Go drives is a slow process by essence, mostly because the WIM API is pretty slow (which has to do with the WIM format itself being a rather poor design when it comes to performance), so if your CPU is slow, or if your source image was on a very slow media, then it doesn't matter how fast your target drive or target connection will be, as the bottleneck happens upstream.

I'm afraid that you are going to find out on your own what caused the slow speed, as I can vouch that Rufus is completely agnostic when it comes to the target (doesn't care if it's a super fast SSD or slow flash drive) and will always pump data out as fast as Windows allows it to do so.

pbatard, 2019-08-01:
I'm going to close the issue then, since there's really not much additional data I can provide.

closed 2019-08-01 by pbatard

ckrzen, 2019-08-03:
I just wanted to confirm that my main problem was with my USB 3.0 memory
stick that has abysmal small file write speeds, despite boasting a 400/MB/s
read & 300/MB/s for sequential large files.

PATRIOT Supersonic Rage 2 USB 3.1, Gen. 1
(USB 3.0) Flash Drive 128GB

*** NOT RECOMMENDED FOR WINDOWS-TO-GO ***

[image: image.png]

pbatard, 2019-08-06:
Many thanks for following up on this. Indeed, this is something I should also have mentioned: There can be a world of difference between sequential and random write speed, and since the WIM extraction process is very reliant on random writes (as opposed to Windows installation drives, where you're copying one large ~4 GB file and only a relatively small amount of small files), if the flash drive's random write speed is limited, then you will observe slow Windows To Go creation.

I am quite amazed at how dramatic that effect appears to be for this specific flash drive though...

For the record, though I haven't tested that much hardware, if you are looking for a flash drive that works well for Windows To Go, I can state that [Mushkin Ventura Ultra drives](http://www.poweredbymushkin.com/Home/index.php/products2/33-ventura-ultra) seem to perform with decent enough speed, both for Windows To Go creation and runtime. The one thing to know about these drives however is that their speed might be reduced if the controller detects flash memory is overheating, which may happen if you are copying a large amount of data for some time. For standard Windows To Go usage, that shouldn't be much of an issue however, and, though my testing has been limited, I don't recall ever seeing a degradation in performance.

ckrzen, 2019-08-06:
Indeed! I switched to using an ADATA SU800 128GB SSD via a USB3.0
adapter(passive) and I got blazing fast WTG creation and runtime.
Previously, I used a Sandisk Cruzer Ultra 32GB usb stick and got decent, if
underwhelming, performance for WTG.

With the price of a 128GB SSD at $25 or less and USB-to-SSD adapters at $15
or less, I'm done with USB sticks!