pbatard/rufus #1354: Is it possible to add a burning option, e.g. Windows 10 for GPT and MBR simultaneously?

RemixPL1994, 2019-08-01:
Is it possible to add a burning option, e.g. Windows 10 for GPT and MBR simultaneously?

It's about a solution like in the Microsoft Creation Tool thanks to which we can record on one flash drive one image of Windows 10 which at the same time supports MBR and GPT.

pbatard, 2019-08-01:
Alas, alas, if only people followed the checklist that has an explicit check mark for

- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

inviting them to have a proper look at the FAQ and only open an issue when they have done so.

So, let me make a few things clear here:
1. It's not because you have a general question that the check list doesn't apply to __you__.
2. GPT and MBR are incompatible. You can't have a drive that is both GPT and MBR (though you can have a GPT drive with an MBR protective record, but then the disk cannot be used an as MBR drive).
3. What you mean, really, is that you want a drive that works for both BIOS and UEFI. Again there, be mindful that BIOS and UEFI are not synonyms of MBR and GPT. For instance, even though UEFI did introduce GPT, an UEFI system is perfectly fine with an MBR drive. As a matter of fact, to achieve what you want (dual BIOS+UEFI bootable Windows drive), you will most certainly want to use an MBR drive.
4. https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode
5. The feature you request is already there. Just press <kbd>Alt</kbd>-<kbd>E</kbd>

So, next time, when you see a check list that explicitly asks you to perform some steps, such as having a look at the FAQ, please try to spend some time doing so, as you may find what you seek is already answered there.

closed 2019-08-01 by pbatard

RemixPL1994, 2019-08-01:
Very Thanks You.