pbatard/rufus #1357: Can't update syslinux

gifthitao, 2019-08-10:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```
Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
NOTE: No Windows Language Pack is installed for Chinese Simplified (简体中文) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Notice: The ISO download feature has been deactivated because the remote download script can not be accessed.
0 devices found
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 972, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x6CA53599
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 2.8 MB (2899968 bytes)
  Start Sector: 23136, Boot: No
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 972, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x6CA53599
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 2.8 MB (2899968 bytes)
  Start Sector: 23136, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Debian 10.0.0 amd64 1'
  Size: 3.6 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: debian-10.0.0-amd64-DVD-1.iso (3.6 GB)
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling

pbatard, 2019-08-10:
The problem is 100% on your side. https://rufus.ie/files/syslinux-6.04/ldlinux.sys __IS__ accessible and is being served by GitHub/AWS (so for it to be down, then AWS would need to be down, which would be major news).

The fact that https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys is not accessible is normal (it's part of the standard fallback mechanism of Rufus to first look if a __custom__ version of ldlinux, in `files/syslinux-6.04/<custom version string>/` is available, just in case, and, if it that file doesn't exist (99.9% of cases) download the mainline `files/syslinux-6.04/ldlinux.sys` version, which does exist and is currently being served without issues.

Possibly, since it looks like you are residing in China, your government is blocking access to Amazon Web Services. Or it could be the firewall at the location you've attempted to use Rufus from.

At any rate, I'm afraid the problem remains 100% on your side, and since there is nothing for me to fix here, I will close this issue.

closed 2019-08-10 by pbatard

gifthitao, 2019-08-10:
Dear Pete:
    Thanks for your email.I am in China ,so I do it again over the hotspot
shield (VPN). It does net work.

Rufus x86 v3.6.1551 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
NOTE: No Windows Language Pack is installed for Chinese Simplified (简体中文)
on this system.
This means that some controls may still be displayed using the system
locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Notice: The ISO download feature has been deactivated because the remote
download script can not be accessed.
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 972, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: NTFS (0x00)
  Size: 7.5 GB (8002732032 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Debian 10.0.0 amd64 1'
  Size: 3.6 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to
file system limitations.
  Because of this, some features from this image may not work...
Using image: debian-10.0.0-amd64-DVD-1.iso (3.6 GB)
Checking for Rufus updates...
Checking release channel...
Unable to send request: The attempt to connect to the server failed.
Updates: Unable to access version data
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling

gifthitao, 2019-08-10:
Dear Pete :
     I downloaded the idlinux.sys, Can  I tell me how to use the rufus with
the idlinux.sys offline? Thx.

pbatard, 2019-08-10:
https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them

gifthitao, 2019-08-10:
Thx. U are so  nice .