pbatard/rufus #1358: Rufus 3.6 does not quick format

yagyaxt1034, 2019-08-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I used Rufus 3.6 to format 2 Ubuntu USB drives. I noticed that Rufus was performing a full format regardless of whether "Quick format" was a selected option or not. I performed this test with both 3.6 and 3.5, but 3.6 would always do a full format. The persistent partition was enabled for both.
Log
---
```
<FULL LOG>
Rufus x86 v3.6.1551
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'WD My Passport 25E1 USB Device' (1058:25E1)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SanDisk Cruzer Force USB Device' (0781:557D)
Found USB 2.0 device 'Generic- SD/MMC/MS PRO USB Device' (0BDA:0153)
Device eliminated because it appears to contain no media
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0184319F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.5 GB (3708278784 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: FAT32 LBA (0x0c)
  Size: 4.0 GB (4294950912 bytes)
  Start Sector: 7244780, Boot: No
Found USB 3.0 device 'WD My Passport 25E1 USB Device' (1058:25E1)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'SanDisk Cruzer Force USB Device' (0781:557D)
Found USB 2.0 device 'Generic- SD/MMC/MS PRO USB Device' (0BDA:0153)
Device eliminated because it appears to contain no media
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
[... 780 lines cut ...]
Found USB 2.0 device 'SanDisk Cruzer Force USB Device' (0781:557D)
Found USB 2.0 device 'Generic- SD/MMC/MS PRO USB Device' (0BDA:0153)
Device eliminated because it appears to contain no media
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1FE727EB-3D1C-49F2-93EC-28244D6226A1}
Max parts: 128, Start Offset: 17408, Usable = 8004270592 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {16BFAEE2-25A7-4FC9-B3E2-C724667B8B8B}
  Size: 2.0 GB (2095220736 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Linux Persistence'
  ID: {41FBECCB-9756-4882-9CB9-A1F39FA9BD1A}
  Size: 5.5 GB (5908008960 bytes)
  Start Sector: 4094276, Attributes: 0x0000000000000000

```

pbatard, 2019-08-12:
What do you call full format? And for which partition?

Unless you have quick format unchecked, both the `ext3` partition and the `FAT32` partitions are being formatted in quick format mode.

I'm certainly not seeing any issue with quick format in 3.6 when using Ubuntu or any other Linux distros for that matter. If quick format is selected both the `ext3` and FAT32 partitions are formatted as fast as the system lets us do it. So I'm inclined to think that, because you see the inodes and journal blocks being created for `ext3`, you believe that slow format is being used, whereas we're __actually__ performing a quick format of the `ext3` partition.

A "slow" format is always very noticeable, especially with a USB 2.0 device such as the one you used. So I doubt you saw a "slow" format but instead you saw `ext3` being formatted, which is a much slower process than FAT32, especially on Windows (for the record, we're using the same library as `mkfs.ext3` for Linux, with very very few changes, and still it's about 2 to 8 times as slow on Windows compared to Linux).

Finally, if slow format is effectively selected, you will see this in your log for the FAT32 partition:
```
Formatting (FAT32)...
Using cluster size: 16384 bytes
Slow format was selected
```

I see no mention of "Slow format was selected" in your log, so I can pretty much __guarantee__ that, despite what you believe, slow format was not active.

pbatard, 2019-08-13:
I'm going to close this issue now, as I believe it boils down to user misunderstanding of what constitutes a "slow" format.

closed 2019-08-13 by pbatard