pbatard/rufus #1360: Rufus 3.6.1551 fails when trying to create a persistent Kubuntu live USB

LordCozyCat, 2019-08-14:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 3.6.1551 on Windows doesn't have any issue's creating a live USB without a persistent partition, but when I try to add persistence to it, it fails with this error, as seen in the full log below: `Could not read device size: [0xC0030002] The system cannot find the file specified.`

Log
---
```
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 0, size: 29.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 13.3 GB)
● Creating Linux Persistence Partition (offset: 14277392896, size: 16.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (ext3)...
Could not read device size: [0xC0030002] The system cannot find the file specified.
Re-mounted volume as F: after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3824, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0073708F
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 13.3 GB (14276344320 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 16.0 GB (17179868160 bytes)
  Start Sector: 27885533, Boot: No
```

yoyokenny, 2019-08-14:
You should probably specify which version of Windows you use (which is supposed to be in the full log for convenience)

I have no idea how it works on Windows 10, but on Windows 7 that is the exact same message when trying to format to any of the ext filesystems.

It is marked as an experimental feature, with known issues.

To quote @pbatard: 

> with regards to ext2/ext3 formatting errors, please be mindful that I have slapped a big [EXPERIMENTAL] in the release notes about that feature. What you have found is part of the reason why.

pbatard, 2019-08-14:
Please provide an untruncated log __or I will close this issue without further notice__. The guidelines are extremely clear on that. If you can't follow the guidelines, you are wasting my time, because I am missing important data that I need to help __YOU__.

closed 2019-08-16 by pbatard

pbatard, 2019-08-26:
You might start by copy/pasting your log. Also you may want to try the latest BETA of Rufus [that was released earlier today](https://github.com/pbatard/rufus/releases).

But you also should be aware that, unless you use a Ubuntu derivative ISO __that was released after 2019.08.01__ (because any releases of Ubuntu and friends before that are using a [buggy version of casper](https://bugs.launchpad.net/ubuntu/+source/casper/+bug/1489855)), it doesn't matter whether you are able to create the a drive with persistence in Rufus or not, as you won't be able to boot it anyway. Oh and the same restrictions as I explained above apply, with Rufus 3.6 not modifying the config files in the first place anyway, and therefore persistence being ignored (since it requires a kernel flag to be added). Only Rufus 3.7 will add that flag automatically for you.

In other words, right now, the __only__ Ubuntu derivative that will actually work out of the box with a persistent partition is the latest Ubuntu [daily-live](http://cdimage.ubuntu.com/daily-live/current/), and only if you are using Rufus 3.7. So if you are using anything like an official ISO release (too old), or Rufus 3.6 (won't add the relevant kernel flag) it won't work anyway, even if you do manage to create the drive.