pbatard/rufus #1362: Tooltip won't always show up in the Volume label

Sopor, 2019-08-17:
﻿

Checklist
---------
- [ x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
It seems that the tooltip for the `Volume label` won't always work. As long as i don't activate the volume label row and put the cursor in it, it will work, but then it is impossible to get the tooltip to show up. I need to restart Rufus or change language to get the tooltip back (maybe there are other ways too).

Rufus 3.6
Windows 10 version 1903

pbatard, 2019-08-17:
I'm afraid I can't replicate the issue.

If I activate the `Volume label`  field and have an active cursor in it, I still see the tooltip without issues.

Tested with Rufus 3.6 on Windows 10 1903.

You may want to elaborate on what you are doing, because, from where I stand, what you describe is far from being the systematic issue you describe, so there must be some additional component you haven't reported that makes the problem appear.

Sopor, 2019-08-17:
This is what i do:
1. Insert USB-stick.
2. Start Rufus.
3. Select ISO file.
4. Press the mouse in the Volume label to set the cursor in that field.
5. Now it won't show any tooltip even if i'm trying to get the cursor away from the volume label field by selecting something else (in the below picture i press on the Device so the cursor is not in the volume label field any more).

[image]

pbatard, 2019-08-17:
I've repeatedly tried to do exactly what you shown above (also testing with or without all the advanced options showing), but I'm afraid I'm still unable to replicate what you see. And of course I tested using the exact same version of Rufus as you do. Whether `Volume label` is active or not, the tooltip for that field always appears for me.

So this looks like something that might be specific to your environment (or maybe an issue with Windows having trouble tracking the cursor position to decide if it should show a tooltip).

Have you tested on more than one system?

Sopor, 2019-08-18:
I tested it on another Windows 10 and it worked, but after playing around with it, it stopped working as on my other Windows 10. If you make the tooltip popup and you move the mouse up instead of down the tooltip will persist and it seems to have something to do with that. If i press in the Volume label field and instead of moving the mouse down i move it up and down several times and after awhile it will stop popping up. Why it is so easy to reproduce on one of my Windows 10, i don't know, but i do see the behavior on both computers even if it isn't so obvious as it is on the first computer. where i found it. If it was like on my second computer i shouldn't have reported it.

pbatard, 2019-08-18:
I don't think I'm gonna bother then. Rufus does not detect mouse movements or has any fine control over tooltips (else, it would be a major pain in the ass to program anything for Windows. Be mindful that most of what the code of an application does is invoke the Windows APIs to delegate something (e.g. "Please associate this tooltip to this control"), rather than control the whole thing (e.g. "I'm going to monitor the mouse all the time and check whether the user is hovering over this or this control. Then, if I detect they are doing just that, I'm going to create a balloon, and then paint some text into it for X number of seconds").

In other words, this looks like a Windows issue to me rather than a Rufus issue, and even if it was a Rufus issue, since it's a minor UI, that can't be replicated all the time, it's just not something I can see myself spending time on.

Closed as NeverFix.

closed 2019-08-18 by pbatard