pbatard/rufus #1363: Attempting to add persistent partition ends in failure

no-good-usernames-left, 2019-08-17:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Attempting to add a persistent partition to the USB drive fails.
Error: 0xC00300002 The system cannot find the file specified. 

Log
---
```
Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 6.5 GB)
● Partition 0 (offset: 6930540544, size: 1024.0 MB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 6.5 GB)
● Creating Linux Persistence Partition (offset: 6930540544, size: 1024.0 MB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (ext3)...
Could not read device size: [0xC0030002] The system cannot find the file specified.
Re-mounted volume as D: after error

Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {CAAA5461-10A1-46EB-A9D2-B6A9131F2BC7}
Max parts: 128, Start Offset: 17408, Usable = 8004270592 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {A0866884-62B8-4F1B-BDC3-17DC5A4E7420}
  Size: 6.5 GB (6929491968 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Linux Persistence'
  ID: {007BE690-1C39-40C5-B72F-B617C318DC9F}
  Size: 1024.0 MB (1073737728 bytes)
  Start Sector: 13536212, Attributes: 0x0000000000000000
Closed Windows format prompt

```
[image]

[image]

pbatard, 2019-08-17:
Yes, this can happen, which is part of the reason why the Persistence feature is tagged __[EXPERIMENTAL]__ (as you will see on https://rufus.ie or the [ChangeLog](https://github.com/pbatard/rufus/blob/master/ChangeLog.txt) or the [3.6 release notes](https://github.com/pbatard/rufus/releases/tag/v3.6)).

It's byproduct of Windows being an ass when it comes to multiple partitions, and I'm not sure I'll ever had a good solution for that.

What might help:
* Trying to cycle the drive with <kbd>Alt</kbd>-<kbd>C</kbd>
* Trying to unplug and replug the drive
* Trying to zero the drive with <kbd>Alt</kbd>-<kbd>Z</kbd> for a while, and then cancelling.
* Trying to create a FreeDOS drive before trying to recreating the Linux drive with persistence.

Also, and this is something that is clearly mentioned in the release notes, persistence will __NOT__ work with Ubuntu at this stage, because all of the official releases of Ubuntu have a major bug with `casper` that effectively prevent persistent partitions from being mounted (and this issue has nothing to do with Rufus).

Only the very [latest daily live](http://cdimage.ubuntu.com/daily-live/current/) has a non-buggy version of `casper`... but that does not mean it will work straight out of the box with Rufus 3.6, as you'll still need to add the `persistent` kernel option to the GRUB or ISOlinux config file for the system to detect it. Having that option being applied automatically is something that will only happen with Rufus 3.7 (i.e. one more reason why the persistence feature was clearly labelled as __[EXPERIMENTAL]__).

In 3.7, provided that Windows doesn't interfere with the creation of the persistent partition, you should find that persistence works out of the box with Debian 10.0 Live or versions of Ubuntu that have been released __after 2019.08.01 (but it probably won't ever work with versions that have been released before that).

pbatard, 2019-08-17:
Also, you __did__ truncate the log. Please don't do that as I have no idea what version of Windows you are using, which could be crucial in trying to help you out. The checklist explicitly states that you should __NOT__ truncate the log in any way...

pbatard, 2019-08-22:
I'm going to close this issue as I don't think there's anything else I can add.

closed 2019-08-22 by pbatard