pbatard/rufus #1367: New Feature Request - a UEFI Shell Boot USB option

hlittler, 2019-08-26:
﻿New Feature Request - a UEFI Shell Boot USB option

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
New Feature Request - a UEFI Shell Boot USB option

Log
---
```
<FULL LOG>
```

pbatard, 2019-08-30:
Okay, I guess it's time to address this one.

I could do it in one word (well two): _NinjaTurtle_, which is the codename I was planning to use for the feature (and which I have now referenced in the [Rufus project page](https://github.com/pbatard/rufus/projects/1)).

Now, why this needs to hide being a code name is because this is a feature that I am not sure that I'm going to actually ever introduce in Rufus, on account that it's way too early to tell if it will really happen or not, and generally I'd rather not have people find out that I once had plan to introduce a feature they might have wanted, and then dropped it. In fact, while _NinjaTurtle_ has always more or less been on my horizon, I wasn't planning to look into it that soon. But now that the cat's out of the bag...

At this stage, you might be wondering, why does this seemingly straightfoward feature need a codename in the first place? Surely it should be a fairly easy affair, like support for DOS.

However, I'm afraid that, if you are expecting to see a new "UEFI Shell" option in the same boot menu as the one where you can find "FreeDOS", you're going to be quite disappointed, because that not at how I'm planning to add the feature at this stage.

The reason for that is that, unlike DOS, there's not just one universal UEFI Shell that you can just add, and that will work for everybody. Instead, you need to add at least 4 different Shell binaries, for x86 32-bit (`ia32`), x86 64-bit (`x64`), ARM 32-bit (`arm`) and ARM 64-bit (`aarch64`), since Rufus strives to support all of these archs when it comes to bootable media. And each of those is around 1 MB in size. Therefore, if Rufus were to embed these in the same manner as we embed the FreeDOS files, even wit compression, the size of the Rufus binary executable would suddenly find itself multiplied byby 2 or 3. Well, not everybody in the world has fast broadband, so doubling or tripling the size of the application (and the overall download times for Rufus) just that a limited subset of people can boot into UEFI Shell is not exactly what I would call nice...

So that pretty much establishes the need for downloading the Shell binaries vs embedding them.

Thus, at this stage, you might think: Well, Rufus could still present the option in the menu and simply download the relevant Shell as needed, like it does for some of the files that GRUB/Syslinux need.

Except, it's still not quite that simple. Because now you are either relying on people having an idea of the specific UEFI target they are going to boot from (`ia32`, `x64`, `arm` or `aarch64`), which I defy most people using a `ia32` UEFI firmware to properly identify, or you have to download all these files at once... which doesn't seem like a big deal except there's yet another hurdle: There actually exists 2 major versions of the UEFI Shell: 1.x and 2.x, and some UEFI platforms can only boot the 1.x version. So not only do you now need to suddenly add 4x more files, but you also need to add 4 additional bootloaders (1 for each archs) that will check whether a platform can run Shell 1.x or 2.x and launch the relevant version. This is starting to become quite a lot of files to download, or a large archive, if you do it through a zip file or something.

And the thing is, at this stage, you are pretty much downloading a complete bootable media archive (i.e. something you can use 7-zip to extract all the files on a media and effectively boot from)... so you might as well turn it into an UEFI bootable ISO that people can download separately (and that Rufus will be able to handle just fine to produce the result you seek). And thus, considering that Rufus recently added the ability to download ISO images, by changing the _SELECT_ button into a split _SELECT/DOWNLOAD_ button, it is starting to make a lot of sense to propose the UEFI Shell ISO there, instead of trying to hide that behind the _Boot selection_ dropdown, and that is exactly what I am  planning to do to satisfy the feature you request __IF__ I decide to satisfy it.

In other words, I __may__ provide you with the feature you want, but it will only be proposed through the _SELECT/DOWNLOAD_ button, not the _Boot selection_ dropdown, and you will have to download an ISO to create the media.

And now that all the above has been mentioned, I can expose the list of the steps that need to be carried out before the feature find its way to Rufus, which should also give an idea as to why this is not a guaranteed shot (because, as you can see, it'll take quite a bit more work than it seems):

1. Because the EDK2 people currently do not provide __up to date__ Shell binaries, and because I suspect that people wanting that feature will prefer to have all the latest bells and whistle, the first thing I'll need to do is create a new project, similar to what I've recently started to do for the Raspberry Pi 3 [UEFI firmware](https://github.com/pbatard/RPi3/) or [ARM Trusted Firmware](https://github.com/pbatard/pitf), that automates the building of both Shell 1.0 and Shell 2.0 binaries from source. This is especially crucial as security minded people (but it should really be everyone) will want to validate that the Shell they are trying to run is not something that has seen any malicious code added. So, from the get go, that's one extra repo I need to maintain (because I'm going to have to generate builds on regular basis when the EDK2 code is updated)...
[... cut ...] t.

Oh, and for those who want to use a Shell on x86 and can't wait, you should be able to simply download the latest rEFInd ISO [from its project page](http://www.rodsbooks.com/refind/) and use that with Rufus, which is exactly what I advised the last person to ask for Rufus to offer a UEFI Shell option to do.

Finally, I'll conclude by stating that the internal _NinjaTurle_ codename we settled on is of course because it's a stealth attempt at adding a something that has to do with a shell... 😄

closed 2019-08-30 by pbatard

hlittler, 2019-09-04:
Pete

Thanks for the Long reply message.

I see what you mean about bloat in rufus.
Just including one version of the Boot64.efi file would double its size!
Rufus is very impressive at 1.08MB.

Downloading an ISO with the shell files is definitely the way to go.

My interest with the UEFI shell is that I’m working on a UEFIShell FileManager project like the FarManager for CMD Windows.
To be used for the recovery of user data from flty laptops,
instead of  running a WindowsPE (UEFI Boot:NTFS).

It’s early days yet  as I’ve just started, checking out the GraphicsConsole functions.

A Raspberry Pi3/4  UEFIShell  - I got to check that out.

Can’t confirm this but I think I read some ware that the new Pi4 supports GPT.

I won’t hold my breath, cross my fingers, or cross my eyes and legs.
But I will wistfully check the NinjaTurtles.

Many thanks for the Rufus

Hughie Littler
IT Support Tech
ED QLD

This message (including attachments) is intended for the addressee named above. It may also be confidential, privileged and/or subject to copyright. If you wish to forward this message to others, you must first obtain the permission of the author. If you are not the addressee named above, you must not disseminate, copy, communicate or otherwise use or take any action in reliance on this message. You understand that any privilege or confidentiality attached to this message is not waived, lost or destroyed because you have received this message in error. If you have received this message in error please notify the sender and delete from any computer. Unless explicitly attributed, the opinions expressed in this message do not necessarily represent the official position or opinions of the State of Queensland or the Department of Education, Training and Employment. Whilst all care has been taken, the Department of Education, Training and Employment disclaims all liability for loss or damage to person or property arising from this message being infected by computer virus or other contamination.

IMPORTANT: This email and any attachments may contain legally privileged, confidential or private information, and may be protected by copyright. You may only use or disclose this information if you are the intended recipient(s) and if you use it in an authorised way. No other person is allowed to use, review, alter, transmit, disclose, distribute, print or copy this email and any attachments without appropriate authorisation.

If you are not the intended recipient(s) and the email was sent to you by mistake, please notify the sender immediately by return email or phone, destroy any hardcopies of this email and any attachments and delete it from your system. Any legal privilege and confidentiality attached to this email is not waived or destroyed by that mistake.

The Department of Education carries out monitoring, scanning and blocking of emails and attachments sent from or to addresses within the Department of Education for the purposes of operating, protecting, maintaining and ensuring appropriate use of its computer network. It is your responsibility to ensure that this email does not contain and is not affected by computer viruses, defects or interference by third parties or replication problems (including incompatibility with your computer system).

The Department of Education does not accept any responsibility for any loss or damage that may result from reliance on, or the use of, any information contained in the email and any attachments.

***************************************************************************************************