pbatard/rufus #1372: Feature Proposal: Validate ISO checksum from clipboard

stdedos, 2019-09-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] ~I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.~
- [ ] ~The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.~

Issue description
-----------------
Feature proposal: Add an input box, where someone can copy-paste the supposed checksum of the file they are checking.
[image]
Add a validation method if the checksum matches something (e.g. make that box green?), or if it doesn't (make the input box red?)

I feel that that's more straightforward (and a bit more bulletproof) than trying to compare with your eyes (or faster)

Log
---
```
<FULL LOG>
```

pbatard, 2019-09-01:
I disagree. If you feel like copy/pasting checksums, and have trouble verifying a match, just open a text file anbd copy both the checksums on 2 individual lines. Any different will become exceedingly apparent.

Also see [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want). You're asking me to add a rather prominent change to the UI for a feature that is unlikely to benefit that many people.

&rarr; Rejected.

closed 2019-09-01 by pbatard

stdedos, 2019-09-01:
I have actually read your FAQ multiple times, because I enjoy doing so (even though it's extremely lengthy, it is extremely complete too).

I believe it is much less of a hassle when you don't have an `.*sum` file, more straightforward UX, and it offloads the burden from the user (very much so on the way you insist not providing SHA256 sums for your releases, because the application being signed means Windows will verify your executable's sum already).

But as you keep mentioning (indirectly) to your FAQ:
Your time, your software, your call