pbatard/rufus #1373: Broke usb...?

UnchartedScars, 2019-09-05:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I honestly don't know how to describe his, but basically i was formatting my 32gb Samsung usb from a bootable usb back to a super floppy disk so it could be used as a normal usb again, after running a speed test with CristalDiskMark from Cristal Dew World, it 1) ran slower 2) after being plugged in would fail in the system. To see if this is something from my side I'm currently doing a full format (slow) I can keep you posted to see if it fixes it or not I'm running the beta version so i thought id let you know so you'd be ready for future updates. maybe its something to be looked into?
Ruining beta v3.7.1570

Log
---
```
Rufus x86 v3.7.1570 (Beta)
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: OpenLocalMachineGPO failed - error 80004005
Localization set to 'en-US'
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#4&191c6f3b&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[059]: USB\VID_046D&PID_C52B\5&35BF0E71&0&1
  Found ID[107]: USB\VID_05C8&PID_038E\200901010001
  Found ID[250]: USB\VID_8087&PID_0A2A\5&35BF0E71&0&7
  Found ID[180]: USB\VID_0EEF&PID_C04D\5&35BF0E71&0&8
  Found ID[179]: USB\VID_2109&PID_2817\5&35BF0E71&0&2
  Found ID[228]: USB\VID_2109&PID_0817\5&35BF0E71&0&14
Processing Hub 3:
  Hub[2] = '\\?\usb#vid_2109&pid_2817#5&35bf0e71&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[246]: USB\VID_090C&PID_1000\0347018040020272
Processing IDs belonging to 'USBSTOR':
  USB\VID_090C&PID_1000\0347018040020272
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSamsung_Flash_Drive_____1100'
  Matched with ID[246]: USB\VID_090C&PID_1000\0347018040020272
  Matched with Hub[2]: '\\?\usb#vid_2109&pid_2817#5&35bf0e71&0&2#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Samsung Flash Drive USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
Could not open GUID volume '\\?\Volume{33e4e073-cedf-11e9-ba0c-ec8eb55495bb}': [0x00000005] Access is denied.
No logical drive found (unpartitioned?)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskHGST____HTS541010A9E680_JA0O'
Found non-USB non-removable device 'HGST HTS541010A9E680' => Eliminated
1 device found
Could not open GUID volume '\\?\Volume{33e4e073-cedf-11e9-ba0c-ec8eb55495bb}': [0x00000005] Access is denied.
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 29.9 GB (32080200192 bytes)
  Start Sector: 0, Boot: No

```

UnchartedScars, 2019-09-05:
Sorry about the log I realized that I needed It after I closed rufus, that's my bad.

pbatard, 2019-09-05:
I'm afraid USB flash drives fail all the time, through no fault of the application that is being used to access them. It's just the nature of flash memory to have a limited number of write cycles, which is __much shorter__ than people tend to think it is, especially for cheap drives. If you are lucky, your drive will last till well after the __mean__ number of write cycles has been reached. But if you aren't, a brand new drive may very well fail not long after it was purchased.

There's is literally nothing I or Rufus can do about these failures. The only thing Rufus will do, which I __strongly__ encourage you to perform, is run a bad blocks checks. Have you done that?
You can find the bad blocks checks under _Show advanced format options_.

Now, with regards to the speed test then one possibility is that the clusters are no longer aligned to sectors. But really, there again, this has nothing to do with Rufus, since, unlike what is the case for MBR or GPT (where Rufus does try to align partitions), you simply cannot do that for Super Floppy, since there are no partitions to start with.

Therefore, I'm afraid that I don't see much I can help you with here.

UnchartedScars, 2019-09-07:
After the complete format leaving the computer off for a day and rebooting it the second time with repairing the drive, its finally running back at 255 instead of 25. just to keep you updated.

UnchartedScars, 2019-09-07:
thanks for the comment you gave, that was really informative and im sorry for being a bother
have a great day

closed 2019-09-07 by UnchartedScars