pbatard/rufus #1374: parameter is not correct with 3.7 and persistence

sonoro1234, 2019-09-09:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to create a bootable usb with persistence (12 gb) with ubuntu-18.04.3-desktop-amd64.iso returns with error 0xC0030057 (parameter is not correct)
This only happens if persistence is requested.

Log
---
```
<Rufus x86 v3.7.1576 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0C0A (es-ES)
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
Localization set to 'en-US'
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1937, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000BA949
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.8 GB (3052417536 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 12.0 GB (12884884992 bytes)
  Start Sector: 5963801, Boot: No
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 2.8 GB)
● Partition 0 (offset: 3053466112, size: 12.0 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 2.8 GB)
● Creating Linux Persistence Partition (offset: 3053466112, size: 12.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Unexpected volume name: '(null)'
Re-mounted volume as F: after error

Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1937, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00113A13
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.8 GB (3052417536 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 12.0 GB (12884884992 bytes)
  Start Sector: 5963801, Boot: No
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1937, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00113A13
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.8 GB (3052417536 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 12.0 GB (12884884992 bytes)
  Start Sector: 5963801, Boot: No
>
```

pbatard, 2019-09-09:
Well, first of all, persistence will __NOT__ work with Ubuntu 18.04 (yes, even 18.04.03 that was released after August 1st). This is because, for now, the fix for [bug 1489855](https://bugs.launchpad.net/ubuntu/+source/casper/+bug/1489855) has only been included in the 19.10 daily ISOs.

As to the `0xC0030057` (and other) codes, yeah, this is a result of Windows being an ass when it comes to creating partitions it doesn't know how to handle. The trick is to insist or unplug/replug the drive or zero the drive (<kbd>Alt</kbd>-<kbd>Z</kbd>) for a while or try to first create a FreeDOS drive, and then create a drive with persistence.

At this stage I am not sure I'll ever had a good solution to avoid these error codes, but I can tell you that, if you insist, you should managed to get a persistent drive (at least, in all my testing, I have __never__ found a drive where I couldn't add persistence with a little bit of cajoling).

sonoro1234, 2019-09-09:
Now with eoan-desktop-amd64.iso after first creating a FreeDOS drive.
Says: the device is in use by another process.

pbatard, 2019-09-09:
You're going to have to figure out what other process is using the drive.

But I'm also re-testing on Windows 7, and I'm getting the feeling that the change I applied in 94e2015edf6c919b1cd82adf3cfdd28848b7d177 where I switched to using `AltMountVolume()` instead of `GetPartitionName()` to get the drive to format doesn't sit too well with Windows 7...

The thing is, I did that because I found it made Windows 10 more stable with the creation of `ext3` partitions...
That would certainly explain the `Unexpected volume name: '(null)'` message (and of course, if we don't get a volume name, we can't format much of anything).

If other people confirm they have the same issue, I guess I'm going to have to declare that persistence creation is not compatible with Windows 7 for the time being, and try to revert that change in Rufus 3.8... 

I can only wish people had reported that issue during the 2 weeks BETA we had for Rufus 3.7, as it would have given me an opportunity to fix this before release.

sonoro1234, 2019-09-09:
>You're going to have to figure out what other process is using the drive.

It is explorer saying that usb is not formated and offering: Do you want to format?

pbatard, 2019-09-09:
Makes sense, though there is code in Rufus that is designed to prevent these prompts from appearing (and last time I checked, these __did__ work on Windows 7 for me).

pbatard, 2019-09-15:
I believe I have a fix for the above. It will be part of the upcoming Rufus 3.8 BUGFIX release, which I'm planning to publish in the coming week, since I consider that both this issue and #1375 are rather important matters that require an out-of-band release.

If you want to test, you can download an early preview of this BUGFIX release [HERE](https://rufus-web.akeo.ie/testing/rufus-3.8_TEST1.exe).

Note that, as I push the commit with the fix, this issue will be automatically closed. But that won't prevent anyone from commenting further if you wish (for instance to report whether 3.8 works or if you still see an issue).

closed 2019-09-15 by pbatard

sonoro1234, 2019-09-16:
I have tested succesfully the BUGFIX release!!!
Thanks!!

sonoro1234, 2019-09-16:
As I could not install Jack in ubuntu 19.04 I tried with ubuntu studio 19.04 from https://ubuntustudio.org/download/

I could made the usb but after booting it only shows inittransfm console
Should work with ubuntu studio?

pbatard, 2019-09-16:
As per what I said above:

> Well, first of all, persistence will __NOT__ work with Ubuntu 18.04 (yes, even 18.04.03 that was released after August 1st). This is because, for now, the fix for [bug 1489855](https://bugs.launchpad.net/ubuntu/+source/casper/+bug/1489855) has only been included in the 19.10 daily ISOs.

commit bf8d888b (2019-09-15): [ext2fs] fix creation of persistent partitions for pre-1703 platforms

* Windows platforms prior to Windows 10 1703 cannot access any logical partition besides the
  first one (we don't even get a volume for those).
* This fix enables the use of physical + offset for ext# formatting to work around this,
  which is file since we don't actually need to mount the partition.
* Also fix ext2fs_open2() not handling normalized versions of Windows drive paths ("\\?\...")
* Also fix an issue where we would make the drive letter unavailable after formatting a
  standalone partition to ext#.
* Also ensur [... cut]