pbatard/rufus #1375: Botable deboan work only with old rufus

szymon1051, 2019-09-11:
﻿

Checklist
---------
- [✓ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [✓ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [✓] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [✓ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Try make bootable [debian 10.1](https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/debian-10.1.0-amd64-netinst.iso). With rufus 3.7 show syslinux "SYSLINUX: No DEFAULT or UI configuration directive found!" but with rufus 3.4 work normal.

Log
---
```
Rufus x86 v3.4.1430
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0415
Will use default UI locale 0x0415
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pl-PL'
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:CB10)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1886, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0040D182
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes
0 devices found
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:CB10)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1886, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0040D182
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:CB10)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1886, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0040D182
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
[... 2359 lines cut ...]
Extracting: F:\win32-loader.ini (233 bytes)
Extracting: F:\.disk\base_components (5 bytes)
Extracting: F:\.disk\base_installable (0 bytes)
Extracting: F:\.disk\cd_type (13 bytes)
Extracting: F:\.disk\info (72 bytes)
Extracting: F:\.disk\mkisofs (1 KB)
Extracting: F:\.disk\udeb_include (34 bytes)
Created: F:\syslinux.cfg
Finalizing, please wait...
F:autorun.inf already exists - keeping it

Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:CB10)
Using autorun.inf label for drive F: 'Install Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1886, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003691C4
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2019-09-11:
Can you please provide the log of how you created the drive with Rufus 3.7? In most cases, the developer is only interested in the log from when you experienced an issue, not the one that works. Thanks.

pbatard, 2019-09-11:
I think I have an idea of why that is happening, and what you can do to work around it.

On your USB, please open the file `syslinux.cfg` with a text editor. You should see that its content is as follows:
```
DEFAULT loadconfig

LABEL loadconfig
  CONFIG /isolinux/txt.cfg
  APPEND /isolinux/
```

You should change the `CONFIG` line so that it says
```
  CONFIG /isolinux/isolinux.cfg
```

Alternatively, if you create the USB drive in DD mode (as Rufus does suggest you try if you encounter an issue during boot in ISO mode), it should also avoid the problem.

Now, the reason why this happening is because Ubuntu are using a `txt.cfg` file to specify kernel options with Syslinux/Isolinux, and, in order to finalize persistence support in 3.7 we needed to detect that this file is a Syslinux config file. So it was added to the list of files we consider as Syslinux config.  However, our current code to decide which config file should be referenced for Syslinux boot is to always use the one with the shortest path, therefore, since `/isolinux/txt.cfg` is shorter than `/isolinux/isolinux.cfg`, we are ending up referencing the former in the `syslinux.cfg` we create hence the issue.

Apparently, this is not a problem for Ubuntu, but it is one for Debian...

I'll try to fix this for Rufus 3.8.

pbatard, 2019-09-15:
For those interested, I have a pre-release of Rufus 3.8 that you can test to confirm that it addresses the issue. You can download it [HERE](rufus-3.8_TEST1.exe).

szymon1051, 2019-09-17:
With 3.8 debian work.

commit 0a24940c (2021-05-27): [iso] fix detection of 'txt.cfg' as a main Syslinux config file

* While we need to detect that 'txt.cfg' is a Syslinux config file, so that
  we can alter it for persistence, it should never be used as a main config
  file, such as the one we link to when we create /syslinux.cfg.
* Closes #1375