pbatard/rufus #1377: Rufus detects multple unrelated drive letters as a single device.

esgeroth, 2019-09-13:
We use a document management system called M-Files that creates a mapped drive containing its data vault. 
When I run Rufus to write an iso to a usb stick, in the device drop down is "Multiple Partitions (D:) (M:) [32GB]"
The D: drive is my usb drive and the M: drive is the mapped drive for the M-Files vault. They have nothing to do with each other. The M: drive is mapped to a network share.
Fortunately I can continue with writing the iso without damaging the data in the M: drive but then I have to restart to get my M: drive back. 

﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2019-09-13:
Can you please attach a new log after pressing <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus? No need to recreate the Windows bootable USB, but you should have the USB plugged in so that Rufus displays that `Multiple Partitions (D:) (M:) [32GB]` message.

This will turn on enumeration debug which will hopefully give a better idea as to what is going on.

esgeroth, 2019-09-13:
Sure here it is. I probably shouldn't refer to the M: drive as a mapped drive. It's actually a virtual drive that is listed under devices and drives, not network locations. I'm not surprised that rufus would detect the M: drive, only that it thinks the M: and D: volumes are both on the same drive. 

Rufus x86 v3.7.1576 (Portable)
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Found USB 2.0 device '' (FFFF:5678)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 489, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {385F8A6D-FD93-44B5-8CB5-F83CB88E4F79}
Max parts: 128, Start Offset: 17408, Usable = 4026497536 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {F2A537AF-3821-47D1-B629-21F4FB7B5FCD}
  Size: 3.7 GB (4025466368 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&4d26b37&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[149]: USB\VID_FFFF&PID_5678\9207056015624010356
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&261c85dd&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[129]: USB\VID_8087&PID_0024\5&3B073210&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#root_hub20#4&22069ec8&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[012]: USB\VID_8087&PID_0024\5&4D26B37&0&1
Processing Hub 4:
  Hub[4] = '\\?\usb#root_hub30#4&8b3247a&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[044]: USB\VID_413C&PID_2003\5&854F772&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#vid_8087&pid_0024#5&3b073210&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[156]: USB\VID_03F0&PID_094A\6&1E01D&0&5
Processing IDs belonging to 'USBSTOR':
  USB\VID_FFFF&PID_5678\9207056015624010356
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskINTEL______SSDSC2KW480H6_LSF'
Found non-USB non-removable device 'INTEL SSDSC2KW480H6' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\Disk________________________2.00'
  Matched with ID[149]: USB\VID_FFFF&PID_5678\9207056015624010356
  Matched with Hub[1]: '\\?\usb#vid_8087&pid_0024#5&4d26b37&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device '' (FFFF:5678)
1 device found
Windows volumes from this device:
● \\?\Volume{9bcefbfb-c7fa-11e9-b85c-4437e6c39780}\ @1048576
● \\?\Volume{14cb27db-b3ce-11e9-b84a-4437e6c39780}\ @0
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 489, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {385F8A6D-FD93-44B5-8CB5-F83CB88E4F79}
Max parts: 128, Start Offset: 17408, Usable = 4026497536 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {F2A537AF-3821-47D1-B629-21F4FB7B5FCD}
  Size: 3.7 GB (4025466368 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&4d26b37&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[149]: USB\VID_FFFF&PID_5678\9207056015624010356
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&261c85dd&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[129]: USB\VID_8087&PID_0024\5&3B073210&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#root_hub20#4&22069ec8&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[012]: USB\VID_8087&PID_0024\5&4D26B37&0&1
Processing Hub 4:
  Hub[4] = '\\?\usb#root_hub30#4&8b3247a&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[044]: USB\VID_413C&PID_2003\5&854F772&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#vid_8087&pid_0024#5&3b073210&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[156]: USB\VID_03F0&PID_094A\6&1E01D&0&5
Processing IDs belonging to 'USBSTOR':
  USB\VID_FFFF&PID_5678\9207056015624010356
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskINTEL______SSDSC2KW480H6_LSF'
Found non-USB non-removable device 'INTEL SSDSC2KW480H6' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\Disk________________________2.00'
  Matched with ID[149]: USB\VID_FFFF&PID_5678\9207056015624010356
  Matched with Hub[1]: '\\?\usb#vid_8087&pid_0024#5&4d26b37&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device '' (FFFF:5678)
1 device found
Windows volumes from this device:
● \\?\Volume{9bcefbfb-c7fa-11e9-b85c-4437e6c39780}\ @1048576
● \\?\Volume{14cb27db-b3ce-11e9-b84a-4437e6c39780}\ @0
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 489, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {385F8A6D-FD93-44B5-8CB5-F83CB88E4F79}
Max parts: 128, Start Offset: 17408, Usable = 4026497536 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {F2A537AF-3821-47D1-B629-21F4FB7B5FCD}
  Size: 3.7 GB (4025466368 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

pbatard, 2019-09-13:
Thanks. I'll see what I can do.

pbatard, 2019-11-01:
I finally had a look into this, but it would look to me like the issue is with the M-Files software.

The way Rufus associates drive letters with a drive is by issuing an `IOCTL_STORAGE_GET_DEVICE_NUMBER` request to get the device number and then reporting all the letters that are associated with this device number, which is meant to be unique per device type.

Now, Rufus does filter out device numbers that belong to classes such as `DRIVE_CDROM`, `DRIVE_REMOTE` or `DRIVE_RAMDISK`, and, IMO, if I understand what M-Files does, it should be using the `DRIVE_REMOTE` class rather than `DRIVE_REMOVABLE` or `DRIVE_FIXED` which is the 2 classes Rufus lists partition from. So that's already one issue.

And I believe the second issue that that, even as they use one of the `DRIVE_REMOVABLE` or `DRIVE_FIXED` class, M-Files decided to assign a device number without registering it with Windows (so that it knows that this device number is in use and won't assign it to the next removable drive being mounted), which means that, ultimately, the disk device used by M-Files is seen as the same disk device as the one from your USB.

I'm also a bit puzzled with your USB 2.0 device with VID `0xFFFF` (this is not an officially registered VID) as well as an empty name, though there's nothing really problematic with it, and Rufus doesn't care about that.

As such, since I believe the issue is with the M-Files software, I will close this issue.

closed 2019-11-01 by pbatard