pbatard/rufus #1378: More verbose complex fallback mechanism

samuelharmer, 2019-09-26:
Rufus must run under an Administrator account (logical). Administrator accounts are not permitted through the proxy in my situation (logical). So when Rufus tries to download a newer ldlinux.* it fails. That's fine; that's my problem.

This request is for the [complex fallback mechanism][1] to
- try all URL permutations that it would do so were the HTTP connection to succeed
- and state which directory it would put that downloaded file into within `rufus_files`.

That way, in a situation like mine I could craft `rufus_files` by reading the log, downloading the necessary files manually and placing them in the expected directory structure. I managed to do this with a little bit of trial and error but the above feature would be extremely helpful.

The alternative which automatically uses [PAC][2] combined with a 'prompt user for net access' is horrible and well beyond the remit of Rufus so I'm not asking for that. Another alternative would be running and downloading using the regular user account, and only the device accessing bit prompting for Administrative access. I suspect I'm unaware of good reasons why that's not the way Rufus works though!

[1]: https://github.com/pbatard/rufus/wiki/FAQ#Rufus_says_that_the_download_files_for_SyslinuxGRUB_are_missing_on_the_server
[2]: http://findproxyforurl.com/

JonnyTech, 2019-09-26:
FYI: https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them

pbatard, 2019-09-26:
> This request is for the [complex fallback mechanism](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_says_that_the_download_files_for_SyslinuxGRUB_are_missing_on_the_server)...

That one's on me. You were almost there with the FAQ, but the information you are seeking was actually in the [Usage Notes](https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them) (please don't ask me why some stuff gets relegated there, it's a long story).

I have now added a link to that in the FAQ entry you linked, which should avoid others running into the same predicament. 

> That way, in a situation like mine I could craft rufus_files by reading the log, downloading the necessary files manually and placing them in the expected directory structure.

That's what the Usage Notes entry is about. I'm not planning to go further than this at this stage. Especially, I'm not planning to add extra info in the app to help people who want/need to manually copy files. Rufus expects that it will be able to connect to the internet to carry out its operation. If, for whatever reason, it can't, my position is that you are on your own.

> I suspect I'm unaware of good reasons why that's not the way Rufus works though!

Too much complexity added to solve an issue that is only going to help __you__ and a very limited number of people.

Yeah, if I had __infinite__ development, testing (don't forget that if I add a feature, I need to test that it works) and maintenance time (make sure it still works with new releases, which, the more you add features and workarounds like that, the more it becomes a MAJOR HEADACHE to validate this), I wouldn't mind trying to help with every single user corner case.

But the fact is I really don't, so I'll continue to estimate how many people are __really__ going to be affected by a specific issue, and if that estimated count is not high enough, I'll use the time I save __not__ doing that on one of the many many other things I could use my development time for.

Hopefully, the Usage Notes entry is what you really wanted. Therefore I will close this issue.

closed 2019-09-26 by pbatard

samuelharmer, 2019-09-26:
Thanks both for your responses. [Where to manually copy the Syslinux/GRUB files, if Rufus can't download them][1] looks to have the information needed.

Following up regarding the information found there:
> Especially, now that most Linux distributions have upgraded to Syslinux 6.03, and considering that Rufus does embed generic Syslinux 6.03 files, the remote download from our servers is mostly there as a safety measure, in case a distribution has added some very custom patches on top of Syslinux that they really won't be able to boot with otherwise, which I am not aware any of the major distributions (Debian, Arch, Ubuntu, etc.) has ever done.

I don't know if you're aware that [Debian 10.1.0][2] seems to have a newer Syslinux version than Rufus'.

> ISO analysis:
> Image is an ISO9660 image
> Will use '/isolinux/isolinux.cfg' for Syslinux
> Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')

[1]: https://github.com/pbatard/rufus/wiki/Usage-Notes#where-to-manually-copy-the-syslinuxgrub-files-if-rufus-cant-download-them
[2]: https://cdimage.debian.org/images/unofficial/non-free/images-including-firmware/10.1.0-live+nonfree/amd64/iso-hybrid/

pbatard, 2019-09-26:
I am well aware of it. Debian are using a __pre-release__ of 6.04 (since Syslinux 6.04 has __NOT__ been released and has been in pre-release for more than 3 years now...). [These](https://mirrors.edge.kernel.org/pub/linux/utils/boot/syslinux/) are the __official releases__ of Syslinux, and as you can see, 6.04 is not there.

But then again, it doesn't matter since the internet connection and fallback mechanism requirement is precisely for that kind of stuff. We've been providing Syslinux 6.04 support for the distros using the pre-release from about the very first day the pre-release was published. Heck, we're even supporting pre-2, even though there's no formal release for that.

See https://github.com/pbatard/rufus-web/tree/gh-pages/files/syslinux-6.04.

__THAT__ is the whole point of wanting people to sort their internet access when using Rufus, so that, if the Syslinux people formally release Syslinux 6.04 tomorrow (or consider that 6.04 is such a mess that they might as well jump straight ahead to 6.05), I can __immediately__ get that supported, in __ALL__ existing versions of Rufus, by dropping the required files on the server.

samuelharmer, 2019-09-26:
And to clarify (at least in my situation), when prompted to download the newer Syslinux version, pressing Yes (which fails) also halts the process of writing the ISO. This means one is forced to fabricate `rufus_files` rather than just giving 6.03 a shot as the Usage Notes advise.

samuelharmer, 2019-09-26:
I can't fix the policy of not granting internet access to Administrative accounts. I wouldn't want to either. That is sensible in any security conscious situation.

But thank you for your time. Much appreciated!

pbatard, 2019-09-26:
For the 6.03 automatic fallback, try with your ethernet cable unplugged. I must admit that I haven't tested the fallback mechanism for a while, and it might only apply on specific non-connectivity conditions (because it'd take too much of my time to try to replicate the many different conditions under which a network connection can be broken, and, IIRC, I rely on the native Windows libraries to tell me that there is an issue with the connection, which it may not always do).

samuelharmer, 2019-09-26:
Yep. It must just be hanging. Thanks again.