pbatard/rufus #1379: Cannot get rufus-3.x.exe to open on Windows 10

klopes91, 2019-09-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have attempted to open the rufus-3.5.exe, and the rufus-3.8.exe executables on at least 3 different Windows 10 machines. All of them have failed to launch the executable after prompting to allow admin privileges. No GUI element pops up after accepting the admin privileges.

Log
---
```
<FULL LOG>
```

JonnyTech, 2019-09-26:
Do any older versions work?

pbatard, 2019-09-26:
Okay. I've had very sporadic reports of this happening (not recently, but I've gotten maybe 5-6 reports like yours over the past 3-4 years), but so far, I have never ever managed to replicate the issue and this is not a widespread issue. Especially, I have reasons to believe that it could have to do with the security solution you are using or the way your environment is configured, and how it may deviate from a vanilla Windows installation.

One thing that may help is if you have DebugView running while you launch Rufus, as Rufus will send its log data there, and it may give a better idea of the stage at which it seems stuck (though I'm afraid it would require more instrumentation to find out precisely where it is stuck). You can download and install DebugView from https://docs.microsoft.com/en-us/sysinternals/downloads/debugview.

Of course, __the most helpful thing one can do__, if someone happens to see this issue systematically, is install [Visual Studio 2019](https://visualstudio.microsoft.com/vs/) (the Community Edition is completely free to install and run) on the computer where you are seeing the problem, then clone the github project and recompile/run Rufus from source, in Debug mode. Then you will then be able to find out precisely where in the code Rufus seems to be stuck on launch.

Short of this, because those reports are exceedingly sporadic and appear to be tied to specific environment factors, it's next to impossible for me to troubleshoot the problem...

pbatard, 2019-10-16:
Closing the issue. Will reopen if someone who experiences it can provide the data requested.

closed 2019-10-16 by pbatard