pbatard/rufus #1381: "BOOTMGR is compressed" error on boot

CyberTacoX, 2019-09-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

(I know using NTFS compression isn't officially supported, but it works great and this is an easy one, I already know the fix and it's simple.)
Hello! When the NTFS compression cheat key is enabled, if Rufus is putting a file called "BOOTMGR" in the root of a partition, could that one file be not compressed, please? Otherwise, you get a "BOOTMGR is compressed / Press Ctrl-Alt-Del to restart" message on boot. (Everything else works great once that file is uncompressed.) Thank you! :-)

Log
---
```
(N/A: No log because problem and solution are very obvious.)
```

pbatard, 2019-09-28:
I'm gonna assume that this happens when using a Windows ISO, right (please don't assume that I don't need to see your log. If your issue happens after creating an image from an ISO, __ALWAYS__ provide the log period). Also have you tested both BIOS and UEFI? Recent Windows ISOs have a `bootmgr.efi` as well. Can you tell if that one can be left compressed?

CyberTacoX, 2019-10-05:
Fair enough about the log. I just made a usb boot stick to test what you asked about, so I'll put the log from that below.
Yes, it's for Windows ISOs. After some testing, here's what I've learned: For bios, bootmgr (no extension) must not be compressed, for uefi, it doesn't matter. For bootmgr.efi, it doesn't matter if it's compressed for both bios and uefi, either way boots fine (which is interesting; I didn't expect that to work for uefi). 
Also, could we please have the setting to use ntfs compression carry over between runs of Rufus? Besides this one issue, it works so well that I have no reason to turn it off under any circumstance I've come across. :-)

```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
1 device found
Disk type: FIXED, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3892, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004D6548
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014630912 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GSP1RMCHPXFRER_EN_DVD'
  Size: 3.1 GB (Projected)
  Uses: EFI (win7_x64)
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Windows 7 64bit SP1 (original).iso (3.1 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 29.8 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 29.8 GB)
● Creating UEFI:NTFS Partition (offset: 32014612480, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Enabled NTFS compression
Format completed.
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume \\?\Volume{30ad0dee-e675-11e9-8ee5-681ca2075c6e}\
\\?\Volume{30ad0dee-e675-11e9-8ee5-681ca2075c6e}\ is already mounted as G:
Disabling file indexing...
[... 915 lines cut ...]
     68185 KB in use by the system.
     65536 KB occupied by the log file.
  28020336 KB available on disk.
      4096 bytes in each allocation unit.
   7815811 total allocation units on disk.
   7005084 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
1 device found
Disk type: FIXED, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3892, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x08399067
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32013563904 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 62528540, Boot: No

```

josephsmendoza, 2019-10-16:
When I had this issue, it was enough to run `compact /u bootmgr bootmgr.efi` to decompress the bootmgr files.

pbatard, 2019-10-22:
@josephsmendoza, that's the simplest solution, so that's what I'm going to with, thanks.

At this stage, I don't really think that `bootmgr.efi` actually needs to be left uncompressed (or rather, to be consistent we'd also need to have the various `/efi/boot/boot[...].efi` uncompressed as well), but I'll uncompress it as well, since it doesn't cost much. I would assert however that someone who wants to boot compressed NTFS on UEFI can just create a UEFI:NTFS drive with Rufus, since the NTFS driver we use supports compression.

I have applied the needed changes to the source, which will be part of the next release, so I will close this issue with the next commit.

closed 2019-10-22 by pbatard

josephsmendoza, 2019-10-29:
It hadn't occurred to me, but now that I think about it, I don't think I've had this issue on EFI. Especially if the included driver supports compression, I'd leave `bootmgr.efi` compressed, and if it still has issues, forward the issue to the driver vendor (assuming that isn't you)

CyberTacoX, 2019-11-02:
> I have applied the needed changes to the source, which will be part of the next release

Fantastic! Thank you very much! :-)

josephsmendoza, 2019-12-20:
Leaving this comment to update later. Got `file not found /efi/boot/bootx64.efi` or something like that at boot, whether `bootmgr.efi` was compressed or not. I decompressed the entire USB to fix it, will do more investigation at a later date. Will update today with the mobo info.

josephsmendoza, 2019-12-20:
ROG STRIX B350-F GAMING

pbatard, 2019-12-20:
Well, of course you will still get this issue since Rufus 3.9, which will be the first version to fix this problem, has not been released yet (and no, I don't know exactly when it will be released... maybe February?)

josephsmendoza, 2019-12-21:
Actually, it's the same issue but on different files. The fix as stated here would not prevent this error. Specifically, at least `efi/boot/bootx64.efi` needs to also be decompressed. I'll test and report back tomorrow.

josephsmendoza, 2019-12-21:
While testing, I noticed that the `compact` command will not compress `bootmgr`, but will do so for `.efi` files, indicating that the issue may not be the compression, but rather is in this motherboard's built-in NTFS EFI driver, which in this case was used instead of the driver provided by Rufus.

pbatard, 2019-12-21:
Yeah, you are seriously asking for trouble trying to use compressed NTFS for EFI installation.

This is very different from BIOS installs, where the file system driver is provided by the Microsoft bootloader, and therefore compressed NTFS support is provided by the installer itself and can be expected to be supported.

But for EFI boot, the NTFS file system drivers are __NOT__ provided by Microsoft themselves as part of their installation media, and therefore, unlike what is the case for BIOS, there is __NO__ expectation a user should have about being able to use compressed NTFS for EFI install.

Hence your issue. Depending on the driver being used (the one provided by Rufus should be fine, but the one provided by your mobo manufacturer might not), you will run into all sort of issues.

I'll apply your fix. But let this be clear: expect exactly __ZERO__ level of support from me when it comes to EFI Windows installations and compressed NTFS.

josephsmendoza, 2019-12-23:
Thank you for the prompt replies, and for applying the fix, and obviously for Rufus.

commit 87194126 (2019-10-22): [iso] leave bootmgr and bootmgr.efi uncompressed on compressed NTFS

* When using compressed NTFS, having a compressed bootmgr prevents BIOS boot, so we
  now call `compress -u` where needed to leave the relevant bootmgr files uncompressed.
* Closes #1381
* Also fix a minor warning in ext2fs