pbatard/rufus #1382: Rufus 3.8 (&3.7) fail to write compressed image with "error: invalid magic"

coxtehude, 2019-10-04:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 3.7 (tested) & 3.8 (tested) fail to write an image file to an mSD card with "Error: invalid magic".
Previous version 3.4 (tested, have yet to test interim 3.5 or 3.6 to assess their behaviour) completes the write just fine...

Log - 3.8 (failing with error: invalid magic)
---
```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found card reader device 'SD Card'
1 device found
Disk type: Removable, Disk size: 1GB, Sector size: 512 bytes
Cylinders: 123, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 968 MB (1015021568 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  'xxxxxxxxxobfuscatedzzzzzzzzzzz.gz' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: xxxxxxxxxobfuscatedzzzzzzzzzzz.gz.gz (71.9 MB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 0, size: 968 MB)
Could not delete partitions: [0x80070032] The request is not supported.
Opened \\.\PhysicalDrive1 for exclusive write access
I/O boundary checks disabled
Writing compressed image...
Error: invalid magic
Could not write compressed image: -1
Remounted \\?\Volume{4c1eeca5-37ff-11e8-a91a-9061aecc2cbf}\ as F:
\\?\Volume{4c1eeca5-37ff-11e8-a91a-9061aecc2cbf}\ is already mounted as F:
Re-mounted volume as F: after error

Found card reader device 'SD Card'
1 device found
Disk type: Removable, Disk size: 1GB, Sector size: 512 bytes
Cylinders: 123, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 968 MB (1015021568 bytes)
  Start Sector: 0, Boot: No
Found card reader device 'SD Card'
1 device found
Disk type: Removable, Disk size: 1GB, Sector size: 512 bytes
Cylinders: 123, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 968 MB (1015021568 bytes)
  Start Sector: 0, Boot: No
```

Log - 3.4 (completing the write just fine)
---
```
Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Unable to apply NoDriveTypeAutorun policy - error 80070020
Localization set to 'en-US'
Found card reader device 'SD Card'
1 device found
Disk type: Removable, Disk size: 1GB, Sector size: 512 bytes
Cylinders: 123, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 968 MB (1015021568 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  'xxxxxxxxxobfuscatedzzzzzzzzzzz.gz' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: xxxxxxxxxobfuscatedzzzzzzzzzzz.gz (71.9 MB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partitions from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 0, size: 968 MB)
Could not delete partitions: 0x80070032
Opened \\.\PhysicalDrive1 for exclusive write access
I/O boundary checks disabled
Writing compressed image...
Remounted \\?\Volume{4c1eeca5-37ff-11e8-a91a-9061aecc2cbf}\ on F:\

Found card reader device 'SD Card'
1 device found
Disk type: Removable, Disk size: 1GB, Sector size: 512 bytes
Cylinders: 123, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 968 MB (1015021568 bytes)
  Start Sector: 0, Boot: No
Found card reader device 'SD Card'
1 device found
Disk type: Removable, Disk size: 1GB, Sector size: 512 bytes
Cylinders: 123, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 968 MB (1015021568 bytes)
  Start Sector: 0, Boot: No

```

pbatard, 2019-10-04:
I'm sorry but I'm going to have access to that `.gz` image of yours to address this issue.

Can you upload it somewhere and e-mail me the link to: pete@akeo.ie?

pbatard, 2019-10-16:
Closing issue. Will reopen if OP provides the image.

closed 2019-10-16 by pbatard

geoleven, 2019-12-14:
I can provide a link to an image that does this.
https://download.freenas.org/11.2/STABLE/U7/x64/FreeNAS-11.2-U7.iso
It is the stable version of the current FreeNAS.

I confirmed that with my previous version (3.1) I could write it, with version 3.8 it fails every time with the error 0x80070032.

pbatard, 2019-12-14:
Sorry, but that's __not__ the same issue.

Unless you are seeing the message "error: invalid magic", you will need to open a new issue (and make sure you follow the checklist, as at the very least I'll need to see your full log.

Please don't piggyback on an issue if the title of the issue doesn't match what you are observing. This creates problems for everybody, and actually makes it less likely for you to get help.