pbatard/rufus #1384: Fail to boot in UEFI mode - Start failure: [17] No mapping

LordBurrito, 2019-10-06:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm trying to boot windows 10 in UEFI mode with CSM disabled but I receive this error:

[image]

I've tried to use different ISO all taken from the windows website.

I'm able to boot an Ubuntu image with CSM disabled.

I'm using:
Motherboard - Asus Rog Z370-H BIOS version 2102
CPU - Intel 8700K
SSD - Samsung 960 EVO 250GB m.2

let me know if you need something else

Thanks

Log
---
```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'Verbatim USB External SSD USB Device' (18A5:0246)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15346, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {9AD48E5B-3284-4097-86AF-CD2D0968C581}
Max parts: 128, Start Offset: 17408, Usable = 126231738880 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {50EEF292-1B39-4719-B112-74DCE48D431D}
  Size: 117.6 GB (126229630464 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {313DBC52-622C-4235-9CF7-216A49497E63}
  Size: 512 KB (524288 bytes)
  Start Sector: 246544295, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
[... 1033 lines cut ...]
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {5765FAF5-D406-4E66-849E-CD28E6B0A6E9}
  Size: 512 KB (524288 bytes)
  Start Sector: 246544295, Attributes: 0x0000000000000000
Found USB 3.0 device 'Verbatim USB External SSD USB Device' (18A5:0246)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15346, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {16E5A311-517A-4302-979B-6603938C83FE}
Max parts: 128, Start Offset: 17408, Usable = 126231738880 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {C0AA8258-B345-481E-BA45-3C79D48AFCEC}
  Size: 117.6 GB (126229630464 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {5765FAF5-D406-4E66-849E-CD28E6B0A6E9}
  Size: 512 KB (524288 bytes)
  Start Sector: 246544295, Attributes: 0x0000000000000000
```

pbatard, 2019-10-06:
Thanks for the report.

The fact that you get the warning `An NTFS service is already loaded` indicates that the UEFI firmware advertises that it __already__ has an NTFS driver, and therefore that the one that UEFI:NTFS (which is the custom UEFI bootloader that Rufus installs to boot from NTFS partitions) should not be necessary.

As a result, because you can't have 2 separate drivers (the one from the ASUS UEFI firmware and the one from Rufus) installed for the same purpose (providing access to NTFS volumes), the UEFI:NTFS bootloader attempts to use the ASUS advertised one to boot.

However, it seems that this driver, which should be able to handle the generic calls issued by UEFI:NTFS to locate and launch the Windows bootloader residing on the NTFS partition, doesn't play nice and returns a mapping error...

Now, if you are lucky, the above means that when you are trying to boot your Windows USB drive, you should see __2__ entries for UEFI boot of that device: One for the UEFI:NTFS bootloader (that resides on the second small FAT partition that Rufus created on the drive, and which is the one that produces the messages above) and one for the Windows bootloader that resides on the primary NTFS partition. If that is the case, you should try the other bootloader, as, if Asus didn't screw up their NTFS driver, it should allow you to boot Windows without using the UEFI:NTFS workaround.

One the other hand, if you only see one entry for the UEFI boot of your USB drive, then it means that Asus screwed up their driver, as it means that, as far as UEFI applications are concerned, they do advertise that an NTFS driver is available to access NTFS partitions, yet, if that driver cannot be used to enumerate the Windows native bootloader (which is what should produce the second UEFI boot entry for your USB drive), or to allow UEFI applications to access NTFS files, then it means that their driver is broken.

So, do you see two boot entries for the USB drive or not? And if you see two, what happens when you select the other one?

LordBurrito, 2019-10-06:
Hi pbatard,

I do see two entries but I have the same error:

[image]

pbatard, 2019-10-06:
That's not possible. The 2 entries clearly reference 2 different partitions which means that it's impossible for the UEFI:NTFS bootloader to be executed when you select Partition 1.

This means that, no, you are not seeing the same error when you select Partition 1, because the UEFI firmware will try to run a __completely different__ bootloader then.

So, what error do you get then?

LordBurrito, 2019-10-06:
I've manually tried to boot into the two partition.
Probably one doesn't really show anything and it immediately tries to boot the second one; that's why I see the same error

pbatard, 2019-10-06:
Then you'll need to take it with Asus, because that's an Asus issue.

The fact that you can select the first partition means that their own NTFS driver managed to locate `\efi\boot\bootx64.efi` on the NTFS partition (which is Partition 1). Yet, even as it did locate that bootloader, it doesn't seem to be able to launch it, which it should (because that's the whole point of having an NTFS driver in the firmware).

In other words, you should see the exact same issue (failure to boot) if you simply create an NTFS drive with a single partition and then manually copy the whole content of your Windows ISO, in which case you need to tell Asus support that their NTFS driver doesn't work and needs to be fixed.

LordBurrito, 2019-10-06:
Thanks, I'll try to contact them even if I don't think they could really help me.

Do you think downgrading the BIOS I could resolve the problem and would you recommend it to me?

pbatard, 2019-10-06:
A downgrade is unlikely to resolve your problem. If the issue is what I think it is, it probably has to do with a lack of maturity of the UEFI NTFS driver used by motherboard manufacturers, in which case you want a BIOS with a more up to date NTFS driver, rather than a more ancient one, and therefore one that might be even more problematic.

At any rate, I would encourage you to run the test I advised (copy the whole content on a single NTFS partition and see if you can boot from that), because if, as I suspect, you can't boot from such a media, then you have every right to vehemently complain to Asus support about their NTFS driver and ask them to fix it ASAP by indicating that it directly prevents you from installing Windows.

LordBurrito, 2019-10-06:
What do you mean with "copy the whole content on a single NTFS partition" ?

What do you think it's the best way to let them know about the problem ? Could I give them some information/log to really prove the issue ?

Thanks

pbatard, 2019-10-06:
1. Format your drive as non bootable with a single NTFS partition (you can do that with Rufus, just select _Non bootable_ under _Boot selection_ and make sure the file system is set to NTFS)
2. Using something like 7-zip open your Windows ISO and extract the whole content onto the NTFS  driver you just created.
3. Because it's UEFI and your Asus motherboard is supposed to be able to boot from NTFS in UEFI mode (the whole point of them adding an NTFS driver), you can just plug that drive and try to boot from it. You should at the very least see an UEFI option to boot from Partition 1.

If you do that and it doesn't boot, then you have demonstrated that their NTFS driver is __broken__, plain and simple.

You may also validate the SHA-1 of your `Win10_1903_V2_EnglishInternational_x64.iso` image to demonstrate to them that you are using a genuine unmodified Microsoft Windows 10 ISO and that therefore it's not an issue with the source media.

Do that, and you should have a very clear case with Asus support, that will be hard for them to dismiss as something external to their UEFI firmware.

LordBurrito, 2019-10-15:
Hi pbatard,

sorry to bother you again, I was wondering if you could explain why I'm able to boot Ubuntu ,on the same motherboard and same nvme, in UEFI mode.

pbatard, 2019-10-15:
I don't see why you shouldn't as long as you use FAT32, which I suppose is what you use to create your drive, since you didn't say otherwise. The problem is with using NTFS.

And again, the one test that can remotely try to pin this issue on Rufus/UEFI:NTFS rather than Asus is if you create a Windows bootable drive using a single NTFS partition, using the steps I highlighted above (i.e. not using Rufus) and you find that it boots. Did you run that test? Because it's when reporting that test to Asus support that you can irrevocably demonstrate to them, without them being able to try to push back the issue on some third party utility being used in the process, that their UEFI NTFS driver is broken and needs to be fixed.

pbatard, 2019-10-21:
Since I don't think there is much that can be added to what's already been pointed out, I will now close this issue.

closed 2019-10-21 by pbatard