pbatard/rufus #1385: How to dump raw image

narcisgarcia, 2019-10-07:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [xx] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [xx] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have an already hybrid .ISO image and don't want persistent space and no other adaptation feature. How can I let Rufus to make a simple byte-per-byte copy same as DD does?

Thank you.

pbatard, 2019-10-07:
If you want to dump an existing drive to an uncompressed DD image, you can do that in Rufus. See [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Saving_an_existing_drive_to_VHD) (which you should have seen if you read the FAQ) knowing that, for all intent and purposes, you can use an uncompressed VHD file in __exactly the same manner as you would use a DD `.img` file__, as the only difference is a 512-byte footer, which, I can guarantee, and even if you might think otherwise, will NEVER EVER get in the way of using a `.vhd` like a `.img`.

But really, if you _"have an already hybrid .ISO image"_, then Rufus will let you copy it byte-per-byte same as DD does. Heck (unless you are using Manjaro or PopOS where DD mode is selected by default, and therefore you will not be asked to choose a mode), as soon as you press start with an ISOHybrid image, Rufus __WILL__ produce a prompt that asks you explicitly if you want to write it in ISO or DD mode. Did you not see that prompt?

Can I please ask you to spend a little more time doing your research next time, before asking a question that you should have been able to answer on your own?

I will now close this issue.

closed 2019-10-07 by pbatard

narcisgarcia, 2019-10-07:
I was cancelling when being prompted about isolinux and/or another software to be downloaded because of the nature of .ISO image. I was assuming no DD mode was coming there.

Okay, I've tried now by accepting that other prompt, and found next prompt about ISO/DD modes.