pbatard/rufus #1387: DD mode for aarch64 Centos not work

lastguy, 2019-10-09:
A case titled "DD mode gone in version 3?" is locked so I create a new post here.
I downloaed arm centos from here: http://isoredirect.centos.org/altarch/7/isos/aarch64/

On version 3 there is no DD mode and above iso burnt cannot boot.
I also tried version 2.16, if I choose DD image then it report error for above iso (dd image)
Follow Centos readme, I use ubuntu 16.04 dd to write this iso into USB, and it boots up and I finished installation.

pbatard, 2019-10-09:
Next time please provide a log (yes, even if you're not able to create the drive &mdash; There's a very good reason why I make a point to ask for it, __ALWAYS__) and follow the checklist, coz you're definitely not getting any sympathy from me for thinking rules do not apply to you.

Also, I'm not seeing any issue with `CentOS-7-aarch64-NetInstall-1908.iso` which, in terms of boot functionality and detection, should behave __EXACTLY THE SAME__ as `CentOS-7-aarch64-Everything-1908.iso`. So, until proven otherwise (Have you checked the SHA-256, which, if you had elected not to ignore the checklist, you were supposed to validate?), I am going to assume that your download got corrupted or something when you transferred between Ubuntu and Windows. Or do you also see the same issue with NetInst?

Here's my log with NetInst, and I got the ISO vs DD prompt (NB: The `Timeout while waiting for logical drive` after writing the image is benign and simply indicates that Windows wasn't able to recognize a partition it knows how to handle):

```
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'CentOS 7 x86_64'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0159DA25
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/0x5bd8f633 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'CentOS 7 x86_64'
  Size: 610.3 MB (Projected)
  Note: File on disk is larger than reported ISO size by 256 KB...
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
Using image: CentOS-7-aarch64-NetInstall-1908.iso (552 MB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive8':
● Partition 1 (offset: 1048576, size: 29.2 GB)
Opened \\.\PhysicalDrive8 for exclusive write access
Writing Image...
Timeout while waiting for logical drive

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x75247844
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 8.5 MB (8912896 bytes)
  Start Sector: 121664, Boot: No
Local time: 13215121759
Next update check in 47100 seconds.
```

pbatard, 2019-10-09:
Well, now that I have downloaded the 8 GB "Everything" image, it looks like the CentOS people forgot to use the __established convention__ for indicating that an image is bootable by having the bytes at 0x1FE set to `0x55`, `0xAA`. So, as opposed to the NetInst image, which has the boot marker, the Everything doesn't which means Rufus will not propose the DD mode.

As far as I'm concerned, this is a CentOS problem. If they want to deviate from de-facto convention, with the result of creating problems for applications downstream (because if I start to allow any image to be written in DD mode with Rufus, a lot of users will have no idea why the non-bootable __random__  file they tried to use doesn't actually boot), so be it, but then their images will remain incompatible with Rufus' DD mode. Please understand that Rufus is not a meant to be a DD equivalent (it was certainly not designed that way) but a boot utility creator, so, unlike DD, it will not accept any random data you throw at it.

However, my take is that CentOS probably simply screwed up their "Everything" image without realizing it, because it makes very little sense that they would set the boot marker for NetInst and not for "Everything", so I'll leave you report that issue to them, so that they can fix it.

Closing.

closed 2019-10-09 by pbatard