pbatard/rufus #1388: Status bar will show using image even if i have selected non bootable

Sopor, 2019-10-09:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When opening an IMG file it will show the name of the loaded file in the status bar
Using image: file name.img
but if i now change the boot selection (it doesn't matter what i choose it will not change) to non bootable i can still see that the img file is loaded and this can be quite confusing. Now i think it will still install the IMG file but it will be non bootable.

As soon as i change boot selection the Using image should change until i select the image file again in the boot selection.

If i continue to use non bootable and press start Rufus will format the USB stick and when it is finished what will the status bar show, 1 device found. If the IMG was loaded it should still show the IMG file in the status bar but it does not.

Log
---
```
Rufus x86 v3.7.1576
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16021192704 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  'D:\Image\OPNsense-19.7-OpenSSL-vga-amd64.img\OPNsense-19.7-OpenSSL-vga-amd64.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: OPNsense-19.7-OpenSSL-vga-amd64.img (1.3 GB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive4':
● Partition 4 (offset: 1048576, size: 14.9 GB)
Opened \\.\PhysicalDrive4 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{50f6cb19-ea29-11e9-b403-049226d44339}\
Successfully remounted \\?\Volume{50f6cb19-ea29-11e9-b403-049226d44339}\ as G:

Found USB 2.0 device 'USB DISK Pro USB Device' (13FE:4123)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16021192704 bytes)
  Start Sector: 2048, Boot: No
```

Sopor, 2019-10-09:
Update:
It seems that when i select the IMG file in the boot selection it will not change to Using image: file name.img. I want to see what kind of file it has loaded so i know what file it will use.

pbatard, 2019-10-09:
I'm afraid this is by design. The Status bar is meant to be a shortcut to reporting stuff that happened, and not, as you seem to expect, be a reflection of the current status of the UI.

The status bar is meant to be updated when a new status message is issued. Therefore, as long as no operation had been enacted that gives rise to a new status message (and switching between boot selection is no such operation), the last status message will remain.

Please consider this: The status bar is really a __one-line mini log__. For the most part, when you change UI settings, the log does not get updated, and the same applies to the status bar.

> I want to see what kind of file it has loaded so i know what file it will use.

You can see that from the Boot selection drop down. When an image has been selected, the drop down is updated to display the name of that image.

I'm sorry, but it looks like you have a misconception of what the Status bar is really meant to be used for, and as such, I can not accept your request.

&rarr; Rejected.

closed 2019-10-09 by pbatard

Sopor, 2019-10-09:
Yes, that explains the behavior ...

pbatard, 2019-10-09:
Yeah. I can understand why you'd want the UI to behave in a specific way, but there's really so much I want to do in an application like Rufus, and I believe the use of a Status bar as a one-line mini log is what most application that have a status bar tend to use it for.

Plus, considering that there's also an Info field in ufus (on top of the progress bar) that, on the other hand, exists to reflect the current status, having the Status bar updated for what you wish for wouldn't make much sense. And no, I'm afraid I'm not planning to go out of my way to update the Info field to reflect whether image or non-bootable is selected either... :wink: