pbatard/rufus #1389: USB Drive not appearing in Disk List nor Disk Management after Rufus error mid-copy

Tactical-Myzzrym, 2019-10-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello,

I just bought a brand new USB Drive (64 gb) to use Rufus - Windows To Go on it. In the middle of the operation, Rufus had an error and unfortunately at that time I didn't think about snatching the log. Now, the USB just doesn't get recognized by my computer (Windows 10) anymore, here is what happens when I plug it in:
- It briefly appear in "My Computer" with a Device Letter (D)
- After around 15 seconds, the USB Drive disappears from "My Computer"

During that 15 sec period:
- If Rufus is open at that moment, it freezes completely (my computer still works fine, just Rufus freezes)
- If I try to open Rufus, Rufus doesn't launch until the USB Drive disappears from "My Computer"
- If I try to write Disk List in Diskpart, it freezes until the end of the period (and doesn't show the USB drive)
- If I try to write chkdsk d:/f, it freezes until the end of the period (and says "Cannot open volume for direct access" afterwards)

Other information:
- The USB Drive is not visible either in Diskpart => Disklist nor in Disk Management
- It does appear in the Device Manager as "Unknown USB Device" (uninstalling then restarting the PC doesn't solve the issue)
- This behavior is consistent regardless of which USB port I put it in.
- Other USB Drives work, and the Key Assigned by Windows is consistent (if I put a valid USB Drive it gets assigned the letter D and works fine, plugging the faulty USB Drive after assigns the letter F (E is already used) but it has the same issue as described above).

Is there any way to fix this USB Drive?

Many thanks!

Log
---
```
<FULL LOG>
```

pbatard, 2019-10-12:
Looks like the flash memory of your drive failed. Please see if you can access the drive with a different OS such a Linux, but otherwise, it's probably time to replace that drive.

If the drive is no longer visible with Windows native tools, it's most likely dead.

Tactical-Myzzrym, 2019-10-12:
Darn, I just received it today. Guess Rufus isn't as safe as I hoped for :(

pbatard, 2019-10-12:
I'm afraid it's not Rufus that's the problem. It's the reliability of flash memory (which is a lot less durable and sturdy than people tend to think it is).

It doesn't matter what application you use, if you're unlucky with your flash memory, your drive will die even if all you're doing is using File Explorer to copy a bunch of files, which pretty much exactly the same as what Rufus was doing when your drive died.

See also https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive

I will now close this issue.

closed 2019-10-12 by pbatard