pbatard/rufus #1393: RUFUS 3.8 and older versions dont want to show all my san disk USB devices

sgdp, 2019-10-19:
**WHAT IS THE PROBLEM?**

_I have different USB devices. 16 GB / 32GB / 64GB, but i cant pick one from the drop down.. yeah and all that crappy tricks with alt+z or uncheck all settings etc.. nope.. going down to older versions.. down to 2.x dont help.. :( PLS HELP ME!_

**WHAT I DO?**

_If i replug the USB Device following msg:
1. found non-USB removable device 'SanDisk Cruzer Facet SCSI Disk Device' => Eliminated 
2. 0 devices found_

**SOME DUMB QUESTIONS, PLS DONT ASK**
Do you restart PC and turn it off and on? Do you tried on different PC? Do you tried with System Admin? Do you turned Virus Detection off? Do you checked Firewall? Do you have ports blocked? System have all Updates? Do you tried win10? Do you tried to use USB2.x or 3.x directly on Backplate of MB? Do you checked DeviceManager? Do you checked formatting it? Do you tried to...

**LOG of one device:**

_Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
0 devices found
Found non-USB removable device 'SanDisk Cruzer Facet SCSI Disk Device' => Eliminated
0 devices found
Found non-USB removable device 'SanDisk Cruzer Facet SCSI Disk Device' => Eliminated
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
0 devices found
0 devices found
Found non-USB removable device 'SanDisk Cruzer Facet SCSI Disk Device' => Eliminated
0 devices found_

pbatard, 2019-10-20:
Well, I'm going to start by asking you the "dumb" questions, because I think it is very relevant to your issue, and, strangely, you chose to __WILLINGLY DISREGARD__ the checklist that contains said question, which won't bring you nay favour here (the checklist exists for a reason &mdash; If you think you are above following it, don't be surprised if help doesn't arrive fast... or never at all).

Did you actually try on a different Windows PC?

From what I can see, the most logical explanation, since your USB device is not detected as a regular USB device (hence your issue) is that you have a special driver installed on your system to access these specific drives (it may be a SanDisk specific drive, or it may be something like one of those various motherboards' _"We'll speed up your USB 3.0 drives by using a non USB conformant driver if you let us"_ drivers that these manufacturers like to use).

At any rate, you are using Windows 7, which, last time I checked, didn't have native support for USB 3.0, and the fact that you are talking about USB 3.0 leads me to think that your device is being accessed from a USB 3.0, which means __CUSTOM USB 3.0 DRIVER__, which means the possibility of non-compliance with what's expected from a proper USB 3.0 driver. So, yes, if you have a USB 2.0 port, it's not "dumb" to ask whether your device is properly detected there (or through a USB 2.0 extension cable, which will force Windows to use its native USB 2.0 drivers to access your device). And it's also not "dumb" to ask you to validate, as I strongly suspect, that you don't have that issue on a vanilla Windows 10 computer, since it provided native USB 3.0 drivers.

But more importantly:
* Press <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus with your device plugged in. This will enable enumeration debug.
* Copy paste the new log as it will contain insightful information about the driver that is actually used.

pbatard, 2019-10-24:
Any chance of getting the <kbd>Alt</kbd>-<kbd>.</kbd> log? if not, I'm going to have to close this issue.

closed 2019-11-01 by pbatard