pbatard/rufus #1394: Rufus 3.8.1580 USB killed?

Fsadju, 2019-10-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Plugging the USB into a different port.

Issue description
-----------------
Last night installed HBCD.iso from "hirensbootcd.org" with Rufus , that writing task was successful (unfortunately i did not expect that error occurs following morning so no logs for that).
Over night i did left HBCD run on my laptop to wipe laptop hard disk completely ("0" type wipe).
While on desktop i downloaded last windows 10 .iso file from it's provider.
Today in the morning task wiping was completed on laptop so i turned off everything from laptop and backed to desktop with usb. So then i connect normal ether high speed usb port device storage won't appear , "My Computer" doesn't show partition (with letter). I ignored that missing partition of usb and run Rufus anyway.

Things i have tried despite checklist:
[*] Format device trough various tools.
as result , occurs errors such as "writing-protected" or "check windows logs for more information".

To understand more i could attach photos - if needed.

What i could do with this unexpected error ?

I don't understand why it happen.

Tho device storage is accessible trough Free-ware such as 

Am writing here as issue happen with Rufus.

Log
---
```
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x84
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1893, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {F2542726-0E65-4833-A9D7-9BD93F91BD25}
Max parts: 128, Start Offset: 17408, Usable = 15571319296 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Main Data Partition'
  ID: {774FEB55-9F71-4B76-96F8-98B22A51FBE9}
  Size: 14.5 GB (15570288128 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 3.9 GB (Projected)
  Note: File on disk is larger than reported ISO size by 556 KB...
  Uses: EFI
[... 299 lines cut ...]
Block 241: write error
Block 242: write error
Block 243: write error
Block 244: write error
Block 245: write error
Block 246: write error
Block 247: write error
Block 248: write error
Block 249: write error
Block 250: write error
Block 251: write error
Block 252: write error
Block 253: write error
Block 254: write error
Block 255: write error
Too many bad blocks, aborting test
Check completed: 256 bad block(s) found

  0 read error(s)

  256 write error(s)

  0 corruption error(s)Rufus bad blocks check ended on: 2019.10.21 13:32:22

```

pbatard, 2019-10-21:
The bad blocks check indicates that your device is defective. And yes, things that used to be working do happen to fail, especially when it comes to flash memory, which is __A LOT__ less everlasting that people tend to think to is.

If the Rufus bad blocks check report errors, then your USB flash drive is defective, period. It's a pure hardware issue, and no matter of wishful thinking that, maybe, there's something that Rufus could do, will detract from the fact that your hardware is failing and should no longer be used.

Please use a different flash drive, and either recycle this one or send it for RMA, because your flash memory __is__ defective. Flash memory going bad happens, all the time. And it is not something software can fix.

I will now close this issue.

closed 2019-10-21 by pbatard

Fsadju, 2019-10-21:
Yes, thank you for replay. Indeed. Will keep that in mind.
Anyway , checking trough friends laptop that USB working properly. Tried run other USB's on My Desktop still Failing to load ANY USB Device in fact. Both devices laptop and desktop operation system is failing/lagging that's why i must to reinstall 'em all . Awkward situation .  hahah . 

,- In the end .Thanks for this freeware !