pbatard/rufus #1396: The ext3 file system in the partition for persistence has errors

sudodus, 2019-10-24:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.

        $ sudo badblocks -sv /dev/sdc 
        Kontrollerar block 0 till 58615703
        Letar efter dåliga block (skrivskyddad test): klar                                                 
        Pass avslutat, 0 dåliga block hittade. (0/0/0 fel)

- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.
*I tested the checksum of the iso file separately at/after downloading*

        $ md5sum lubuntu-19.10-desktop-amd64.iso 
        8b8b03366298e7489df6bc34760f89aa  lubuntu-19.10-desktop-amd64.iso

    Actually a got it via zsync, which has a built-in checksum test.

Issue description
-----------------
When creating persistent a live drive Rufus makes an ext3 file system in the partition for persistence. This file system has errors. It is still possible to use the file system for example with
Lubuntu 19.10, but

1. Part of the drive space in the file system is lost, wasted, compared to a healthy file system, when Lubuntu or other flavours of Ubuntu is booted. This might not be seen when connected to an operating system that is already running. The size of the lost drive space is small in a 4 GB pendrive, significantly bigger in a 16 GB pendrive, and huge ina 60 GB SSD. I used an SSD in several test runs because the loss is so big, that it very easy to see, and also because it is a high speed device, so that I could do several tests within a rather short time.

2. The errors in the file system may not create any corruption of files in the beginning, but I am afraid, that it may cause severe problems later on. When booting live-only I noticed that a log directory was mounted (and some of the errors are that some inodes have 2 links).

I tested various things in three different computers and another user tested it in a fourth computer. I ran standard Windows 10 fully up to date in two computers and a 'Windows Insider' version 10 fully up to date in the third computer. After some repetitions things started to look good, but when I specified the starting conditions in a strict way I could reproduce similar (not exactly the same) size of lost drive space.

1. Original drive a standard storage device with FAT32
2. Oriignal drive zeroised (completely overwritten with zero bytes)

In these cases between 20 GiB and 40 GiB were lost, already marked as used by df -h when booting into the persistent live system for the first time. A single test starting from exFAT showed 12 GiB lost drive space.

Thinking aloud
---
It seems to me that the tool to create the file system is buggy, or is using some buggy or not compatible system software (dll-file?).Or maybe you could fix it by compiling with some flags for more strict, less optimized for speed operation. Or you could test if there is a version for ext2 or ext4, that is more robust. I think that ext4 gets much more attention than ext3 nowadays.

Repair
---
I tested that it was possible to repair the file system with e2fsck -f and then there was no lost drive space, only the normal amount reserved to management (and similar size as that of the linux tool mkusb).

Re-format
---
I tested also that it was possible to re-format the ext3 file system with AOMEI Parition Assistent (the freeware version), and that file system was working almost correctly, there was a "Resize inode not valid" message. But no other complaint, not the huge amount of errors caused by Rufus, and there was no obvious loss of drive space for the user.

I wish that you will fix this bug or find a way around it soon, and I am willing to help with testing, when you have new versions of Rufus or simply new ways to set the conditions to make it work.

Good luck
Nio

Log
---
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x041D (sv-SE)
Will use default UI locale 0x041D
[... 710 lines cut ...]
Extracting: E:\README.diskdefines (226 bytes)
Extracting: E:\ubuntu (0 bytes)
  Ignoring Rock Ridge symbolic link to '.'
Created: E:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found UAS (USB 3.0) device 'OCZ-AGIL ITY3 SCSI Disk Device' (174C:55AA)
Using autorun.inf label for drive E: 'Lubuntu 19.10 amd64'
Found non-USB removable device 'LITEONIT LMT-128M6M mSATA 128GB' => Eliminated
1 device found
Disk type: FIXED, Disk size: 60 GB, Sector size: 512 bytes
Cylinders: 7297, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000592E1
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.9 GB (2039385600 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 54.0 GB (57982030848 bytes)
  Start Sector: 3985223, Boot: No

pbatard, 2019-10-24:
Thanks for logging an issue.

As I explained in the e-mail I sent earlier, the "tool" we use to create the `ext` file system is the exact same code as Linux uses which comes from [e2fsprogs](https://git.kernel.org/pub/scm/fs/ext2/e2fsprogs.git/tree/lib/ext2fs) and which is maintainer by Theodore Ts'o, who is the main developer for the ext file systems in Linux. So, in terms of "tool", we could hardly be closer to what Linux uses for ext file system creation as we already are. Especially it should be pointed out that, because e2fsprogs has some Windows support, the core of the code was barely modified for our usage. As a matter of fact, if you clone the repository I linked to above and compare it to the [code used by Rufus](https://github.com/pbatard/rufus/tree/master/src/ext2fs), you'll see that the only differences are for the backend. So really, for all intent and purposes, the code used by Rufus and the code used by the native Linux tools to create an ext file system is virtually identical and comes straight from same people.

Also, as opposed to what many other utilities do, and since Rufus doesn't have an installer in the first place anyway, we don't use external DLLs apart from the system ones (but of course, Windows does not have a system DLL that can be used to format a drive to `ext`). As a result, anything we reuse is pretty much the same as the official code, which we compile within the application itself. So I hope this will finish to convince you that, no, the issue is not the result of some "lazy" reuse of a dodgy DLL. Instead we spent a lot of effort making sure that we used the code that is closest to what Linux uses. Plus, you have to be very conscious that because Rufus is GPLv3, we can't really go around and reuse elements from close source utilities, like AOMEI Parition Assistant, which, last time I checked wasn't Open Source at all (please be mindful of confusing "freeware" with "Open Source", as it's not because something is gratis that it is actually free in the sense of software freedom).

Therefore, our expectation has been that, considering that the only code that we changed was for the I/O backend, and that this I/O backend doesn't appear to have much of an issue when we're writing root inodes and other stuff, the various ext2fs initialization calls we make, which again are supposed to perform the exact same operations as they do on Linux, might result in a file system that is properly initialized. If anything, this would lead us to think that the issue might be with the official code, whose Windows side probably hasn't received as much love as the Linux side, which might explain why the created file system trips e2fsck.

Which means that we now have to analyse the internals of ext2fs file system creation, to figure out why code that, for all intent and purposes, we do expect to behave in the same manner regardless of the platform, might not, and this is going to take a while since we have to familiarize ourselves with exactly the kind of stuff we were trying to avoid when we chose to go with "the reference" for formatting an `ext` file system (i.e. getting super-knowledgeable about ext2/ext3 intrinsics).

Realistically then, unless someone helps us with that (for instance by providing some reports of _"Rufus writes data X into inode Y at position Z whereas it should write data X' into inode Y' at position Z'"_, or even better, explaining precisely what we might be missing in the code to make our sequence of calls to what should be the same functions as `mkfs.ext3` calls on Linux work), I expect that troubleshooting the issue is likely to take a couple of months at best, because figuring out what exactly might be wrong with ext bitmaps or inodes __is__ not something you'll achieve with a shortcut.

Also, as I mentioned elsewhere, whereas I did indeed see `e2fsck` errors (mostly `Block bitmap differences` on pass 5), I didn't really consider them a major issue considering that we have yet to receive a single report of effective data issues for ext partitions created with Rufus and I've seen some reports from Theodore Ts'o advising people [with similar errors](https://www.redhat.com/archives/ext3-users/2008-October/msg00024.html) that _"block allocation bitmap differences and free block/inode accounting information being wrong is normal"_ under some circumstances and that it is _"unlikely (to be) anything serious"_ [3].

All this to say that, whereas we will be investigating the issue to see if we can fix the e2fsck results, it's unlikely to happen as soon as you wish it will and that if some Linux folks, with knowledge of the ext2/ext3 intrinsics, want to shed some technical light as to what exactly e2fsck seems to complain about, and what we might be missing from our [ext initialization](https://github.com/pbatard/rufus/blob/7ec7331200c047e920590cf65a2b1dee3162213a/src/format.c#L843-L1106) (which, it should be pointed out, is not directly derived from `mke2fs` since, whereas the parts of the e2fsprogs code we reuse are licensed under LGPL, which allows us to include them into a GPLv3 application, unless told otherwise, `mke2fs` itself is licensed under GPLv2 __only__, which is not compatible with GPLv3, so we did have to take some guesswork about what needs to be called during ext init) we would appreciate it.

sudodus, 2019-10-25:
Have you seen that e2fsprog is part of Cygwin? It might help in debugging (to check if it is subject to the same bug, and in general how it behaves. I guess Cygwin is FOSS (freer than AOMEI's tool).

https://cygwin.com/cgi-bin2/package-cat.cgi?file=x86%2Fe2fsprogs%2Fe2fsprogs-1.42.12-1&grep=hat

pbatard, 2019-10-25:
Yes, that's where the Windows support comes from, coz it sure isn't compatible with MSVC and MinGW. My worry however is that cygwin is doing so much to fool an application into believing that it runs on Linux (that's basically the whole point) that even the issue doesn't manifest itself there it won't tell much of anything except that code that was mostly designed for Linux like environments appears to work well in Linux-like environment... Oh, and if your idea is that Rufus could/should use cygwin behind the scenes, I'm going to have to disappoint you because any application that relies on cygwin has to rely on a (rather large) `cygwin1.dll` (which basically does the Windows abstraction to a Linux/Posix-like), which is a nonstarter for an application that both doesn't come with an installer and is very size conscious (i.e. I'm not going to double or triple the size of Rufus just so that I can take a shortcut in not identifying what exactly might be wrong when using MSVC/MinGW, which remains the proper way to try to address the problem).

sudodus, 2019-10-26:
I understand, that you don't want to rely on any dll, and I think you have good reasons for that. As I said, it might help in debugging.

sudodus, 2019-10-26:
Debugging results
---
I installed Cygwin and e2fsprogs and fdisk into it.

Then I could

1. Repair the ext3 file system created by Rufus using `e2fsck` so that it worked in the persistent live drive. This process was rather slow.

2. Overwrite the ext3 file system created by Rufus using `mkfs.ext4` and it worked without errors in the persistent live drive and as checked afterwards with `e2fsck` in my main linux computer. This process was faster.

So it seems to me that Cygwin can do it. It means that the source code is healthy, and you should be able to use it successfully. Maybe the compiler options/flags used by Cygwin are available (in a makefile or similar). Maybe there is file system corruption because the write operations are not flushed completely, and you should `sync` and allow some time before and after creating the `ext` file system and before finishing Rufus.

pbatard, 2019-10-26:
> Maybe there is file system corruption because the write operations are not flushed completely, and you should sync and allow some time before and after creating the ext file system and before finishing Rufus.

I appreciate the help but I would also appreciate if you weren't trying so hard to clutch at straws.
 
Fixing issues is not "guesswork". It is deducing the cause from properly analysed effects, which still hasn't been done here. And for the record, we are [opening the partition with `FILE_SYNCHRONOUS_IO_NONALERT`](https://github.com/pbatard/rufus/blob/master/src/ext2fs/nt_io.c#L211-L213) precisely so that having to flush the drive should not be needed. Also, if it was a flushing problem, we'd see a lot more inconsistencies than we are seeing in the `e2fsck` report (after all, leaving the drive plugged in for a while should make the errors go away), so it is exceedingly unlikely that this has to do with flushing the drive.

Also, a much more relevant test, since of course you've been using `mkfs`, which, as I explained above, we cannot use as is due to licensing, would be to port [this code](https://github.com/pbatard/rufus/blob/master/src/format.c#L843-L1106) to cygwin and use that to format a drive, which is something I might do if I can't identify the issue through other means (because what I am not short of are ideas on how I can proceed to troubleshoot the issue. What I am short of is time).

Also, please be mindful that we're using `ext3`, not `ext4`. I am not planning to add `ext4` support at this stage because this requires a lot of extra work and additional code, which doesn't seem very relevant to add for a persistent partition. As a matter of fact, my original plan for Rufus and persistence was to go for `ext2` only, and a much better analysis (since it'll avoid a lot of the pitfalls that come from adding extra complexity) is to compare what Rufus does for ext2 with what Linux does for ext2 only, and fix that if needed, before moving to journalled. So that too is part of my current plan, which I will get too when I have a chance to do, but which will always be much slower than you'd like due to prioritization and time constraints.

At the moment, one of my first planned tests is going to be this: Create a virtual image in Linux (e.g. 32 MB) set with `0xFF` bytes (to make it easier to see stuff that's init'd to zero) and format it to ext2 in Linux. Then do the same in Rufus and compare the inode and bitmap data, to see how they differ. Then, from that data, figure out the parts of the code used by Rufus that may introduce differences.

If you really want to help, you can actually carry out that test yourself, since, if you have a Windows platform, you can install VS2019 to run Rufus & debug Rufus, and I already have [code to deal with the formatting of a 32 MB image in Rufus](https://github.com/pbatard/rufus/blob/master/src/format.c#L866-L881) which all you have to do to use is uncomment the define for `RUFUS_TEST` at the top of `rufus.h`.

sudodus, 2019-10-26:
1. I think it is OK with `ext2`, at least temporarily, if you think it is much easier to make it work than to make `ext3` work.

2. I was only showing (with the most advanced case `ext4`) that the code works as it is implemented by Cygwin in Windows. I think it would have been worse, if Cygwin produced the same or similar errors as your implementation.

3. I have to think twice about taking the step into compiling and debugging the source code of Rufus.

pbatard, 2019-10-26:
> 1. I think it is OK with `ext2`, at least temporarily, if you think it is much easier to make it work than to make `ext3` work.

That's not exactly what I'm saying. What I am saying is that, to properly troubleshoot this issue it is much better to start by figuring out what needs to be done for the lowest common denominator (`ext2`) and then work our way up, than start with the more complex versions of the filesystem, and either have to figure out everything at once or sort out the elements that are irrelevant.

I am not planning to drop `ext3` support from Rufus. I'm just saying that it will be easier to figure out what the problem is if we analyse the e2fsck issues for `ext2` instead of `ext3` when trying to troubleshoot this problem.

> 2. I was only showing (with the most advanced case `ext4`) that the code works as it is implemented by Cygwin in Windows.

Well, if the issue is the inode/bitmap layout (which is what e2fsck seems to report) then I don't think testing `ext4` features tells us much, since my understanding is that `ext4` doesn't fundamentally exert that aspect further compared to ext2. 

> I think it would have been worse, if Cygwin produced the same or similar errors as your implementation.

I would have been very surprised if cygwin produced an issue, as, if e2fsprogs does have a Windows implementation, and that one implementation is for cygwin, I'd expect there are quite a few people testing it for errors, as opposed to my MSVC/MinGW, which is something that I had to add myself for Rufus usage (and that I had to implement in a semi-blind manner because only part of the e2fsprogs source is compatible with Rufus). I'm still kinda hoping the issue is a simple missing call during all the various steps that are required to set up an `ext` file system (as you can see from the code I linked to, there are quite a few that need to be leveraged), but I'm not exactly counting on the fix being that simple, as I'd expect very different results from e2fsck then...

It might be interesting to know if the e2fsprogs folks have a test suite that they use to validate file system creation (I haven't really looked into that), as opposed to create the file system and see if e2fsck complains...

> 3. I have to think twice about taking the step into compiling and debugging the source code of Rufus.

Which is __not__ something I am requesting you to do. I am planning to do that when I have a chance. If you do have time to spare and want to help (and don't want to suffer the annoyance of having to contend with Windows), what I'd be more interested into is a technical explanation of what the e2fsck errors mean. For instance: _"If e2fsck reports error X then it means that element Y, which is used to provide the file system data about <some specific thing>, is not set to its expected value Z"_.

That's actually one of the other avenues of what I'm planning to do: look at the e2fsck implementation to understand in details what it's really complaining about, as if we properly understand this data, it may become a lot easier to tell _"Here, this is what's wrong!"_

MatthewForrester, 2019-11-06:
In the FAQ, are urged to "remind distro maintainers that an open issue is affecting them too, rather than assume that things will get fixed on their own". So although I am basically a satisfied 'customer' of Rufus, I would like to report how this issue may be affecting me. 

When Ubuntu 19.10 came out, I used Rufus to create a persistent live USB. I got a workable system with Firefox and LibreOffice. However, even a 32GB USB stick did not have enough space for Steam (which should need <2 GB). I have [described the situation in more detail in an Askubuntu question](https://askubuntu.com/questions/1181998/how-do-i-create-enough-space-for-steam-on-a-persistent-live-usb/1182005). Something seemed to be consuming huge amounts of space in the persistent partition. After posting on Askubuntu, I tried maxing out the persistent partition (same problem reoccurred) and creating the persistent 18.04LTS system recommended by Steam (triggering the expected casper bug). I have some screenshots showing oddities in disk utilization, but none of the low-level information requested above by @pbatard 

This week I tried [the answer recommended by @sudodus ](https://askubuntu.com/a/1182005/1007001). By using mkusb+dus, I have been able to get persistent 18.04LTS on the same USB stick as before, and to successfully install Steam.

Although this thread suggests that the two of you (@pbatard  and @sudodus ) have had no previous contact, I have relied on both of your tools in order to get from Windows 7 to a working live Ubuntu USB, and I am grateful to both of you for the time that you have donated to your projects. Thank you!

In my opinion, Rufus is much easier to use for users coming from an English-speaking Windows background. But at the moment, it seems only mkusb is reliable. I wish both of you the best of luck in finding ways to work (together or separately) for the benefit of Linux users.

To finish, I would like an 'obiter dictum' on the FAQ comment chiding users for not reporting the Ubuntu casper bug. I guess that poor reporting will always be a particular headache for Rufus, and mkusb, because many of their users will also be encountering Ubuntu and/or Linux for the first time. In my case, I only started using Ubuntu/Linux as a sysadmin a few months ago, so my default assumption is that "I'm doing it wrong", rather than "I've encountered a bug"!

sudodus, 2019-11-06:
I hope that all of us can co-operate for the benefit of Linux users :-)

Do you remember Alexander the great and the Gordian knot?

https://en.wikipedia.org/wiki/Gordian_Knot

I suggest that you let an experimental Rufus create a partition just big enough to extract everything from the iso file, create the BIOS bootloader and add the boot option 'persistent'. Then skip creating a partition behind it (no partition, no ext3 file system) and let Ubuntu 19.10 create it automatically. It will work rather well, but maybe not perfect yet. You can see details about it at the following link,

https://help.ubuntu.com/community/Installation/iso2usb/diy

and particularly

https://help.ubuntu.com/community/Installation/iso2usb/diy#Let_Ubuntu_create_the_partition_for_persistence_automatically

If we find that it works well enough, this can be an option in Rufus (for Ubuntu 19.10 and later versions) alongside the current options.

pbatard, 2019-11-06:
>  I would like to report how this issue may be affecting me.

Thanks. User reports like yours are an essential way to gauge the severity of an issue, as it provides additional elements to try to replicate an issue and come up with something that can be replicated for troubleshooting.

I'll start by restating that my main limiting factor to try to address the problem here is time. Unfortunately, even as it may look so externally, a file system intrinsic issue like this one is unlikely to be resolved in a _"Oh, it behaves like this? Then I know exactly what section of the code needs to be patched and this shouldn't take more than 5 minutes to fix"_.

Instead, what's likely to need to happen is something closer to _"In order to fix this problem, I need to understand how `ext#` inodes and bitmaps are created, where exactly they should be created and what content they need to hold. And I also need to understand, at the technical level, how the various elements of the file system are linked together. __Then__ I can look at the outcome from different tests and try to devise what might actually be happening behind the scenes in order to fix it"_. Right now, I expect this to be a multiweek effort, which means I need to allocate enough time for it, which is the prime reason why I haven't started to look into it in earnest. Oh and sorry @sudodus  but there is simply no way that, after I already spent many months adding support for ext formatting in Rufus, I'm going to drop the whole feature and tell users that they should just format the casper-rw partition in linux. First of all this is super inconvenient and second of all this type of quick and dirty workarounds does not align with what I strive for with Rufus. Plus this would require modifying the UI to add new elements which equates to waiting at least 2 months to get everything translated so that I can actually put a release out. So instead of speeding up the process it is much more likely to delay it... And yes, I do understand that it is frustrating to see your pet issue not being worked on, but that's just how real-life is. There are just too few developers for too much work, which forces prioritization and delays. But that does not mean that I do not consider this issue as an important matter that I want to see fixed ASAP. It's just the ASAP that's the root of the matter.

Now, this being said, I tried to run a few tests similar to what @MatthewForrester described (32 GB drive with a 17 GB partition) and I can confirm the problem with space allocation. Curiously, during my first test, everything seemed to be good, even after copying the Ubuntu 19.10 ISO to the casper partition (in order to see how the amount of used/free space would be reported), until I started to copy a 5 GB Windows ISO and found that I ran out of space. Then, after running the same thing a second time I found that the used space jumped way beyond what was expected after copying the Ubuntu ISO. So this would seem to indicate that the file system is pointing to inodes that haven't been set properly, which can either mean we're creating the inodes properly, but then not properly setting the table or whatever ext# uses as a means of pointing to inodes, or we are setting that table properly, but we are missing an inode initialization step so that Linux doesn't attempt to read garbage data from those inodes. Or it could be an issue with the bitmap tables (which are used in what manner, I don't know yet) if this is what's being used to report used/free space.

At any rate, as I pointed above, there's no way around getting familiar to what exactly is going on behind the scenes, by, for instance, comparing a sane file system with the one created by Rufus, to try to understand where exactly the problem lies, and __then__ try to venture a fix. And there's no shortcut to accomplishing that besides devoting it the required time.

sudodus, 2019-11-06:
"... tell users that they should just format the casper-rw partition in linux. First of all this is super inconvenient ..."

I'm sorry, but you did not read the link. **Ubuntu 19.10 creates the partition and file system automatically** the first time that the [persistent] live system is booted. It is not inconvenient, only a little delay (for this action to run).

pbatard, 2019-11-06:
Ah, I heard the Debian maintainers talk about adding this but I didn't realize that they had done it for 19.10. Still, it's going to be too confusing for users to handle things differently or even subvert the expectation that, if you set a persistent partition in Rufus, then it should be formatted before you boot the OS. Plus my other worry is that the bugfix will be retrofitted to 18.04, but not the automated partition creation (or other Ubuntu-based distros might not follow with the automated formatting), in which case we're going to have another slew of issues.

I'm still going to point out that what you are proposing is a workaround, whereas the proper way to address an issue is to fix it. If I had stated that I'm not planning to look into fixing this, it would make sense to try to go with a workaround. But not when my plan is and continues to be to try to address the issue when I have enough time to do so.

MatthewForrester, 2019-11-07:
Thank you for taking the time to respond to my post.

_> I'll start by restating that my main limiting factor to try to address the problem here is time. Unfortunately, even as it may look so externally, a file system intrinsic issue like this one is unlikely to be resolved in a "Oh, it behaves like this? Then I know exactly what section of the code needs to be patched and this shouldn't take more than 5 minutes to fix"._

As I know nothing about filesystems, I completely accept that a solution will take time. I have made some very minor contributions to another open-source community and I know that my heart sinks when I realize that I'm going to have get my head around yet another subsystem in order to move ahead with the main project. I guess you must have a similar feeling.

Just a thought about how to manage this. As an interim step, would it perhaps be worth updating the FAQ so that Rufus users know that there is a problem? I may try to do this myself, but I won't be offended if you roll back since 'a little knowledge is a dangerous thing' and my text might be counterproductive.

pbatard, 2019-11-07:
Nah, what I'm planning to do is add the __EXPERIMENTAL__ tag back to the feature in the ChangeLog because that's where you want to advertise that a new feature may still have bugs __before__ people try to use it.

It actually used to be flagged __EXPERIMENTAL__ until the last release, since it was still too new to declare as stable, so I'm just going to reinstate it as such for a while.

MatthewForrester, 2019-11-08:
OK, I won't touch the FAQ then. Thank you for your software and best of luck.

sudodus, 2019-11-08:
> OK, I won't touch the FAQ then. Thank you for your software and best of luck.

:+1:

pbatard, 2020-02-09:
Just a small update for those who must be despairing to see progress on this. First of all to tell you that I am working behind the scenes to try to fix this issue, but, as indicated above, it does require time.

I can however provide a few elements of what I've found so far:
* One of the `e2fsck` warnings seems to be due to having the `resize_inode` feature enabled. Once you remove that feature, the warning no longer appears.
* The issue only seems to manifest itself for partitions that are larger than 4 GB, and it looks like the problematic inodes are all located above 4 GB on the partition. So there's a possibility we have a 32-bit variable somewhere that's overflowing, though we're mostly using block indexes rather than byte indexes, so with 1024 or 4096-byte blocks, we should have some margin at the block level before we see an overflow. Besides, I tried to make sure to use 64-bit wherever I could (e.g. I switched from `blk_t` to `blk64_t` in parts of the code). Then again, the current code (i.e. the stuff that I didn't write) does generate a lot of warnings about arithmetic overflows from using operators on a 32-bit value and then casting the result to 64-bit value. Those are usually benign, so most codebases ignore them (with the right warning level, gcc on Linux will produce those as well) but maybe there's are instances somewhere where these warnings are relevant.
* I think I probably enabled or forgot to enable an `ext` option/feature that makes the formatting much slower than it could be, so I'll be trying to identify that as well.

Once again, I have no idea when I'll get a chance to dive more deeply into these, but, despite what it may look like, progress is being made on this issue, albeit slowly.

marcosfrm, 2020-02-10:
With EXT4, `uninit_bg` (superseded by `metadata_csum` in recent e2fsprogs/kernel) will make huge difference in formatting time on large devices.

FSArchiver has some logic for this:

https://github.com/fdupoux/fsarchiver/blob/master/src/fs_ext2.c

pbatard, 2020-02-10:
I appreciate the help, but I'm not using `ext4` at all in Rufus and I currently have no plans to add that feature (too much complexity for too little benefit).

Also the code you pointed to is GPLv2 __ONLY__ (not GPLv2 or later) which means that, even if I wanted, I couldn't use any of it in Rufus (GPLv3 or later)... which actually is probably part of the reason we're having trouble with `ext2`/`ext3` formatting, as `mke2fs` is GPLv2 only, so I couldn't reuse that code for formatting (but the `ext2fs` library parts are LGPLv2. which makes them usable in a GPLv3 project).

pbatard, 2020-02-12:
Okay, I think I have finally pinpointed the issue.

The issue has to do with [this specific line of code](https://github.com/pbatard/rufus/blob/9edd7492db84a5f3b8cf5d71fee98cf51882043f/src/ext2fs/nt_io.c#L700):
```C
offset.QuadPart = block * channel->block_size + nt_data->offset;
```

As I mentioned above, it has to do with a 32-bit overflow when computing values, and more specifically with __DUMB__ compilers not performing 64-bit computations when they really should.

You see, in the line above, `offset.QuadPart` and `nt_data->offset` are 64-bit values, and, since compilers are allegedly supposed to do the smart thing when performing calculation, one would expect that, even as `block` and `channel->block_size` are 32-bit values, the compiler will be smart enough to compute and move the product of these 32-bit values into 64-bit registers (because, programmatically, there's absolutely no logical reason whatsoever to want to limit that multiplication to 32-bit when the result is going to end up in a 64-bit variable).

Alas, this is exactly what's happening here. And it looks like both MSVC and gcc suffer from that illogical behaviour, because, unless you specifically tell the compiler that you do want to compute `block * channel->block_size` as 64-bit (by adding an explicit cast like `((uint64_t)block) * channel->block_size`), that darn multiplication is performed as 32-bit, which means that, as soon as you try to write to a block of data that resides above 4 GB on the media, you're writing to the wrong address. Hence `e2fsck` finding loads of issues for any inode above 4 GB, since none of them have been initialized... Ouch!

Damn, and I was always told that trying to micro-manage or micro-optimise what compilers do was a waste of time because they are supposed to be designed to do the smart thing. So much for that... Heck, it almost makes one wish we were still programming in pure assembly, because such an issue would certainly not happen then (since we'd very explicitly specify that we are working on 64-bit register for that multiplication)...

Oh and for those who wonder why we're not using a 64-bit block variable in the first place, or more specifically the 64-bit version of the  `write_blk()` call, the reason is that, with a block size of 4096 bytes or more for large volumes (and code that explicitly checks for 32-bit overflow for the `block` variable), you're going to have to create a 16 TB or larger partition before you run into the 32-bit limit, which seems a bit unrealistic __for a persistent partition__. Plus the underlying `ext2fs` code I lifted still has tons of internal calls to the 32-bit version of `write_blk()`, so it didn't make much sense to bother with the 64-bit version... __as long as the `block * block_size` computation was properly carried out using 64 bits__. Now that I've seen the result of trusting compilers (and/or pre-existing code) to do the right thing, I guess I should have taken a page from what `unix_io.c` does, moved the core write block function to 64 bit and just added 32 &rarr; 64 bit invocation for the 32-bit calls like so:
```C
errcode_t unix_write_blk(io_channel channel, unsigned long block,
				int count, const void *buf)
{
	return unix_write_blk64(channel, block, count, buf);
}
```

Now, as to the reason why cygwin seems to be fine, it's because, from what I could see, it uses the UNIX I/O manager rather than the NT I/O manager for low level disk I/O, so of course, the 32-bit truncation of the multiplication, that (unless it's a side effect of adding the offset, which I don't believe it is) you do get in the original ext2fs code, does not apply since this specific code is actually not the one the official e2fsprogs uses on Windows/cygwin.

Oh well, I already had a few fixes I picked, that I was planning to upstream to e2fsprogs once I had finalized `ext` formatting in Rufus, so I guess I'll just have to add one more...

sudodus, 2020-02-12:
Den 2020-02-12 kl. 01:11, skrev Pete Batard:

> Okay, I think I have finally pinpointed the issue.
> 
> The issue has to do with this specific line of code 
> <https://github.com/pbatard/rufus/blob/9edd7492db84a5f3b8cf5d71fee98cf51882043f/src/ext2fs/nt_io.c#L700>:
> 
> offset.QuadPart = block * channel->block_size + nt_data->offset;
> 
> As I mentioned above, it has to do with a 32-bit overflow when computing 
> values, and more specifically with *DUMB* compilers not performing 
> 64-bit computations when they really should.
> 
> You see, in the line above, |offset.QuadPart| and |nt_data->offset| are 
> 64-bit values, and, since compilers are allegedly supposed to do the 
> smart thing when performing calculation, one would expect that, even as 
> |block| and |channel->block_size| are 32-bit values, the compiler will 
> be smart enough to compute and move the product of these 32-bit values 
> into 64-bit registers (because, programmatically, there's absolutely no 
> logical reason whatsoever to want to limit that multiplication to 32-bit 
> when the result is going to end up in a 64-bit variable).
> 
> Alas, this is exactly what's happening here. And it looks like both MSVC 
> and gcc suffer from that illogical behaviour, because, unless you 
> specifically tell the compiler that you do want to compute |block * 
> channel->block_size| as 64-bit (by adding an explicit cast like 
> |((uint64_t)block) * channel->block_size|), that darn multiplication is 
> performed as 32-bit, which means that, as soon as you try to write to a 
> block of data that resides above 4 GB on the media, you're writing to 
> the wrong address. Hence |e2fsck| finding loads of issues for any inode 
> above 4 GB, since none of them have been initialized... Ouch!
> 
> Damn, and I was always told that trying to micro-manage or 
> micro-optimise what compilers do was a waste of time because they are 
> supposed to be designed to do the smart thing. So much for that... Heck, 
> it almost makes one wish we were still programming in pure assembly, 
> because such an issue would certainly not happen then (since we'd very 
> explicitly specify that we are working on 64-bit register for that 
> multiplication)...
> 
> Oh and for those who wonder why we're not using a 64-bit block variable 
> in the first place, or more specifically the 64-bit version of the 
> |write_blk()| call, the reason is that, with a block size of 4096 bytes 
> or more for large volumes (and code that explicitly checks for 32-bit 
> overflow for the |block| variable), you're going to have to create a 16 
> TB or larger partition before you run into the 32-bit limit, which seems 
> a bit unrealistic *for a persistent partition*. Plus the underlying 
> |ext2fs| code I lifted still has tons of internal calls to the 32-bit 
> version of |write_blk()|, so it didn't make much sense to bother with 
> the 64-bit version... *as long as the |block * block_size| computation 
> was properly carried out using 64 bits*. Now that I've seen the result 
> of trusting compilers (and/or pre-existing code) to do the right thing, 
> I guess I should have taken a page from what |unix_io.c| does, moved the 
> core write block function to 64 bit and just added 32 → 64 bit 
> invocation for the 32-bit calls like so:
> 
> errcode_t  unix_write_blk(io_channel channel,unsigned  long  block,
> 				int  count,const  void  *buf)
> {
> 	return  unix_write_blk64(channel, block, count, buf);
> }
> 
> Now, as to the reason why cygwin seems to be fine, it's because, from 
> what I could see, it uses the UNIX I/O manager rather than the NT I/O 
> manager for low level disk I/O, so of course, the 32-bit truncation of 
> the multiplication, that (unless it's a side effect of adding the 
> offset, which I don't believe it is) you do get in the original ext2fs 
> code, does not apply since this specific code is actually not the one 
> the official e2fsprogs uses on Windows/cygwin.
> 
> Oh well, I already had a few fixes I picked, that I was planning to 
> upstream to e2fsprogs once I had finalized |ext| formatting in Rufus, so 
> I guess I'll just have to add one more...

Congratulations to this good catch, Pete :-)

Best regards
Nio

pbatard, 2020-02-12:
Okay, I will close this issue with the next commit I push, since this should now be fixed.

There's a TEST version of Rufus 3.9 you can use, that includes the fix, and which you can download [__here__](https://rufus-web.akeo.ie/testing/rufus-3.9_TEST3.exe) if you want to confirm for yourself that the problem reported above has been addressed.

closed 2020-02-12 by pbatard

sudodus, 2020-02-12:
Den 2020-02-12 kl. 18:55, skrev Pete Batard:

> Okay, I will close this issue with the next commit I push, since this
> should now be fixed.
>
> There's a TEST version of Rufus 3.9 you can use, that includes the fix,
> and which you can download [...]

Hi Pete,

I made a persistent live Lubuntu 19.10 with this TEST version of Rufus.

It looks good.

- I tested the ext3 file system with e2fsck and it was healthy.

- When booted into the persistent live system, it worked as it should.

Congratulations! Now I feel happy with Rufus again :-)

Best regards
Nio

pbatard, 2020-02-12:
Great! Thanks for reporting.

sudodus, 2020-02-17:
Den 2020-02-12 kl. 20:42, skrev Pete Batard:

> Great! Thanks for reporting.

Hi Pete,

I noticed a regression. Our new kind of persistence no longer works in 
the developing version of Ubuntu. I wrote a bug report. I am rather sure 
that it will also affect Rufus, so I suggest that check if it'affects 
you and in that case that you add heat to the bug report by clicking on 
'affects me too'.

https://bugs.launchpad.net/ubuntu/+source/casper/+bug/1863672

You find the current daily iso files via the ubuntu iso tracker

http://iso.qa.ubuntu.com/qatracker/milestones/408/builds

Best regards
Nio

pbatard, 2020-02-17:
Thanks for the heads up. I'll try to test this when I get a chance.

commit 67d324f8 (2020-02-12): [extfs] fix inodes not being initialized above 4 GB

* So, as it happens, when assigning the product of two 32-bit variables into a 64-bit one,
  compilers default to being *DUMB* and, against all reasonable expectations, do not perform
  that multiplication as a 64-bit operation (even when the code is compiled as x64). Wow,
  that's really great decision making by compiler designers if I ever saw some... Whoever
  decided that C developers would much rather want truncation and 32-bit overflows, instead
  of the expected *LOGICAL* behaviour of conducting arithmetic operations as 64-bit when th [... cut]