pbatard/rufus #1398: Blocked at formating 100%

thelittlefireman, 2019-10-27:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus is blocked at "formating 100%" with thoses settings. I try several times.
I changed NTFS to FAT32, i try lower persistant but nothing is working
[https://i.imgur.com/2Tyf31e.png](https://i.imgur.com/2Tyf31e.png)
Log
---
```
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 32 Go, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03215E73
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31027363840 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Lubuntu 18.04.2 LTS amd64'
  Size: 1.1 Go (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: lubuntu-18.04.2-desktop-amd64.iso (1.1 Go)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 28.9 Go)
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 1.9 GB)
● Creating Linux Persistence Partition (offset: 2038434304, size: 27.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Formatting (ext3)...
442367 possible inodes out of 7077884 blocks (block size = 4096)
353894 blocks (5.0%) reserved for the super user
Creating 216 inode sets: [1 marker = 2.7 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 16384 journal blocks: [1 marker = 204.8 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Cancelling
```

pbatard, 2019-10-27:
Persistence will not work with Ubuntu/Lubuntu 18.04 due to an [Ubuntu bug](https://bugs.launchpad.net/ubuntu/+source/casper/+bug/1489855) (you need to use 19.10 or later), so I'm afraid that you are wasting your time trying to use persistence with 18.04.

Also, quick formatting does __not__ apply to ext2/ext3 file system creation, which will always be much slower than NTFS/FAT32 formatting, and it's also not because you have a USB 3.0 drive that it means it's going to be fast. Unfortunately there are many USB 3.0 drives with dreadful write speed, like 5 MB/s or less, which will make any formatting operations very slow. Do you know the actual write speed of your drive?

While we are planning to look into what can be done to improve ext2/ext3 formatting speed (because we know that it is much slower than on Linux), and we are looking into other potential issues with ext2/ext3 (because it was introduced very recently), we haven't received any other reports of a formatting operation being stuck, so, considering the speeds you are observing, I would say that if you wait long enough, you'll probably see Rufus complete its task. But then again, this is pointless if you are using Lubuntu 18.04, as, if you do enable persistence, the USB will fail to boot anyway.

thelittlefireman, 2019-10-28:
huge thank for your quick response :) OK i will try with 19.10.

Should i use FAT32 or NTFS format ?

I try to burn same iso with balenaEtcher and the speed was on avg about15MB/s (2m59 to burn the iso on the key)

pbatard, 2019-10-28:
FAT32.

balenaEtcher doesn't use persistence and doesn't format a file system. You __will__ be able to achieve the same speed in Rufus if you select 'DD mode' when prompted (Remember that prompt you got that asked you whether you wanted to write the media in ISO or DD mode? That's the one), but of course you won't get persistence and you also won't be able to see the content of the media in Windows after it is written.

thelittlefireman, 2019-10-30:
I try with lubuntu 19.10 and FAT32 and NTFS but nothing works betters. On FAT32 it fails because after format the disk windows mount it and rufus loss acess to it. On NTFS is still freeze at 100% of formating

pbatard, 2019-10-30:
Did you try DD mode?
Also please provide a log of your test with FAT32. I have tested Ubuntu 19.10 a few times and never saw Rufus losing access to the drive after formatting, so I am very puzzled by your report. The log should tell me more about what is happening.

thelittlefireman, 2019-10-31:
I never show the dd mode prompt for all my test. I Will retry and put logs

pbatard, 2019-11-11:
No update &rarr; Closed.

closed 2019-11-11 by pbatard

thelittlefireman, 2019-11-11:
The dd mode prompt never appear. How can i get it ?