pbatard/rufus #1400: Fails after deleting partitions

Gordon-Beeming, 2019-10-29:
I know this doesn't follow the format, not sure how else to share the info though... please read quick, more a heads up.

- I used 3.8p version to make a bootable windows 10
- It failed after deleting the partitions
- I went to windows disk manager and created the partition again
- repeated the process and failed again, created the partition again in disk manager
- downloaded 3.7p version, everything worked fine

At this point thought if there is info I can get that can help with the project let me do this in case it's affecting others. 

Opened 3.8p again and this time it worked. I was hoping it would fail so I could get the logs for you but unfortunately not, without logs I'm not sure how helpful this can be but only thought about it after it worked.

Can I suggest maybe adding to the end of error messages that people come here to log the issue? This way least they are able to provide all the details at the time 😁.

Gordon-Beeming, 2019-10-29:
Managed to get it to break 😁... When using 3.7p and it was working I killed the process early so I could run the 3.8p version and get a log... I let the 3.8p version run to completion now and then to see if it only broke once it was finished before I run 3.8p again and this time it broke. 

**NOTE:** I didn't close the app before running it again, but I did clear logs first from the successful complete, this seems to be a different issue maybe because the partition was created when it failed and after closing the app and opening it again it's working again

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive3':
● Partition 1 (offset: 1048576, size: 14.9 GB)
● Partition 2 (offset: 16016958976, size: 512 KB)
Opened \\.\PhysicalDrive3 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Could not zero Main Data Partition: [0x00000015] The device is not ready.
● Creating UEFI:NTFS Partition (offset: 16016958976, size: 512 KB)
Writing UEFI:NTFS data...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Write error: [0x00000015] The device is not ready.

Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A81D)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 14.9 GB (16018046976 bytes)
  Start Sector: 0, Boot: No
Device not ready → Trying to cycle port...
Cycling port 4 (reset) on \\?\usb#vid_8087&pid_8008#5&217efabb&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Please wait for the device to re-appear...
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A81D)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 14.9 GB (16018046976 bytes)
  Start Sector: 0, Boot: No
Closed Windows format prompt
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A81D)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 14.9 GB (16018046976 bytes)
  Start Sector: 0, Boot: No

The bits of the log that would have now been trimmed out from the first run are below (copied from the top of the new run that is working)

Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A81D)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 14.9 GB (16018046976 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_10_consumer_editions_version_1903_updated_oct_2019_x64_dvd_2055d1cf.iso (5.1 GB)

pbatard, 2019-10-29:
Yes, Windows 10 can be a major pain in the ass when there exist multiple partitions or a partition using a file system it cannot recognize. It would appear that there's a privileged process in Windows (that is too elevated to be listed) that seems to like accessing these kind of drives for longer than it should. We are aware of this sporadic problem, but we don't have a workaround for it, besides unplugging and re-plugging the drive until Windows lets you properly access it.

Note that, no, despite what you are inferring, it is not a 3.7 vs 3.8 issue, as there were only 3 sets of changes between 3.7 and 3.8 (3.8 was a BUGFIX release with regards to `ext3` formatting, and the code changes that were applied have not altered the way partitions or file systems are created/deleted for non `ext3` operations), so please be mindful, as you are doing here, about inferring that this is a 3.8 vs 3.7 regression and that so many people are going to be affected by it that reports are just going to pour in (which, if you had followed the checklist and, as it asked, look for similar issue but found none, would have told you that maybe your assertion is wrong).

There's not much you can provide here that we don't already know, so I'm going to close this issue. Next time, please follow the checklist.

closed 2019-10-29 by pbatard

Gordon-Beeming, 2019-11-03:
Noted, next time I'll rather not help with anything 👍🏻. I did mentioned that I repeatedly tried 3.8 and it didn't work and then when trying 3.7 it did, that felt like a regression but as you said there are other things at play in the OS.

It's also probably not a great idea to treat people trying to help like idiots, I was suggestion maybe something was wrong which in a later comment I said it looks like that is not the case because of other behavior I saw.