pbatard/rufus #1403: Unusable USB Stick

SteveF48, 2019-11-10:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I've tried to create a bootable USB from this ISO 4 times, using 2 different USB sticks. Each time the result is that the USB stick is not recognised by Windows.
The disk shows a 4mb partition the remainder is aunallocated

Log
---
```
<FULL LOG>
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 943, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0EE6F08B
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 320, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
  Detected Grub version: 2.02
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Pop_OS 19.10 amd64'
  Size: 2.1 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: pop-os_19.10_amd64_intel_10.iso (2.1 GB)

Computing checksum for 'C:\Users\steve\Downloads\pop-os_19.10_amd64_intel_10.iso'...
  MD5:	 198def0b72161e4bc45d057f7eb845e2
  SHA1:	 f6cbf7daf9009a2dae50c7b6bc459b9a122873ce
  SHA256: 00e426b578338d9291e0532e5079157174f9b200dabb075cb8caf2d3b713046c
```

pbatard, 2019-11-10:
2 things:
1. Not all file systems are recognized by Windows. Especially if an ISO (such as PopOS) decided to create an ISOHybrid image using one of the various Linux file systems such as `ext4` and you write that to an USB, Windows will not be able to mount the `ext4` partition so you __may__ think that the USB is not recognized by Windows whereas it is just the file system that isn't. In other words, there's nothing unusual to Windows not being able to mount a Linux partition and it doesn't mean that your USB drive can't boot. The way the PopOS maintainers probably created their ISO is that they have a small (4 MB) EFI System Partition, which Windows can mount (because it uses the FAT file system) and a larger `ext#` partition with the OS which Windows can't, giving you the impression that there is an issue when everything is perfectly normal. Have you actually tried booting the drive?
2. You can easily restore the drive to a single partition that Windows will recognize by following [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive).

As I see nothing related to Rufus usage here (you will see the same "problem" if you use Etcher for instance) I will close this issue.

closed 2019-11-10 by pbatard

SteveF48, 2019-11-10:
Thanks for the rapid response.
I only ran rufus 4 times, because the usb wouldn’t boot.
I will see whether the developers have a solution.
Regards
Steve