pbatard/rufus #1404: Link to Syslinux files on SSL failure

cup, 2019-11-11:
When I try to use Rufus I get this:

~~~

This image uses Syslinux 6.04 but this application only includes the
installation files for Syslinux 6.03/2014-10-06.

As new versions of Syslinux are not compatible with one another, and it wouldn't
be possible for Rufus to include them all, two additional files must be
downloaded from the Internet ('ldlinux.sys' and 'ldlinux.bss'):
~~~

and then:

~~~
This system's SSL library is too old to be able to access this website.
~~~

my concern is that users shouldnt have to rely on rufus to get these files, as
they can be aquired regard of any SSL library:

- https://rufus.ie/files/syslinux-6.04/ldlinux.sys
- https://rufus.ie/files/syslinux-6.04/ldlinux.bss

perhaps a better solution is to just pop a message with the links where they can
be downloaded, or pop a message whenever the SSL fails.

pbatard, 2019-11-11:
As opposed to browsers that use their own custom SSL libraries, and therefore manage to access modern version of SSL secured servers such as GitHub, Rufus relies on the Windows networking libraries, and, if you happen to be running on Windows 7, an OS that is going to be obsoleted in about two months, __and you haven't applied all the system updates__ (because if you do, then the system SSL libraries __will__ be updated and allow you to access GitHub) then you will face the issue you described above.

However, considering what I explained above, I hope you can appreciate that the issue is far from being the generalized problem that you may imagine it is, but instead is restricted to a __minority__ of Windows 7 users.

As such it makes little sense to take additional steps besides notifying the users that they should keep their OS up to date, because __that is really the core of the issue__.

If _"users shouldnt have to rely on rufus to get these files"_ then I will posit that _"users should keep their OS fully up to date"_ instead of expecting applications like Rufus to work around an issue that only exists due to users choosing to use a non up to date OS (and no, I am not talking about upgrading from Windows 7 to Windows 10. I am talking about making sure that you have applied all the Windows 7 updates that are published by Microsoft as some of those are important if you do want to access SSL websites).

I will therefore close this issue.

closed 2019-11-11 by pbatard

cup, 2019-11-11:
@pbatard honestly I think you are lying to yourself about this. Here are some
essentially duplicates, this is only since November 2018:

- https://github.com/pbatard/rufus/issues/1357
- https://github.com/pbatard/rufus/issues/1378
- https://github.com/pbatard/rufus/issues/1311
- https://github.com/pbatard/rufus/issues/487
- https://github.com/pbatard/rufus/issues/1276
- https://github.com/pbatard/rufus/issues/1232

As you said, my browser is up to date, so that negates a need for my internal
SSL to need to be up to date. But because of your refusal to bend on this, its
causing a lot of pointless pain on the userbase.

Please reconsider.

pbatard, 2019-11-11:
I __tested__ it. As I explained, it's not because browsers, that use their own internal SSL library, can access a modern SSL website that an OS that hasn't been updated will. Please try to access the files with Internet Explorer, you will see.

And I will also encourage you to read through the issues you linked, where I pretty much explain the same thing. If you have an example of a vanilla Windows 7 system, where all the updates have been applied, that can't access a modern SSL website, I'll be happy to hear it, because I did test that a vanilla Windows 7, with all updates applied, could access GitHub in SSL mode (where the files are being served) be it from Rufus or Internet Explorer.

And no, as I explained at length in the issues you linked, I'm not going to add a__custom__ multi MB SSL library so that users of an OS that is going to become obsolete in 2 months can spare themselves the trouble of keeping it up to date.

cup, 2019-11-11:
No one is asking you to bundle anything, thats the point. All that needs to be done is point users to where the files are, and they can work out the best way to get them.

Your demand that users get the files "your way" is pointless and time-wasting, as it requires the user update their OS, when they can simply download the files with their browser if they just know where the files are located.

I agree that its important that users keep the OS up to date, but its not Rufus job to be the OS or security police. It Rufus job to produce a bootable USB drive. Devs involved with Rufus should focus on that and keep "software politics" out of it.

pbatard, 2019-11-11:
So keeping an OS up to date so that malware cannot exploit vulnerabilities is "pointless"?

Do you understand how malware spreads?

You may want to read on [WannaCry](https://en.wikipedia.org/wiki/WannaCry_ransomware_attack#Attack) maybe:

> Organizations that had not installed Microsoft's security update from April 2017 were affected by the attack

As a software developer, I very much see it as my job to invite users to keep their OS up to date when I have the opportunity to do so, rather than, as you suggest, take a backseat and eventually watch these same users fall victim to malware. If I have to inconvenience people so that they keep their OS up to date, so that I can do my part in reducing the spread of all these malware attacks that should have been contained in scope if people simply kept their OS up to date, you bet I will.

If you disagree with that, then I will kindly ask that you use a different software

cup, 2019-11-11:
Ok then, well at least youve made it clear that you wish to intrude into users software decisions. Luckily other projects are available that dont do that:

https://github.com/unetbootin/unetbootin

I will give them a try next time this comes up.

Cheers.