pbatard/rufus #1406: What does "Add fixes for old BIOSes" do ?

SkybuckFlying, 2019-11-13:
There is no information on the internet about this advanced option in rufus, nothing about it in the FAQ. Google returns also pretty much nothing about this.

As far as I can tell it creates some kind of extra mini-partition on the same usb flash stick. Maybe it also makes the usb-stick seem like it's a harddisk instead of a removeable (not sure about that?).

This addition of mini-partition complexifies the installation of a boot manager for usb stick installation ?

Where to place boot manager loader ?

(Since you know so much about paritions and such maybe you can answer these questions below:)

(I want to "burn" Game Over ISO to usb stick, and then install from usb stick to same usb stick, and then boot into usb stick full OS):

What I have tried so far:

Burn Game Over ISO to usb stick/fat32 4 GB partition.
Resize FAT32 to 4 GB in installer.

Use free space to create:

Primary partition needed with mounting point set to: \
Swap partition needed with file system linuxswap

Then I have to set install boot manager ?

Different options for install boot manager occur.

Something with "install the usb stick mbr/dev/dba"
Something with "install to \"
And some others...

There is also a mini partition with 36 kb if I choose "Add fixes for old BIOSes" in RUFUS.

I am unsure if flags need to be set as well ?!

I tried setting \boot to primary partition.

I tried setting swap for linuxswap.

So far the things I tried resulted in the old system booting into the live cd/installer but not into the supposedly full install ?!

So I think it's not booting into the correct partition ???

So how do I setup the installation so that on reboot it will boot into the correct/full installation partition and now the live cd/fat32 ?

I also tried deleting the fat32 and mini partition... but during install it will say: "in mounting use" so that does not seem possible ?

(During boot of livecd there is also "toram" option ? To Ram ? What does this do ? Does it copy the entire ISO to RAM ? and then install from there ? Would this allow me to delete far32 and mini parittion ? I will also ask this in seperate thread, so if you don't know then just cut/skip it)

The old system also had some boot issues with other setups, like it would only show grub, probably when BIOS fixes was not added/set.

Also it would produce some apic issue in other situations, but that might be fixeable by adding "noapic" boot options. I will keep trying to create this.

Bye for now,
  Skybuck.

SkybuckFlying, 2019-11-13:
Addition:

How does booting actually work ?

Does it first boot into mini-image ?

Then into fat 32 image ?

Then into main partition ?

Can booting like this happen in stages/linked ? Like linked booting ?

I am a bit fuzzy about this.

If you know a good link that explains partitions and how boot programs and boots happen that would enlighten me a bit and could help with using rufus and os installation programs better.

Perhaps some links inside rufus to that kind of information would be helpfull to user, right now I am kinda in the dark about all of this.

JonnyTech, 2019-11-13:
That is not possible and is outside of the usage scope for Rufus

SkybuckFlying, 2019-11-13:
What is not possible ? If it's about the visuals, that idea is for a help file or so.

pbatard, 2019-11-13:
You know, as a developer I'm getting a bit tired of spending time trying to describe what options do, by adding __tooltips__, and then being told that there's no help whatsoever on the various UI elements. So here:

[image]

Now, the description is terse because, as opposed to what many people think, __this is not a magic bullet__. This options is really there for:
1. People who are using pre 2000 BIOSes (i.e. "old" BIOSes)
2. People in the group above who are willing to give it a try and accept that if it manages to fix their problem (that has to do with the limitation of using __some__ pre-2000 BIOSes), great, but if it doesn't, then they'll have to figure out some other way to fix their issue on their own.

So, really, all you need to do here is:
1. Find out if you are using a pre 2000 BIOS
2. Try the option and see if it helps in any way with your old BIOS issue and accept that, if it doesn't, __you'll have to spend time figuring out how to fix your specific problem on your own, because trying the option and see whether it works or not to fix an issue that ONLY has to do with using an old BIOS__ is the best you're going to get from me. I genuinely don't have the time to investigate every single user's problem out there or go through time consuming hand holding. The internet contains more than enough data for you to figure thing on your own.

I will now close this issue.

closed 2019-11-13 by pbatard

SkybuckFlying, 2019-11-13:
Funny enough the hint message don't work in certain cases, I am pretty sure this is what happened in my case many times, because I am pretty sure I tried to get the hint message working, but it never worked or I can't remember.

It's very reproduceable.

So new issues opened #1407 see there how to reproduce this behaviour.

This only describes one case where hint message does not pop-up, there might be others ! ;)

SkybuckFlying, 2019-11-13:
So is this all it does ? "Align partitions ? Perhaps by injecting a mini partition in between ?!

Explanation is still a bit shady... but perhaps requires more insight in how partitions and booting from partitions work.

At least there is now some information on when to enable this fix... "if bios is older than 2000" that information should be in hint message, much more usefull.

What does the "etc" mean/hint at in the hint message ?!?

Sounds like it might be doing some else/more...