pbatard/rufus #1407: Hint message for "Add fixes for old BIOSes" does not work when checkbox checked.

SkybuckFlying, 2019-11-13:
HAHA !

I AM BAAAACCCCCCKKKKK.

DO I HAVE A SURPRISE FOR YOU:

THE HINT MESSAGE DON'T WORK !

At least not in the following way:

1. Setup rufus so it's not grayed.

2. Click the checkbox so it's checked.

3. Hold the mouse over the checkbox.

SURPRISE !

NO HINT MESSAGE ! ;)

Bye,
  Skybuck.

SkybuckFlying, 2019-11-13:
It seems to be very funny behaviour and as far as I am concerned this behaviour is completely flawed.

It's very likely a user will checkbox it and then try and read it, but funny enough as long as the cursor remains on the checkbox the pop-up never shows up.

One has to "exit" the checkbox first completely... by moving away the mouse cursor... then moving back into it... and then not clicking on the checkbox.

It gets even worse.... when it's grayed out it doesn't even show anything.

So stop blaming your users for not reading hint messages if the hint implemention is complete garbage ! =D

pbatard, 2019-11-13:
This is standard Windows behaviour. You seem to be thinking that every single application has code that detects mouse position and that creates the tooltip rectangle, whereas all we are doing really is tell Windows "Here's the tooltip that's associated with this control for you to display".

Therefore, feel free to engage with Microsoft on the behaviour you report, rather than pointlessly look for elements to blame a developer you're unhappy with because they told you they don't have the time to do your work for you.

closed 2019-11-13 by pbatard