pbatard/rufus #1408: Can't copy manjaro in DD mode

Hanzemon, 2019-11-13:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
After doing some research I have come to the conclusion that in order to install manjaro in your computer you have to copy it in "DD mode" however, when I try to copy it using rufus there is no option for me to use "DD mode". 

If I try to copy my "Xubuntu" image on the usb stick it asks me if I want to copy it as ISO or DD but when I try to copy manjaro I can't choose DD mode.

Log
---
```
<FULL LOG>
Rufus x86 v3.8.1580
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
Using autorun.inf label for drive E: '4GB'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4006576128 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04~manjaro
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MJRO1811'
  Size: 2.4 GB (Projected)
  Uses: GRUB2
  Uses: EFI
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: manjaro-kde-18.1.1-191015-linux53.iso (2.4 GB)
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
Using autorun.inf label for drive E: '4GB'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4006576128 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
Using autorun.inf label for drive E: '4GB'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4006576128 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
Using autorun.inf label for drive E: '4GB'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4006576128 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
Using autorun.inf label for drive E: '4GB'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4006576128 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
Using autorun.inf label for drive E: '4GB'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4006576128 bytes)
  Start Sector: 2048, Boot: No
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 3.7 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Writing Image...
Remounted \\?\Volume{a3f71467-fcf7-11e9-be1b-806e6f6e6963}\ as E:

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 5011080, Boot: No
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1607)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 5011080, Boot: No

pbatard, 2019-11-13:
The answer: Because Manjaro is incompatible with ISO mode, we only provide the option to write it in DD mode in recent versions of Rufus. And of course, it wouldn't make much sense to pop up a dialog that asks the following with only one option:

Please select the mode you want to write this ISO in:
* DD mode

So we don't.

TL;DR: Your Manjaro ISO __will__ be written in DD mode. You don't have to "force" it or believe the guides that say that if you don't see DD mode you're doing it wrong.

TL;DR2: It never hurts to __try__ before assuming that there's anything wrong. Have you actually tried booting the ISO that was written by Rufus?

As I believe I answered your question, I will close this issue.

closed 2019-11-13 by pbatard

Hanzemon, 2019-11-14:
Yes, I've tried MANY times to install manjaro after copying it with rufus.
The result is always the same: After selecting my usb stick as boot device I get a black screen with the word "GRUB" on the top left corner. It stays there and nothing ever happens.

pbatard, 2019-11-14:
Then I'm afraid you will have to take this issue to the Manjaro people as you would have the same issue if you had used `dd` on Linux or Etcher. This issue tracker is only for Rufus related issue, which your problem doesn't appear to be, and I'm afraid I have no insight as to why a Manjaro ISO written in DD mode may or may not boot on specific systems.

Hanzemon, 2019-11-14:
Well NOW that you say it I know, so thank you. Had I known before I wouldn't have bothered you. Peace.